Rahix/avr-hal #226: Converting analog values to f32 on arduino nano and uno behave strange

ChrisRega, 2021-11-12:
Hello,

I got an ADXL335 on arduino nano and uno and wanted to do some float-math with the analog values. I stumbled over the following:
```
#[arduino_hal::entry]
fn main() -> ! {
    let dp = arduino_hal::Peripherals::take().unwrap();
    let pins = arduino_hal::pins!(dp);
    let mut adc = arduino_hal::Adc::new(dp.ADC, Default::default());
    let a0 = pins.a0.into_analog_input(&mut adc);
    let mut serial = arduino_hal::default_serial!(dp, pins, 9600);
    loop {
        let input_data = a0.analog_read(&mut adc);
        let test_data_f = f32::from(input_data);
        ufmt::uwriteln!(&mut serial, "test: {} ", input_data).void_unwrap();
        let test_data_f_str = uFmt_f32::Three(test_data_f);
        ufmt::uwriteln!(&mut serial, "value as f32: {} ", test_data_f_str).void_unwrap();
        ufmt::uwriteln!(&mut serial, "f32 back to u16: {} ", test_data_f as u16).void_unwrap();
        arduino_hal::delay_ms(500);
    }
}
```

This results in the output:
```
test: 328 
value as f32: 0.000 
f32 back to u16: 0
```

The strange thing is, that if i hardcode the input value to an u16 value in the same range like:

```
let input_data: u16 = 400;
```

and then run the code, the float conversion works:

```
test: 400 
value as f32: 400.000 
f32 back to u16: 400 
```

Am I missing something here?
Many thanks for your help in advance!

Rahix, 2021-11-13:
This is probably one of the known floating point problems in the compiler.  Search the issues in rust-lang/rust, you'll find reports similar to yours.

For now, I would recommend to not use floats - it isn't a great idea on AVR anyway because all floating point operations must be implemented in software which is slow and eats lots of space.  Often you can, for example, use the `Fixed<i16>` type from the `fixed` crate instead.

djdisodo, 2022-06-16:
[image]
it's not related to this issue but this failes to compile if i don't turn of lto

[image]
i wasn't able to represent the issue

Rahix, 2022-06-26:
> it's not related to this issue but this failes to compile if i don't turn of lto

This happens when `compiler-builtins` is not built in release-mode because it then generates references to items from `libcore` which it is not allowed to do.  In release-mode these references are optimized out.  Also check #131.

> i wasn't able to represent the issue

Yeah, with the latest compiler changes I think these problems should be resolved.  Let's close this issue for now.

closed 2022-06-26 by Rahix