Rahix/avr-hal #227: So close to working :-(

RandoSY, 2021-11-19:
This is a great project, thanks for posting!

Very new to rust, but amazed how easy to get setup and testing. Seem to be building OK, but wondering where to set -P to COM12 statement, I've been using the RAVEDUDE_PORT env variable clumsily from the command line to get it flashing OK. 

Here's a run that hangs with no serial  output

`C:\Users\randa\Desktop\avr-hal-main\examples\arduino-uno>cargo` run --bin uno-i2cdetect
    Finished dev [optimized + debuginfo] target(s) in 0.12s
     Running `ravedude uno -cb 57600 C:\Users\randa\Desktop\avr-hal-main\target\avr-atmega328p\debug\uno-i2cdetect.elf`
       Board Arduino Uno
 Programming C:\Users\randa\Desktop\avr-hal-main\target\avr-atmega328p\debug\uno-i2cdetect.elf => COM12

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e950f (probably m328p)
avrdude: erasing chip
avrdude: reading input file "C:\Users\randa\Desktop\avr-hal-main\target\avr-atmega328p\debug\uno-i2cdetect.elf"
avrdude: writing flash (1600 bytes):

Writing | ################################################## | 100% 0.04s

avrdude: 1600 bytes of flash written
avrdude: verifying flash memory against C:\Users\randa\Desktop\avr-hal-main\target\avr-atmega328p\debug\uno-i2cdetect.elf:
avrdude: load data flash data from input file C:\Users\randa\Desktop\avr-hal-main\target\avr-atmega328p\debug\uno-i2cdetect.elf:
avrdude: input file C:\Users\randa\Desktop\avr-hal-main\target\avr-atmega328p\debug\uno-i2cdetect.elf contains 1600 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 0.03s

avrdude: verifying ...
avrdude: 1600 bytes of flash verified

avrdude: safemode: Fuses OK (E:00, H:00, L:00)

avrdude done.  Thank you.

  Programmed C:\Users\randa\Desktop\avr-hal-main\target\avr-atmega328p\debug\uno-i2cdetect.elf
     Console COM12  at 57600 baud

--------------------------------------------------------------------------------------------------------

There it hangs and I don't see any output from the port. 

Ideas?

RandoSY, 2021-11-19:
I just noticed....

avrdude: safemode: Fuses OK (E:00, H:00, L:00)

???

Rahix, 2021-11-20:
Hm, just to rule out any oddities; did you try the `uno-usart` example?

> avrdude: safemode: Fuses OK (E:00, H:00, L:00)

Yeah, we don't program the fuses right now... Try flashing a program with Arduino IDE first to set them, just in case.

RandoSY, 2021-11-20:
Thank you for the kind consideration of your prompt reply!

I know from hard one experience to get on a good, stable **Ubuntu 20.04** machine for testing. 

Once I had jumped  through all the hoops to install rust, and ravedude, it runs like a champ on a low cost UNO clone. 
So excited to bring this sophisticated software technology to the Arduino hardware platform!

  Programmed /home/rasyoung/Codetest/AVR/Rust AVR/avr-hal-main/target/avr-atmega328p/debug/uno-i2cdetect.elf

     Console /dev/ttyUSB0 at 57600 baud

Write direction test:
-    0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:       -- -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- 57 -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --

Read direction test:
-    0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:       -- -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- 57
^Z
[2]+  Stopped                 cargo run --bin uno-i2cdetect

Begs the question what slight irregularity is keeping the Windows 11 console from working properly?

Rahix, 2021-11-21:
Maybe related to https://github.com/Rahix/avr-hal/issues/176?

Fundevoge, 2023-08-27:
This seems to be an issue with `serialport`: When changing the buffer size from 4098 to 1 solved the problem for me. 
console.rs, line 24: `let mut buf = [0u8; 4098];`
Might not be the most efficient solution, maybe adding a cli argument to determine the buffer size could be an option.

Rahix, 2023-08-28:
How about `#[cfg(target_os = "windows")]` (not sure what the exact cfg line was) for the single-byte buffer?  Can you send a PR?

Fundevoge, 2023-08-28:
Seems good, did exactly that in #433

Rahix, 2023-08-29:
Fixed by #433.

closed 2023-08-29 by Rahix