Rahix/avr-hal #231: Heapless Support

SyedAhkam, 2022-01-03:
I have been trying to get [heapless](https://github.com/japaric/heapless) to work.

First off, to get it to compile, I had to disable atomic types by modifying the [build.rs](https://github.com/SyedAhkam/heapless/blob/master/build.rs) file.

Further, everything seemed to be working but then I received this compile error:

```
error: linking with `avr-gcc` failed: exit code: 1
  |
  = note: "avr-gcc" "-mmcu=atmega328p" "-Wl,--as-needed" "-L" "/home/syed/.rustup/toolchains/nightly-2021-01-07-x86_64-unknown-linux-gnu/lib/rustlib/avr-atmega328p/lib" "/home/syed/dev/learning/arduino/rust/lcd/target/avr-atmega328p/debug/deps/lcd-9504018c68ef322d.heapless-552d8457291ab78e.heapless.vkav1gxd-cgu.0.rcgu.o.rcgu.o" "-o" "/home/syed/dev/learning/arduino/rust/lcd/target/avr-atmega328p/debug/deps/lcd-9504018c68ef322d.elf" "-Wl,--gc-sections" "-no-pie" "-L" "/home/syed/dev/learning/arduino/rust/lcd/target/avr-atmega328p/debug/deps" "-L" "/home/syed/dev/learning/arduino/rust/lcd/target/debug/deps" "-L" "/home/syed/.rustup/toolchains/nightly-2021-01-07-x86_64-unknown-linux-gnu/lib/rustlib/avr-atmega328p/lib" "-Wl,--start-group" "-Wl,--end-group" "-Wl,-Bstatic" "/home/syed/dev/learning/arduino/rust/lcd/target/avr-atmega328p/debug/deps/libcompiler_builtins-7df01ff13f6d3d64.rlib" "-Wl,-Bdynamic" "-lgcc"
  = note: /usr/bin/avr-ld: /home/syed/dev/learning/arduino/rust/lcd/target/avr-atmega328p/debug/deps/libcompiler_builtins-7df01ff13f6d3d64.rlib(compiler_builtins-7df01ff13f6d3d64.compiler_builtins.81r5nep0-cgu.7.rcgu.o): in function `compiler_builtins::int::shift::Ashl::ashl':
          /home/syed/.cargo/registry/src/github.com-1ecc6299db9ec823/compiler_builtins-0.1.36/src/int/shift.rs:(.text._ZN17compiler_builtins3int5shift4Ashl4ashl17hc1a361f15810159dE+0x150): undefined reference to `core::panicking::panic'
          /usr/bin/avr-ld: /home/syed/dev/learning/arduino/rust/lcd/target/avr-atmega328p/debug/deps/libcompiler_builtins-7df01ff13f6d3d64.rlib(compiler_builtins-7df01ff13f6d3d64.compiler_builtins.81r5nep0-cgu.7.rcgu.o): in function `compiler_builtins::int::shift::Lshr::lshr':
          /home/syed/.cargo/registry/src/github.com-1ecc6299db9ec823/compiler_builtins-0.1.36/src/int/shift.rs:(.text._ZN17compiler_builtins3int5shift4Lshr4lshr17h89f87de4d526e887E+0x1f8): undefined reference to `core::panicking::panic'
          /usr/bin/avr-ld: /home/syed/dev/learning/arduino/rust/lcd/target/avr-atmega328p/debug/deps/libcompiler_builtins-7df01ff13f6d3d64.rlib(compiler_builtins-7df01ff13f6d3d64.compiler_builtins.81r5nep0-cgu.1.rcgu.o): in function `compiler_builtins::int::specialized_div_rem::u32_div_rem':
          /home/syed/.cargo/registry/src/github.com-1ecc6299db9ec823/compiler_builtins-0.1.36/src/int/specialized_div_rem/norm_shift.rs:(.text._ZN17compiler_builtins3int19specialized_div_rem11u32_div_rem17h6bc3b8db1ec35af8E+0x718): undefined reference to `core::panicking::panic'
          collect2: error: ld returned 1 exit status
```

What could be the issue here? (I have no idea how avr-gcc is configured)

Rahix, 2022-01-03:
Might be #131.

SyedAhkam, 2022-01-03:
Thanks for pointing me there!

It does compile plus 'work' now using the workaround mentioned in #131.

For anyone else stumbling upon this issue:

Adding:
```rs
[profile.dev.package.compiler_builtins]
overflow-checks = false
```
to the `Cargo.toml` file is the workaround.

closed 2022-01-03 by SyedAhkam

stappersg, 2022-01-04:
On Mon, Jan 03, 2022 at 10:16:30AM -0800, Syed Ahkam wrote:
> Thanks for pointing me there!
> 
> It does compile plus 'work' now using the workaround mentioned in #131.
> 
> For anyone else stumbling upon this issue:
> 
> Adding:
> ```rs
> [profile.dev.package.compiler_builtins]
> overflow-checks = false
> ```
> to the `Cargo.toml` file is the workaround.
> 

Consider to convert that to a merge request.

Rahix, 2022-01-04:
> Consider to convert that to a merge request.

Please no, this is not a well working solution as of now - cargo unconditionally complains about this setting.  There needs to be a proper solution in cargo's build-std instead...  It is a bug that `compiler-builtins` is built without `overflow-checks = false` in the first place - this should never be possible.