Rahix/avr-hal #236: "cargo generate --git ... " command not working

exo-cortex, 2022-01-21:
The command "cargo generate --git https://github.com/Rahix/avr-hal-template.git" as described in the "starting your own project" section on the README page does not work for me (" error: no such command `generate` ").
I installed rustup with the curl script on fedora.

stappersg, 2022-01-21:
On Fri, Jan 21, 2022 at 01:18:04AM -0800, exocortex wrote:
> The command "cargo generate --git
> https://github.com/Rahix/avr-hal-template.git" as described in the
> "starting your own project" section on the README page does not work
> for me (" error: no such command `generate` ").
> I installed rustup with the curl script on fedora. 
> 

Please report back after visiting https://github.com/cargo-generate/cargo-generate

exo-cortex, 2022-01-21:
I got it to work by using "cargo install cargo-generate"-command. I didn't know about sub-commands. Thank you very much!!!

Rahix, 2022-01-22:
> Maybe some little hint for newbies like me could be added to the README page?

The README already says _"The best way to start your own project is via the [`avr-hal-template`](https://github.com/Rahix/avr-hal-template) which you can easily use with [`cargo-generate`](https://github.com/cargo-generate/cargo-generate)"_ ... Not sure if it might help to make it more explicit.  If you want, just send a PR rephrasing this part.

stappersg, 2022-01-25:
On Sat, Jan 22, 2022 at 03:59:23AM -0800, Rahix wrote:
> If you want, just send a PR rephrasing this part.

#239

Regards
Stappers

closed 2022-01-26 by Rahix