Rahix/avr-hal #237: External Interrupts

tsemczyszyn, 2022-01-22:
I'm new to embedded programming and rust. I have a rotary encoder and it seems I should be able to use external interrupt pins to tell when the encoder is being turned. I tried to refer to the timer interrupt example and adapt it to work in my use case but nothing I'm doing seems to work.

I thought I could do something like
```
#[avr_device::interrupt(atmega328p)]
fn INT0() {
       CLK_PIN_INT.store(true, atomic::Ordering::SeqCst);
}
```
to toggle a global flag and then read that in my main function but it doesn't seem to be working as I thought. I did enable the interrupts with `unsafe { avr_device::interrupt::enable() };`. I'm a bit stuck. Are there any examples out there for doing this type of thing?

stappersg, 2022-01-22:
On Fri, Jan 21, 2022 at 04:15:30PM -0800, Travis Semczyszyn wrote:
> Are there any examples out there for doing this type of thing?

Please tell how usefull #235 is.

tsemczyszyn, 2022-01-22:
Yes this has worked for me. There's a lot I don't understand but I have been able to get it working. Thanks.

closed 2022-01-22 by tsemczyszyn

stappersg, 2022-01-22:
On Sat, Jan 22, 2022 at 09:58:02AM -0800, Travis Semczyszyn wrote:
> There's a lot I don't understand

That is known as knowledge.  And knowledge is an island.
When it grows, the shore line of the unknown increases.  8^)

Rahix, 2022-01-23:
@tsemczyszyn, if you don't mind, I'd love a pull request with a small example so we have something for others who have the same problem in the future.

tsemczyszyn, 2022-01-25:
Sure I'll try and put something together.