Rahix/avr-hal #241: DummyOutputPin for Chipselect SPI

Fatalon, 2022-01-26:
Hey,
I am still new to Rust and try to play around with your avr-hal library in combination with other public driver crates. 
I wanted to use the available [Sh1106 driver by jamwaffles](https://github.com/jamwaffles/sh1106/) over SPI with your lib.
But it looks like your library does only support SPI, when it is allowed to take ownership over the Chipselect pin and controls its state.
I would like to pass your library kind of a dummy Pin, such that the Sh driver could take ownership over the chipselect pin. I know, that your library does return a `Chipselect` instance, but it looks to me, that it is impossible to get the real port back from this instance. Maybe I could try to transform your `Chipselect` instance in something that would satisfy a OutputPin parameter field, but I currently dont know how to do that. Also, I dont want to fiddle with unsafe pointers in this case, as it looks to me like a typical use-case.

The mentioned library already offers a possible way to pass a dummy output pin (named NoOutputPin, that overwrites the set_high() and set_low() methods and replaces the object with a PhantomData object, respectively.) (https://github.com/jamwaffles/sh1106/blob/master/src/builder.rs around line 129).

Would something like this be possible/ make sense in your library? I think a way to get the ownership over the CS pin by bypassing your constructor is better then getting a struct back, that defines a different API to what a normal driver would expect.
Sincerely, Fatalon

Fatalon, 2022-01-27:
Found your comment inside spi.rs, that ChipselectPin is just a wrapper object for the OutputPin, but implements the same traits.
```
/// Wrapper for the CS pin
///
/// Used to contain the chip-select pin during operation to prevent its mode from being
/// changed from Output. This is necessary because the SPI state machine would otherwise
/// reset itself to SPI slave mode immediately. This wrapper can be used just like an
/// output pin, because it implements all the same traits from embedded-hal.
```
Therefore I created a new issue in the rust forum, as I just unable to properly use your lib instead of an actual missing feature
: https://users.rust-lang.org/t/cast-struct-to-dynamic-trait-object/70944

Fatalon, 2022-01-28:
after some conversation in the forum, it turned out, that there may be an issue with the existing Error type of the Chipselect pin.
Your OutputPin implementation uses the ErrorType `Infallible`
```
// Implements OutputPin from embedded-hal to make sure external libraries work
impl<PIN: PinOps> OutputPin for Pin<mode::Output, PIN> {
    type Error = core::convert::Infallible;

    fn set_high(&mut self) -> Result<(), Self::Error> {
        self.set_high();
        Ok(())
    }

    fn set_low(&mut self) -> Result<(), Self::Error> {
        self.set_low();
        Ok(())
    }
}
```

But the ChipselectPin only provides a Void Type as Error

```
impl<CSPIN: port::PinOps> hal::digital::v2::OutputPin for ChipSelectPin<CSPIN> {
    type Error = crate::void::Void;
    fn set_low(&mut self) -> Result<(), Self::Error> {
        self.0.set_low();
        Ok(())
    }
    fn set_high(&mut self) -> Result<(), Self::Error> {
        self.0.set_high();
        Ok(())
    }
}
```

Is there a motivation behind the different error types that are used for the OutputPins or could you/we change the type Error for Chipselect to `Infallible`

Rahix, 2022-01-28:
We can move it to `Infallible`, `void::Void` is essentially its legacy predecessor.  Would you mind sending a PR?

Fatalon, 2022-01-29:
Sure, no problem. 
One other thing I noted, when trying to apply the workaround solution of the forum, was that it looks like `PinOps` is currently not available outside of the library, even though it is marked public.
Is there a reason behind that or would you mind, if I include that in the PR aswell?

Rahix, 2022-01-30:
> One other thing I noted, when trying to apply the workaround solution of the forum, was that it looks like PinOps is currently not available outside of the library, even though it is marked public. Is there a reason behind that or would you mind, if I include that in the PR aswell?

Hmm, as far as I can see, [`avr_hal_generic::port::PinOps`](https://rahix.github.io/avr-hal/avr_hal_generic/port/trait.PinOps.html) is publicly reachable without problems.  Or am I missing something?

closed 2022-01-30 by Rahix

Fatalon, 2022-01-30:
Ah found my issue. I didn't recognized, that this repo contains multiple libraries, that need to be added independently. After adding `avr_hal_generic` `PinOps` was indeed reachable, thanks for the clarification.