Rahix/avr-hal #245: This project has a fundamental problem

MarkuBu, 2022-02-06:
In my opinion, this project has a fundamental problem. It is called "avr-hal" but in reality it is an Arduino hal. The focus of this project is too much on Arduino instead of the AVR controller itself. The embedded-hal crate starts at the board level and not the controller level as expected, unless I missed something.

I strongly recommend separating Arduino boards from AVR controllers and implementing the embedded-hal for the controllers, not the boards. 

The boards should get their own HAL depending on their additional hardware. Also, the avr-hal and the Arduino boards should be separated. 

After all, the AVRs are also used on other boards, not just Arduino boards.

stappersg, 2022-02-06:
On Sun, Feb 06, 2022 at 07:29:58AM -0800, MarkuBu wrote:
> ...
>
> I strongly recommend separating Arduino boards from AVR controllers
> and implementing the embedded-hal for the controllers, not the boards.
>
> The boards should get their own HAL depending on [...]

Just do it

Rahix, 2022-02-06:
> The embedded-hal crate starts at the board level and not the controller level as expected, unless I missed something.

You indeed missed something.  Please read through the [Repository Structure](https://github.com/Rahix/avr-hal#repository-structure) section of the README.  Only `arduino-hal` is related to "Arduino" boards.  For all other boards, we have HAL crates for families of microcontrollers - `atmega-hal` and `attiny-hal` at the moment.

The idea is that you build your own "board support crate" ontop of these, similar to how `arduino-hal` is doing it for "Arduino" boards.

> The focus of this project is too much on Arduino instead of the AVR controller itself.

If you think the MCU HALs are lacking features, please send PRs.  We focus our work where the largest part of the "community" lies.  And to put it bluntly, not many people outside the "Arduino" world would choose AVR these days anymore - understandably.

MarkuBu, 2022-02-06:
Yes, there are hal's for atmega and attiny, but are they compatible to embedded-hal?

As far as I can see they don't depend on embedded-hal.

Rahix, 2022-02-06:
They do implement e-h traits.  The dependency happens through `avr-hal-generic` where all the actual trait impls live.

MarkuBu, 2022-02-06:
> They do implement e-h traits.  The dependency happens through `avr-hal-generic` where all the actual trait impls live.

I must apologise. I didn't see that it was imported in line 14 in Cargo.toml.
 
  [dependencies.embedded-hal]

I'm really really sorry.

closed 2022-02-06 by MarkuBu

stappersg, 2022-02-06:
On Sun, Feb 06, 2022 at 10:48:43AM -0800, MarkuBu wrote:
> Closed #245.

Please consider to document (better) what **was missing** for you.

Rahix, 2022-02-06:
> I must apologise. I didn't see that it was imported in line 14 in Cargo.toml.
> 
> [dependencies.embedded-hal]
> 
> I'm really really sorry.

No hard feelings :)  Next time, maybe don't title the issue "This project has a fundamental problem", though.

MarkuBu, 2022-02-10:
But I still think AVR and Arduino should be separated into two projects.

Not all Arduino boards have AVR and not all AVO are on Arduino boards.

stappersg, 2022-02-10:
On Thu, Feb 10, 2022 at 03:17:34AM -0800, MarkuBu wrote:
> But I still think AVR and Arduino should be separated into two projects.

Fine.

It is up to you to choose "I still think".  Please keep thinking,
the good thing is it doesn't interfere with those who actual do.

 
> Not all Arduino boards have AVR and not all AVR are on Arduino boards.

That observation is correct.  And completely useless.
As useless, as counter-productive, as this response.

Groeten
Geert Stappers
-- 
Silence is hard to parse