Rahix/avr-hal #246: Ravedude can't resolve port

Baptistemontan, 2022-02-07:
I have an arduino uno connected to port COM3 and ravedude can't find it, I tried setting the RAVEDUDE_PORT environnement variable to COM3 but still don't work. The arduino IDE can find it and I'm using windows.

stappersg, 2022-02-07:
I don't understand what is expected from the people behind this project.

I think it waits for another Microsoft Windows user saying something
like:

It works for me.  That is:
* MS Windows 9
* with the foo extension
* configured COM2
* avrdude version 7

Regards
a happy Linux user

P.S.

I challenge Original Poster to state
* Windows version number
* if needed / if installed   which extension (and their version number)
* which COM ports have been tried
* which COM ports the system besides "the AVR" has
* avrdude version number
* successes with avrdude with ravedude

P.P.S.

The title should change from
  Ravedude can't resolve port (Issue #246)
to
  Ravedude can't resolve port on MS Windows version ... (Issue #246)

Rahix, 2022-02-08:
Hey @Baptistemontan,

>  Ravendude don't bubble up errors from avrdude, and I have no Idea on how to debug without them.

Ouch, yeah, this is a valid point.  But please do not repurpose existing issues for reporting related problems - can you open a new issue about this specifically instead?

Note that I cannot give support for Windows systems, maybe search through the other Windows related issues here (also closed ones) to see if you can piece together what the story is.  I'll gladly take patches to make it work on MS systems as well if you find anything.

stappersg, 2022-02-08:
On Tue, Feb 08, 2022 at 02:02:44AM -0800, Baptistemontan wrote:
>   ... and I have no Idea on ...

Please share with us the things you have more then just idea about.
Like:
* Version of avrdude
* WSL2  or  WSL1 or whatever environment being used
* comport number

Groeten
Geert Stappers

P.S.
It would be nice if
  Subject: Re: [Rahix/avr-hal] Ravedude can't resolve port on MS Windows version (Issue #246)
gets
  Subject: Re: [Rahix/avr-hal] Ravedude can't resolve port on MS Windows 10 (Issue #246)

Yeah, I'm telling that the github.com email interface truncates issue titles.
-- 
Silence is hard to parse

Baptistemontan, 2022-02-08:
I don't know much about WSL 1 vs WSL 2, I just know it worked by switching beetween them.
I don't think that comport number are relevant as I tried a bunch of them and the result were the same.

I'm sorry it is my first time doing github issue, I don't know the "rules", I'have renamed this on to the first issue and will open a new one for ravedude errors

Baptistemontan, 2022-02-08:
Will looking for the avrdude version I found that I was missing "libusb0.dll", I have been hable to resolve this problem by installing WinAVR wich look like the Windows version of avrdude. Making the switch I now have errors appearing from ravedude:
```
PS C:\Users\Baptiste\Documents\code\Rust\arduino\tests> cargo run
    Finished dev [optimized + debuginfo] target(s) in 0.21s
     Running `ravedude uno -cb 57600 target\avr-atmega328p\debug\tests.elf`
       Board Arduino Uno
 Programming target\avr-atmega328p\debug\tests.elf => COM6
avrdude.exe: invalid file format 'e' in update specifier
avrdude.exe: error parsing update operation 'flash:w:target\avr-atmega328p\debug\tests.elf:e'
Error: avrdude failed

error: process didn't exit successfully: `ravedude uno -cb 57600 target\avr-atmega328p\debug\tests.elf` (exit code: 1)
```

which make me think that ravedude does transfer the error, but why does'nt it transfer them before ?

Rahix, 2022-02-08:
> I don't know much about WSL 1 vs WSL 2, I just know it worked by switching beetween them.

Yeah, others have reported this as well in the past.

>  I don't think that comport number are relevant as I tried a bunch of them and the result were the same.

Agreed, I do not think it should make a difference.  But again, I don't have a clue about windows...

> I'm sorry it is my first time doing github issue, I don't know the "rules", I'have renamed this on to the first issue and will open a new one for ravedude errors

You're not doing anything wrong, nothing to be sorry about.  I just requested a separate issue because it helps with keeping a better overview over ongoing topics.

> ```
> avrdude.exe: invalid file format 'e' in update specifier
> avrdude.exe: error parsing update operation 'flash:w:target\avr-atmega328p\debug\tests.elf:e'
> ```

Okay, in absolutely most cases this means your version of avrdude is too old.  Actually, ravedude now learned to check the avrdude version to notify people of this but I just noticed that this feature is not yet part of a ravedude release... I need to take care of that.

Baptistemontan, 2022-02-08:
I've updated to 6.3 and it went further in the process but still failed:

```
PS C:\Users\Baptiste\Documents\code\Rust\arduino\tests> cargo run 
    Finished dev [optimized + debuginfo] target(s) in 0.17s
     Running `ravedude uno -cb 57600 target\avr-atmega328p\debug\tests.elf`
 Programming target\avr-atmega328p\debug\tests.elf => COM6

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude.exe: Device signature = 0x1e950f (probably m328p)
avrdude.exe: erasing chip
avrdude.exe: reading input file "target\avr-atmega328p\debug\tests.elf"
avrdude.exe: can't handle ELF file target\avr-atmega328p\debug\tests.elf, ELF file support was not compiled in
avrdude.exe: read from file 'target\avr-atmega328p\debug\tests.elf' failed

avrdude.exe: safemode: Fuses OK (E:00, H:00, L:00)

avrdude.exe done.  Thank you.

Error: avrdude failed

error: process didn't exit successfully: `ravedude uno -cb 57600 target\avr-atmega328p\debug\tests.elf` (exit code: 1)
```

Rahix, 2022-02-08:
> ```
> ELF file support was not compiled in
> ```

Hm, that version also does not support the features we need :/ Maybe you can find another one somewhere else or compile `avrdude` yourself?  Not sure where the people in the other issues got their `avrdude` binary from...

Rahix, 2022-02-08:
> [...] but I just noticed that this feature is not yet part of a ravedude release... I need to take care of that.

Done, ravedude 0.1.4 is available now.

Baptistemontan, 2022-02-08:
I've uninstalled WinAVR and installed a windows clone of avrdude from https://github.com/mariusgreuel/avrdude, AND IT WORK

Rahix, 2022-02-08:
Great news!  And thanks for that pointer, that fork is something we can recommend to other people as well.

Baptistemontan, 2022-02-08:
Appart from avrdude I have not needed to install anything else to make it work on windows, maybe some requirement were already installed with the arduino IDE.

Baptistemontan, 2022-02-08:
I've tried with my old board and it still don't auto find it, but If I pass the port as argument it work.
To be fair even the arduino IDE don't find it, it juste say that there is something connected to the port, but don't say what board it is