Rahix/avr-hal #255: Compile avr-hal on windows 10?

MarkuBu, 2022-04-06:
What do I need to install on Windows 10 to be able to compile avr-hal?

Winavr? Current Arduino toolchain? PlatformIO? Is that even possible?

Rahix, 2022-04-06:
I have no clue... Maybe check the other issues [labeled `windows`](https://github.com/Rahix/avr-hal/issues?q=is%3Aissue+sort%3Aupdated-desc+label%3Awindows+)?

MarkuBu, 2022-04-06:
They are only related to ravedude.

stappersg, 2022-04-06:
On Wed, Apr 06, 2022 at 01:21:08AM -0700, MarkuBu wrote:
> They are only related to ravedude.

Then start.

Yes, you will encounter obstacles.
Take http://www.catb.org/~esr/faqs/smart-questions.html as detour.

Rahix, 2022-06-16:
@djdisodo, if you want you can add a short section to the README with details on installing all the dependencies needed on windows.

djdisodo, 2022-06-16:
@Rahix https://github.com/Rahix/avr-hal/wiki/Setting-up-enviromment

djdisodo, 2022-06-16:
@stappersg i fixed typo in the link right after

stappersg, 2022-06-17:
> @stappersg i fixed typo in the link right after

Ah, _environment_ vs _enviromment_

What about
> add a short section to the README with details on installing all the dependencies needed on windows.

?

Rahix, 2022-06-17:
I think we can just add a link to the wiki page in the README now, that'll be enough.

Rahix, 2022-10-08:
I think with things like #349 hitting, it's time to close this issue.

closed 2022-10-08 by Rahix