Rahix/avr-hal #265: Is there anyway to access EEPROM ?

krishnaTORQUE, 2022-05-04:
Is there anyway to access EEPROM ?

Rahix, 2022-05-04:
Not yet, this still needs to be implemented.

fvilante, 2022-09-13:
I'm sharing a workaround I coded while EEPROM is not implemented in avr-hal. It works for avr328p.

[Low level function to read/write EEPROM on avr328p 8-bits microcontroler](https://gist.github.com/fvilante/d8263588973a3cfd05074f2d3bb7efcd)

Rahix, 2022-09-13:
@fvilante, you might want to use the API from [`avr-device`](https://github.com/Rahix/avr-device) instead of those raw register accesses.

flyingyizi, 2022-09-17:
there are lots of avr eeprom c code. e.g. eeprom.c in grbl. below shows a rust rewritten sample.

```rust

mod eeprom {
    use core::arch::asm;

    /// Read byte from EEPROM.it reads one byte from a given EEPROM address.
    ///
    /// The CPU is halted for 4 clock cycles during EEPROM read.
    pub fn eeprom_get_char(address: u16) -> u8 {
        let ptr = arduino_hal::hal::pac::EEPROM::ptr();

        unsafe {
            // wait until EEPE become to zero by hardware. on other word,
            //Wait for completion of previous write.
            while (*ptr).eecr.read().eepe().bit_is_set() {}
            // set EEPROM address register
            (*ptr).eear.write(|w| w.bits(address));
            //Start EEPROM read operation
            (*ptr).eecr.write(|w| w.eere().set_bit());
        }
        // Return the byte read from EEPROM
        unsafe { (*ptr).eedr.read().bits() }
    }

    /// Write byte to EEPROM.it writes one byte to a given EEPROM address.
    ///  The differences between the existing byte and the new value is used
    ///  to select the most efficient EEPROM programming mode.
    ///
    ///  \note  The CPU is halted for 2 clock cycles during EEPROM programming.
    ///
    ///  \note  When this function returns, the new EEPROM value is not available
    ///         until the EEPROM programming time has passed. The EEPE bit in EECR
    ///         should be polled to check whether the programming is finished.
    ///
    ///  \note  The eeprom_get_char function checks the EEPE bit automatically.
    pub fn eeprom_put_char(address: u16, data: u8) -> () {
        let ptr = arduino_hal::hal::pac::EEPROM::ptr();

        unsafe {
            // do something without interrupts
            asm!("CLI");
            // wait until EEPE become to zero by hardware
            while (*ptr).eecr.read().eepe().bit_is_set() {}

            // set EEPROM address
            (*ptr).eear.write(|w| w.bits(address));

            //Start EEPROM read operation
            (*ptr).eecr.write(|w| w.eere().set_bit());
            let old_value = (*ptr).eedr.read().bits();
            let diff_mask = old_value ^ data;

            // Check if any bits are changed to '1' in the new value.
            if (diff_mask & data) != 0 {
                // Now we know that _some_ bits need to be erased to '1'.

                // // Check if any bits in the new value are '0'.
                if data != 0xff {
                    // Now we know that some bits need to be programmed to '0' also.
                    (*ptr).eedr.write(|w| w.bits(data)); // Set EEPROM data register.
                    (*ptr).eecr.write(|w| {
                        w.eempe().set_bit(); // Set Master Write Enable bit
                        w.eepm().val_0x00() // ...and Erase+Write mode.
                    });
                    (*ptr).eecr.write(|w| w.eepe().set_bit()); // Start Erase+Write operation.
                } else {
                    // Now we know that all bits should be erased.
                    (*ptr).eecr.write(|w| {
                        w.eempe().set_bit(); // Set Master Write Enable bit
                        w.eepm().val_0x01() // ...and Erase-only mode..
                    });
                    (*ptr).eecr.write(|w| w.eepe().set_bit()); // Start Erase-only operation.
                }
            }
            //
            else {
                // Now we know that _no_ bits need to be erased to '1'.

                // Check if any bits are changed from '1' in the old value.
                if diff_mask != 0 {
                    // Now we know that _some_ bits need to the programmed to '0'.
                    (*ptr).eedr.write(|w| w.bits(data)); // Set EEPROM data register.
                    (*ptr).eecr.write(|w| {
                        w.eempe().set_bit(); // Set Master Write Enable bit
                        w.eepm().val_0x02() // ...and Write-only mode..
                    });
                    (*ptr).eecr.write(|w| w.eepe().set_bit()); // Start Write-only operation.
                }
            }

            asm!("SEI");
        }
    }
}

```

DoubleHyphen, 2022-11-06:
@fvilante I was looking at the code example you pasted.

I'm a tad… apprehensive… about the way you set and clear single bits. At least one of us has completely misunderstood what's supposed to be going on, and I don't know who it is.

You define EEPE thusly: `const EEPE: u8 = 0b0000010;`. Thus, quite obviously, it is supposed to be a bit-mask; `|=`ing it with the EEPROM control register ought to set that bit, and `^=`ing it with same ought to flip it. It's because it's so obviously a bit-mask that it's defined as a binary number with only one 1 in it… right?

17 lines later…
```rs
while (read_register(EECR) & (1<<EEPE)) == 1 { };
```
you bit-shift 1 by `EEPE` positions before you binary-and it with `EECR`. And I'm like… are you _very_ sure that's what you need to do? This examines bit no. 2, which is `EEMPE`, not `EEPE`.

Further…
```rs
// normalize is to clamp the 16 bits address to a 9 bits address (1+8)
```
An ATMega328p has 1kiB of SRAM, not 512 bytes. This corresponds to 10 bits' worth of address space, in the range `0..=9`. Other microcontrollers have less, per the data-sheet… but it's kind of queer that you're taking none of that into account.

Speaking of the data-sheet: Its code examples do contain this `1<<EEPE` part, but… `EEPE` is never defined anywhere. It is therefore entirely possible that it's defined with a value of `1`, so that `1<<EEPE == 0b0000_0010.`

Rahix, 2022-11-07:
@flyingyizi, thanks for providing the code snippets.  Just FYI, instead of the raw `cli` and `sei` instructions, you should probably use the [`avr_device::interrupt::free`](https://docs.rs/avr-device/0.4.0/avr_device/interrupt/fn.free.html) helper.  It makes sure interrupts are only re-enabled if they were actually enabled previously.  This is very important for making the code generally reusable.

DoubleHyphen, 2022-11-07:
Speaking of which… is there any way to get the total size of the EEPROM from somewhere? The `EEPROM` struct only appears to contain a pointer.

Rahix, 2022-11-23:
Hm, I don't think there is... We should probably add a constant to the HAL crates alongside a proper EEPROM API...

flyingyizi, 2022-11-25:
> @Rahix ：PR [#371](https://github.com/Rahix/avr-hal/pull/371)

Rahix, 2022-12-09:
A proper EEPROM driver exists in avr-hal now.  Thanks to @flyingyizi!

closed 2022-12-09 by Rahix