Rahix/avr-hal #271: Analog input pulls high

piegamesde, 2022-05-14:
I finally managed to get my code working again on the newest master. However I now run into the following issue: converting one of the pins to analog input pulls the pin high for some reason. Even when no measurements are made yet. I might be holding it wrong though.

```rust
	let dp = arduino_hal::Peripherals::take().unwrap();
	let mut pins = arduino_hal::pins!(dp);

	// Uncomment these to pull A0 high lol
	// let mut adc = Adc::new(dp.ADC, Default::default());
	// let mut a0 = pins.a0.into_analog_input(&mut adc);
```

I'm targeting an Arduino Nano

Rahix, 2022-06-06:
Hey, can you check the resistance between Vcc and the seemingly pulled-up pin (direction is important!)?  I'm very confused how this could happen, we do not touch pin-registers at this point of the ADC initialization.

piegamesde, 2022-06-07:
After way too many hours of debugging, I decided that my Arduino was not to trust anymore. I'll get a new one and reopen if the issue still persists then.