Rahix/avr-hal #273: ravedude/avrdude cannot program trinket with bundled avrdude.conf

dalpil, 2022-05-20:
It seems like the Adafruit Trinket [requires a slightly modified avrdude.conf](https://learn.adafruit.com/introducing-trinket/programming-with-avrdude) for programming, since the built in bootloader "acts a little bit different than other off-the-shelf programmers".

After updating the ravedude-bundled avrdude.conf file with the suggested changes from the Adafruit documentation I can successfully program my Trinket with ravedude on a MacbookPro. According to adafruit these changes should not affect the programming of bare attiny85 chips, so would it be possible to include these changes in the bundled avrdude.conf?

I have published the required changes in a branch in my forked repo; https://github.com/dalpil/avr-hal/commit/24f182fbfa7bc1759f90fb8260744bc208d7d72d

When programming the Trinket from a Ubuntu machine I also had to add [the suggested configuration to udev](https://learn.adafruit.com/usbtinyisp/avrdude#for-linux-632097-7).

Without the suggested changes to avrdude.conf i keep getting output from ravedude/avrdude like this (for the sake of searchability); 

```
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e930b (probably t85)
avrdude: erasing chip

avrdude: error: usbtiny_transmit: Input/output error
avrdude: reading input file "/Users/mac/Code/avr-hal/target/avr-attiny85/release/trinket-blink.elf"
avrdude: writing flash (378 bytes):

Writing |                                                    | 0% 0.00s
avrdude: error: usbtiny_send: Input/output error (expected 64, got -5)
Writing | ########                                           | 16% 0.00savrdude: 7 retries during SPI command
avrdude: 7 retries during SPI command

...

avrdude: verifying ...
avrdude: verification error, first mismatch at byte 0x0000
         0x00 != 0x0e
avrdude: verification error; content mismatch

...
```

Last but not least; **thank you** for developing/maintaining ravedude/avr-hal/avr-device et.al. It has made developing with Rust on my AVR-boards a pleasure. 👍

Rahix, 2022-05-20:
Hey, yeah, I think we can pull in this change.  As long as other ATtiny MCUs still work I don't see a problem here.  We can always revisit this later if something does come up.

Can you create a PR from the commit you linked?

dalpil, 2022-05-21:
PR submitted.

PR #274 (2022-05-21): Use adafruit's values for attiny85 in avrdude.conf
The changes in this PR makes sure that ravedude uses a avrdude.conf that works with the Adafruit Trinket.

Closes #273 .
closed 2022-05-21 by Rahix