Rahix/avr-hal #276: Example is not compiled after adding MQTT dependency

rkudryashov, 2022-05-29:
I'm trying to add support of publishing of MQTT messages to Arduino Uno firmware. There are several crates for this, for example,

https://github.com/00imvj00/mqttrs: ```mqttrs = { version = "0.4.1", default-features = false }```
https://github.com/quartiq/minimq: ```minimq = "0.5.3"```

Both crates depend on https://github.com/japaric/heapless

I add those dependencies to https://github.com/Rahix/avr-hal/blob/main/examples/arduino-uno/Cargo.toml

The error stack:

```
D:\avr-hal\examples\arduino-uno>cargo run --bin uno-blink
   Compiling heapless v0.7.13
error[E0432]: unresolved import `core::sync::atomic::AtomicUsize`
  --> C:\Users\User\.cargo\registry\src\github.com-1ecc6299db9ec823\heapless-0.7.13\src\pool\singleton\arc.rs:88:26
   |
88 | use core::sync::atomic::{AtomicUsize, Ordering};
   |                          ^^^^^^^^^^^ no `AtomicUsize` in `sync::atomic`

error[E0432]: unresolved import `core::sync::atomic::AtomicPtr`
  --> C:\Users\User\.cargo\registry\src\github.com-1ecc6299db9ec823\heapless-0.7.13\src\pool\llsc.rs:10:26
   |
10 | use core::sync::atomic::{AtomicPtr, Ordering};
   |                          ^^^^^^^^^
   |                          |
   |                          no `AtomicPtr` in `sync::atomic`
   |                          help: a similar name exists in the module: `AtomicI8`

error[E0432]: unresolved import `core::sync::atomic::AtomicUsize`
  --> C:\Users\User\.cargo\registry\src\github.com-1ecc6299db9ec823\heapless-0.7.13\src\spsc.rs:92:26
   |
92 | use core::sync::atomic::{AtomicUsize, Ordering};
   |                          ^^^^^^^^^^^ no `AtomicUsize` in `sync::atomic`

error[E0392]: parameter `T` is never used
   --> C:\Users\User\.cargo\registry\src\github.com-1ecc6299db9ec823\heapless-0.7.13\src\pool\mod.rs:263:17
    |
263 | pub struct Pool<T> {
    |                 ^ unused parameter
    |
    = help: consider removing `T`, referring to it in a field, or using a marker such as `PhantomData`
    = help: if you intended `T` to be a const parameter, use `const T: usize` instead

Some errors have detailed explanations: E0392, E0432.
For more information about an error, try `rustc --explain E0392`.
error: could not compile `heapless` due to 4 previous errors
```

Do you have any thoughts on this? Is there any workable config or workaround?

Rahix, 2022-06-04:
The `heapless` crate has a `cas` feature which is enabled by default.  Maybe disabling that with

```toml
[dependencies.heapless]
version = "..."
default-features = false
```

is enough to get it working.  But you might need to do this in the mqtt crate.

If the crates requires atomics from heapless, this won't work of course.

There is a hack but it is very error prone.  In your target spec file (e.g. `avr-atmega328p.json`) you can set

```json
"max-atomic-width": 16
```

Be warned: There is a high chance that this produces corrupt binaries.

The proper fix looks to be updating the `atomic-polyfill` dependeny of heapless.  From what I can tell, the latest version of that has proper avr support and heapless is just lagging behind...

rkudryashov, 2022-06-05:
Thanks! I've tried to bump `atomic-polyfill` version and disable default features locally. After that just one compilation error remained:

```
D:\avr-hal\examples\arduino-uno>cargo run --bin uno-blink
   Compiling heapless v0.7.13 (D:\heapless)
error[E0432]: unresolved import `core::sync::atomic::AtomicUsize`
  --> D:\heapless\src\spsc.rs:92:26
   |
92 | use core::sync::atomic::{AtomicUsize, Ordering};
   |                          ^^^^^^^^^^^ no `AtomicUsize` in `sync::atomic`

For more information about this error, try `rustc --explain E0432`.
error: could not compile `heapless` due to previous error
```

Biut so far I didn't manage to make the example working.

Anyway, it seems the error has nothing to do with `avr-hal` or `mqttrs`.

closed 2022-06-05 by rkudryashov

Rahix, 2022-06-06:
Agreed, I think this needs to be adressed in heapless.