Rahix/avr-hal #277: error: ran out of registers during register allocation

djdisodo, 2022-05-31:
```
PC@DESKTOP-S799JDR  ~/IdeaProjects/untitled5
$ cat .cargo/config.toml 
[build]                                                 
target = "avr-atmega328p.json"                          
                                                        
[target.'cfg(target_arch = "avr")']                     
runner = "ravedude uno -cb 57600"                       
                                                        
[unstable]                                              
build-std = ["core"]                                    
build-std-features = ["compiler-builtins-mangled-names"]

PC@DESKTOP-S799JDR  ~/IdeaProjects/untitled5
$ cat Cargo.toml                            
[package]                                                                                       
name = "untitled5"                                                                              
version = "0.1.0"                                                                               
edition = "2021"                                                                                
                                                                                                
# See more keys and their definitions at https://doc.rust-lang.org/cargo/reference/manifest.html
                                                                                                
[dependencies]                                                                                  
avr-device = "0.3"                                                                              
arduino-hal = { git = "https://github.com/Rahix/avr-hal" }                                      
PC@DESKTOP-S799JDR  ~/IdeaProjects/untitled5
$ cat avr-atmega328p.json                   
{                                                                        
  "arch": "avr",                                                         
  "atomic-cas": false,                                                   
  "cpu": "atmega328p",                                                   
  "data-layout": "e-P1-p:16:8-i8:8-i16:8-i32:8-i64:8-f32:8-f64:8-n8-a:8",
  "eh-frame-header": false,                                              
  "exe-suffix": ".elf",                                                  
  "executables": true,                                                   
  "late-link-args": {                                                    
    "gcc": [                                                             
      "-lgcc"                                                            
    ]                                                                    
  },
  "linker": "avr-gcc",
  "linker-is-gnu": true,
  "llvm-target": "avr-unknown-unknown",
  "max-atomic-width": 8,
  "no-default-libraries": false,
  "pre-link-args": {
    "gcc": [
      "-mmcu=atmega328p",
      "-Wl,--as-needed"
    ]
  },
  "target-c-int-width": "16",
  "target-pointer-width": "16"
}

PC@DESKTOP-S799JDR  ~/IdeaProjects/untitled5
$ cat src/main.rs 
/*!
 * Blink the builtin LED - the "Hello World" of embedded programming.
 */
#![no_main]
#![no_std]

use panic_halt as _;

#[arduino_hal::entry]
fn main() -> ! {
    let dp = arduino_hal::Peripherals::take().unwrap();
    let pins = arduino_hal::pins!(dp);

    // Digital pin 13 is also connected to an onboard LED marked "L"
    let mut led = pins.d13.into_output();
    led.set_high();

    loop {
        led.toggle();
        arduino_hal::delay_ms(100);
        led.toggle();
        arduino_hal::delay_ms(100);
        led.toggle();
        arduino_hal::delay_ms(100);
        led.toggle();
        arduino_hal::delay_ms(800);
    }
}

PC@DESKTOP-S799JDR  ~/IdeaProjects/untitled5
$ rustc --version
rustc 1.63.0-nightly (c35035cef 2022-05-30)

PC@DESKTOP-S799JDR  ~/IdeaProjects/untitled5
$ avr-gcc -v
Using built-in specs.
Reading specs from c:/msys64/mingw64/bin/../lib/gcc/avr/8.4.0/device-specs/specs-avr2
COLLECT_GCC=C:\msys64\mingw64\bin\avr-gcc.exe
COLLECT_LTO_WRAPPER=c:/msys64/mingw64/bin/../libexec/gcc/avr/8.4.0/lto-wrapper.exe
Target: avr
Configured with: ../configure --build=x86_64-w64-mingw32 --prefix=/mingw64 --target=avr --enable-languages=c,c++ --disable-nls --disable-libssp --disable-shared --disable-threads --disable-libgomp --disable-libada --with-dwarf2 --en
able-mingw-wildcard
Thread model: single
gcc version 8.4.0 (GCC)

PC@DESKTOP-S799JDR  ~/IdeaProjects/untitled5
$ cargo build
   Compiling core v0.0.0 (C:\Users\PC\.rustup\toolchains\nightly-x86_64-pc-windows-gnu\lib\rustlib\src\rust\library\core)
   Compiling rustc-std-workspace-core v1.99.0 (C:\Users\PC\.rustup\toolchains\nightly-x86_64-pc-windows-gnu\lib\rustlib\src\rust\library\rustc-std-workspace-core)
   Compiling compiler_builtins v0.1.71
error: ran out of registers during register allocation

error: could not compile `core` due to previous error

Caused by:
  process didn't exit successfully: `rustc --crate-name core --edition=2021 C:\Users\PC\.rustup\toolchains\nightly-x86_64-pc-windows-gnu\lib\rustlib\src\rust\library\core\src\lib.rs --error-format=json --json=diagnostic-rendered-ans
i,artifacts,future-incompat --crate-type lib --emit=dep-info,metadata,link -C embed-bitcode=no -C debuginfo=2 -C metadata=3426b09ae04b4ecb -C extra-filename=-3426b09ae04b4ecb --out-dir C:\Users\PC\IdeaProjects\untitled5\target\avr-a
tmega328p\debug\deps --target \\?\C:\Users\PC\IdeaProjects\untitled5\avr-atmega328p.json -Z force-unstable-if-unmarked -L dependency=C:\Users\PC\IdeaProjects\untitled5\target\avr-atmega328p\debug\deps -L dependency=C:\Users\PC\IdeaP
rojects\untitled5\target\debug\deps --cap-lints allow` (exit code: 0xc0000005, STATUS_ACCESS_VIOLATION)
warning: build failed, waiting for other jobs to finish...
```
rust from rustup
avr-gcc from msys2 repo

djdisodo, 2022-05-31:
seems work when i put --release flag

Rahix, 2022-06-04:
This is technically a problem in LLVM but as you noted, the best solution is to build with optimizations.  Debug builds are usually too big to fit into the AVR's flash memory anyway...

I am going to close the issue but if this pops up again with a different profile, we can reopen it.

closed 2022-06-16 by Rahix