Rahix/avr-hal #282: Unable to use avr-hal-generic as a dependency

tjf801, 2022-06-24:
I am trying to use `avr_hal_generic` as a dependency, due to wanting to implement a struct like this:
```rust
struct SevenSegmentDisplay<D, S, C>
 where D: avr_hal_generic::port::PinOps,
       S: avr_hal_generic::port::PinOps,
       C: avr_hal_generic::port::PinOps 
{
    serial_data: Pin<mode::Output, D>,
    shift_register_clock: Pin<mode::Output, S>,
    storage_register_clock: Pin<mode::Output, C>,
}
```
and I've tried the solution given in #98 to no avail, as it seems like there are actually two implementations of `PinOps`, one from each different dependency. Here's the full compiler output and Cargo.toml if they help at all.

```
error[E0599]: the method `set_low` exists for struct `avr_hal_generic::port::Pin<arduino_hal::port::mode::Output, C>`, but its trait bounds were not satisfied
   --> src\main.rs:138:37
    |
138 |         self.storage_register_clock.set_low();
    |                                     ^^^^^^^ method cannot be called on `avr_hal_generic::port::Pin<arduino_hal::port::mode::Output, C>` due to unsatisfied trait bounds
    |
    = note: the following trait bounds were not satisfied:
            `C: avr_hal_generic::port::PinOps`

error[E0277]: the trait bound `Dynamic: PinOps` is not satisfied
   --> src\main.rs:179:35
    |
179 |                     let display = SevenSegmentDisplay {
    |                                   ^^^^^^^^^^^^^^^^^^^ the trait `PinOps` is not implemented for `Dynamic`
    |
note: required by a bound in `SevenSegmentDisplay`
   --> src\main.rs:101:11
    |
100 | struct SevenSegmentDisplay<D, S, C>
    |        ------------------- required by a bound in this
101 |  where D: avr_hal_generic::port::PinOps,
    |           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ required by this bound in `SevenSegmentDisplay`

error[E0599]: the method `display_hex_digit` exists for struct `SevenSegmentDisplay<Dynamic, Dynamic, Dynamic>`, but its trait bounds were not satisfied
   --> src\main.rs:185:29
    |
100 | / struct SevenSegmentDisplay<D, S, C>
101 | |  where D: avr_hal_generic::port::PinOps,
102 | |        S: avr_hal_generic::port::PinOps,
103 | |        C: avr_hal_generic::port::PinOps 
...   |
107 | |     storage_register_clock: Pin<mode::Output, C>,
108 | | }
    | |_- method `display_hex_digit` not found for this
...
185 |                       display.display_hex_digit(x);
    |                               ^^^^^^^^^^^^^^^^^ method cannot be called on `SevenSegmentDisplay<Dynamic, Dynamic, Dynamic>` due to unsatisfied trait bounds
    |
   ::: C:\Users\REDACTED\.cargo\git\checkouts\avr-hal-f3855d5807fdfd57\d0d2f24\mcu\atmega-hal\src\port.rs:2:1
    |
2   | / avr_hal_generic::impl_port_traditional! {
3   | |     enum Ports {
4   | |         PORTB: (crate::pac::PORTB, portb, pinb, ddrb),
5   | |         PORTC: (crate::pac::PORTC, portc, pinc, ddrc),
...   |
33  | |     }
34  | | }
    | |_- doesn't satisfy `Dynamic: PinOps`
    |
    = note: the following trait bounds were not satisfied:
            `Dynamic: PinOps`

Some errors have detailed explanations: E0277, E0599.
For more information about an error, try `rustc --explain E0277`.
error: could not compile `arduino-uno-rust` due to 3 previous errors
```

Cargo.toml:
```toml
[package]
name = "arduino-uno-rust"
version = "0.1.0"
authors = ["tjf801"]
edition = "2021"
license = "MIT OR Apache-2.0"

[[bin]]
name = "arduino-uno-rust"
test = false
bench = false

[dependencies]
panic-halt = "0.2.0"
ufmt = "0.1.0"
nb = "0.1.2"
embedded-hal = "0.2.3"
avr-device = "0.3.3"

[dependencies.arduino-hal]
git = "https://github.com/rahix/avr-hal"
rev = "d0d2f243bd3e10b49f6a78d44839a6caa8be7d43"
features = ["arduino-uno"]

[dependencies.avr-hal-generic]
git = "https://github.com/Rahix/avr-hal"

# Configure the build for minimal size - AVRs have very little program memory
[profile.dev]
panic = "abort"
lto = true
opt-level = "s"

[profile.release]
panic = "abort"
codegen-units = 1
debug = true
lto = true
opt-level = "s"
```

tjf801, 2022-06-24:
Just tried that and same problem - I don't know why I didn't have that before to be honest

stappersg, 2022-06-24:
Publish a `git clone` URL of your **whole** project to find out how many _"same problem"_ or _"works for me"_ you get.

tjf801, 2022-06-24:
alright [heres](https://github.com/tjf801/arduino-test-project) a stripped down version of the project with just the problematic part where I still get the same error

tjf801, 2022-06-25:
okay... i just set up that project and on compile it seems like the error has mysteriously disappeared and copy/pasting the cargo.toml and code back into the original fixed the problem. that's embarrassing

closed 2022-06-25 by tjf801

stappersg, 2022-06-25:
Thanks for reporting the _"works for me"_.