Rahix/avr-hal #284: toggle() on dynamic output pins is broken

blueberry-fan, 2022-06-26:
For dynamic pins out_toggle() is implemented as
```
w.bits(r.bits() | self.mask)
```
This gets compiled to something like this
```
in	r24, 0x19	; 25
lds	r25, 0x0061	; 0x800061 <_ZN8hal_test14INTERRUPT_LEDS17ha62fa89a5a1d442cE+0x1>
or	r25, r24
out	0x19, r25	; 25
```
This will not only toggle the intended pin but also all pins that happen to read as '1'. 

For non-dynamic pins the implementation is similar
```
w.bits(r.bits() | (1 << $pin_num))
```
But the compiler is clever enough to translate this to the intended
```
sbi	0x19, 0	; 25
```

Rahix, 2022-06-26:
> `w.bits(r.bits() | self.mask)`

Hmm, this seems totally wrong to me... Shouldn't we be writing just `self.mask` into the `PIN` register?  I.e. this:

```rust
$port_pin_reg.write(|w| w.bits(self.mask))
```

And the same goes for the non-dynamic implementation.  It should not work there either... It should be

```rust
$port_pin_reg.write(|w| w.bits(1 << $pin_num))
```

It is actually interesting that it works there:  The compiler assumes that `sbi` and writing the entire register with a single bit changed are the same which they are actually not...  I hope this never leads to problems in correct register accessing code...

blueberry-fan, 2022-06-27:
I have made some experiments. _Writing_ vs _read-modify-writing_ and _constants_ vs _known values_ (to the compiler) vs _unknown values_. (the compiler/optimizer is really clever).

Bottom line seems to be: whenever the compiler knows the operation will affect only one bit (and inside the limited address range of `sbi`) the `in`-modify-`out` is replaced by `sbi`. IIRC avr-gcc has a similar strategy so this _can't_ be wrong, can it? ;-)

The current implementation for dynamic pins would probably work - i.e. get compiled to `sbi` - if the compiler knew the mask has only one bit set. But unless we can prove that to the compiler your suggestion is probably the best we can do.

As for the non-dynamic case, I'm actually quite happy with the current code since `sbi` is slightly more efficient than `ldi` plus `out`.
 
```
#[inline(never)]
unsafe fn test_w_1() {
    let dp = hal::Peripherals::steal();
    dp.PORTA.pina.write(|w| w.bits(0x02));
    // ldi	r24, 0x02
    // out	0x19, r24
}

#[inline(never)]
unsafe fn test_w_2(v: u8) {
    let dp = hal::Peripherals::steal();
    dp.PORTA.pina.write(|w| w.bits(v));
    // This is called with values 4 and 5. Generic code for any value v.
    // out	0x19, r24
}

#[inline(never)]
unsafe fn test_w_2a(v: u8) {
    let dp = hal::Peripherals::steal();
    dp.PORTA.pina.write(|w| w.bits(v));
    // Identical to test_w_2 but called only with value 4.
    // ldi	r24, 0x04
    // out	0x19, r24
}

#[inline(never)]
unsafe fn test_w_3(v: u8) {
    let dp = hal::Peripherals::steal();
    dp.PORTA.pina.write(|w| w.bits(1 << v));
    // This is called with values 3 and 4. Generic code for any value v.
    // Wouldn't shifting be more efficient?
    // ldi	r25, 0x01
    // andi	r24, 0x07
    // dec	r24
    // brmi	.+4
    // add	r25, r25
    // rjmp	.-8
    // out	0x19, r25
}

#[inline(never)]
unsafe fn test_w_3a(v: u8) {
    let dp = hal::Peripherals::steal();
    dp.PORTA.pina.write(|w| w.bits(1 << v));
    // Identical to test_w_3 but called only with value 4.
    // ldi	r24, 0x10
    // out	0x19, r24
}

#[inline(never)]
unsafe fn test_rw_1() {
    let dp = hal::Peripherals::steal();
    dp.PORTA.pina.modify(|r, w| w.bits(r.bits() | 0x02));
    // sbi	0x19, 1
}

#[inline(never)]
unsafe fn test_rw_2(v: u8) {
    let dp = hal::Peripherals::steal();
    dp.PORTA.pina.modify(|r, w| w.bits(r.bits() | v));
    // This is called with values 4 and 5. Generic code for any value v
    // in	r25, 0x19
    // or	r25, r24
    // out	0x19, r25
}

#[inline(never)]
unsafe fn test_rw_2a(v: u8) {
    let dp = hal::Peripherals::steal();
    dp.PORTA.pina.modify(|r, w| w.bits(r.bits() | v));
    // Identical to test_rw_2 but called only with value 4.
    // sbi	0x19, 2
}

#[inline(never)]
unsafe fn test_rw_2b(v: u8) {
    let dp = hal::Peripherals::steal();
    dp.PORTA.pina.modify(|r, w| w.bits(r.bits() | v));
    // Identical to test_rw_2 but called only with value 5.
    // in	r24, 0x19
    // ori	r24, 0x05
    // out	0x19, r24
}

#[inline(never)]
unsafe fn test_rw_3(v: u8) {
    let dp = hal::Peripherals::steal();
    dp.PORTA.pina.modify(|r, w| w.bits(r.bits() | (1 << v)));
    // This is called with values 3 and 4. Generic code for any value v.
    // ldi	r25, 0x01
    // andi	r24, 0x07
    // dec	r24
    // brmi	.+4
    // add	r25, r25
    // rjmp	.-8
    // in	r24, 0x19
    // or	r24, r25
    // out	0x19, r24
}

#[inline(never)]
unsafe fn test_rw_3a(v: u8) {
    let dp = hal::Peripherals::steal();
    dp.PORTA.pina.modify(|r, w| w.bits(r.bits() | (1 << v)));
    // Identical to test_rw_2 but called only with value 4.
    // sbi	0x19, 4
}
```

Rahix, 2022-06-27:
> As for the non-dynamic case, I'm actually quite happy with the current code since sbi is slightly more efficient than ldi plus out.

I agree that it would be nice to keep the `sbi` here but I am not sure if we can trust the compiler enough on this... Relying on an optimization for correctness is a nasty thing to do.

I think we are better off switching to the slightly less efficient but always correct approach.  If there really is a need to have the `sbi` here I think the only right way to do it is using inline assembly.

PR #286 (2022-06-28): generic: port: Fix pin toggle implementation
Only the bit for the pin to be toggled should be set in the PIN register.  The RMW-operation was incorrect here as it would also toggle unrelated pins which happen to read HIGH at that moment.

This went unnoticed because the code for the non-dynamic pin-variant was optimized down to an `sbi` instruction - thus "accidentally" behaving correctly.

I do not think it is a good idea to rely on this behavior though.  If the non-dynamic code should be optimized to an `sbi` again at some point in the future, I would prefer an implementation using inline as [... cut]
closed 2022-06-28 by Rahix