Rahix/avr-hal #287: Unable to trigger the TIMER0_COMPA interrupt on an attiny85

johnnynotsolucky, 2022-07-05:
I have an attiny85 with a RGB LED's red leg connected to PB4, and I'm trying to toggle it using a timer interrupt. I can get `TIMER1_COMPA` to work correctly using Arduino ide and rust, but for trying with `TIMER0_COMPA`, I can only get it working in Arduino. I _think_ I've ported the code to rust exactly how I have it in arduino, but I cannot work out why I can't get TIMER0_COMPA to trigger.

The working Arduino code:

```c
const byte red_led = 4;

volatile byte counter = 0;
volatile bool toggle = false;

void useTimer0() {
  //  TCCR0A = 0b10000010;
  TCCR0A = (1 << WGM01) | (1 << COM0A1); // CTC mode
  //  TCCR0B = 0b00000101;
  TCCR0B = (1 << CS02) | (1 << CS00); // 1024 prescaler
  OCR0A = 250;
  //  TIMSK = 0b00010000;
  TIMSK = (1 << OCIE0A);
}

void useTimer1() {
  //  TCCR1 = 0b10011100;
  TCCR1 = (1 << CTC1) | (1 << COM1A0) | (12 << CS10); // CTC, toggle, 2048 prescaler
  //  GTCCR = 0b00010000;
  GTCCR = (1 << COM1B0);
  //  TIMSK = 0b00000100;
  TIMSK = (1 << TOIE1);
  OCR1C = 255;
}

void setup() {
  pinMode(red_led, OUTPUT);
  //  useTimer1();
  useTimer0();
  sei();
}

void count() {
  counter = counter + 1;
  if (counter > 30) {
    toggle = !toggle;
    counter = 0;
  }
}

ISR(TIMER0_COMPA_vect)
{
  count();
}

ISR(TIMER1_COMPA_vect) {
  count();
}

void loop() {
  digitalWrite(red_led, toggle);
}
```

The code I've ported to rust:

```rust
use core::{cell::Cell, sync::atomic::AtomicBool};

use arduino_hal::pac::{TC0, TC1};
use avr_device::interrupt::Mutex;
use core::sync::atomic::Ordering;
use panic_halt as _;

static COUNTER: Mutex<Cell<u8>> = Mutex::new(Cell::new(0));
static TOGGLE: AtomicBool = AtomicBool::new(false);

fn count() {
    avr_device::interrupt::free(|cs| {
        let counter = COUNTER.borrow(cs);
        let value = counter.get();
        if value > 30 {
            counter.set(0);
            TOGGLE.store(!TOGGLE.load(Ordering::SeqCst), Ordering::SeqCst);
        } else {
            counter.set(value + 1);
        }
    });
}
#[avr_device::interrupt(attiny85)]
fn TIMER0_COMPA() {
    count();
}

/// Use TIMER0_COMPA to toggle an LED on/off
/// Does not work.
fn timer0(tc0: &TC0) {
    tc0.tccr0a.write(|w| unsafe { w.bits(0b10000010) });
    tc0.tccr0b.write(|w| unsafe { w.bits(0b00000101) });
    tc0.ocr0a.write(|w| unsafe { w.bits(250) });
    tc0.timsk.write(|w| unsafe { w.bits(0b00010000) });
}

#[avr_device::interrupt(attiny85)]
fn TIMER1_COMPA() {
    count();
}

/// Use TIMER1_COMPA to toggle an LED on/off
/// This works exactly as it does with the c code used in arduino ide.
fn timer1(tc1: &TC1) {
    tc1.tccr1.write(|w| unsafe { w.bits(0b10011100) });
    tc1.gtccr.write(|w| unsafe { w.bits(0b00010000) });
    tc1.timsk.write(|w| unsafe { w.bits(0b00000100) });
    tc1.ocr1c.write(|w| unsafe { w.bits(255) });
}

#[arduino_hal::entry]
fn main() -> ! {
    let dp = arduino_hal::Peripherals::take().unwrap();
    let pins = arduino_hal::pins!(dp);

    let mut red = pins.d4.into_output_high();

    let tc0 = dp.TC0;
    // timer0(&tc0); // Doesn't appear to work
    let tc1 = dp.TC1;
    timer1(&tc1); // Works correctly

    unsafe { avr_device::interrupt::enable() };

    loop {
        let is_on = avr_device::interrupt::free(|_cs| TOGGLE.load(Ordering::SeqCst));

        // I'm using an RGB LED with a common anode, so set_low() turns the light on.
        if is_on {
            red.set_low();
        } else {
            red.set_high();
        }
    }
}
```

I've used the binary values for setting the registers so I can match them against the datasheet easily, and have tested those exact values in the arduino code too.

I'm pretty sure I'm missing something crucial, but I just cannot see what. Any pointers would be hugely appreciated!

P.S. The actual frequency the LED is toggled is irrelevant, as long as the light toggles

P.P.S This is an amazing library, thank you!

johnnynotsolucky, 2022-07-06:
Quite embarrassingly, I had the wrong avr-spec configured in config.toml. 

I had duplicated my project from another arduino nano project, and seemingly changed everything _but_ the build target.

closed 2022-07-06 by johnnynotsolucky

johnnynotsolucky, 2022-07-08:
Thank you for pointing this out @Rahix! I had assumed that mutating anything from interrupt handler needed to be inside an explicit critical section. This makes perfect sense, and probably should've been quite obvious.

Rahix, 2022-07-11:
I would add the critical section only when it is needed.  After all, you can't safely get to the contents of a mutex without the `cs` token so as long as you're not using `unsafe` code, you only need the critical section where mutex access forces you to use one.

johnnynotsolucky, 2022-07-11:
Perfect. Thank you for all your feedback ❤️