Rahix/avr-hal #292: Unit Testing

roboteng, 2022-07-27:
I'm not sure where to post this, but I think it could be helpful here.  I am new to rust, so I am not 100% proficient with it, but I have not found a way to do unit testing.  I'd totally be willing to write up some documentation once I get it working. I haven't found any references to testing within this repo, nor within the wider rust community.

Any help would be appreciated, and I'd love to make a PR for this repo or the template one if I can figure it out.

Patryk27, 2022-07-31:
I can do a bit of self-advertisement here: https://github.com/Patryk27/avr-tester/ 🙂 

It's not as straightforward as `cargo test`, but it's very close!

djdisodo, 2022-08-05:
you can use qemu, it works
https://github.com/Rahix/avr-hal/discussions/280
it works without no real hardware, yet will have timer issue
https://github.com/Rahix/avr-hal/pull/257