Rahix/avr-hal #293: Can't instantiate `attiny-hal::Adc` for attiny85 application.

tbezemer, 2022-08-03:
Using the examples in the documentation, I am trying to instantiate the ADC, so I can use it to read sensor values in my attiny85 application:

```rust
#[no_mangle]
pub extern "C" fn main() {

    let dp = attiny_hal::Peripherals::take().unwrap();
    let pins = attiny_hal::pins!(dp);

    let mut led = pins.pb3.into_output();
    let mut adc= attiny_hal::Adc::new(dp.ADC, Default::default());
    
    // initialise more things

    loop {
         // do things.
    }
}
```

But when building this code, the compiler then complains because it wants me to be explicit about the types I am using. I don't really understand why this is necessary (as it was not necessary in [the examples in the docs](https://rahix.github.io/avr-hal/avr_hal_generic/adc/struct.Adc.html)):

```
error[E0282]: type annotations needed for `avr_hal_generic::adc::Adc<Attiny, attiny_hal::pac::ADC, CLOCK>`
  --> src/main.rs:16:9
   |
16 |     let mut adc= attiny_hal::Adc::new(dp.ADC, Default::default());
   |         ^^^^^^^
   |
help: consider giving `adc` an explicit type, where the type for type parameter `CLOCK` is specified
   |
16 |     let mut adc: avr_hal_generic::adc::Adc<Attiny, attiny_hal::pac::ADC, CLOCK>= attiny_hal::Adc::new(dp.ADC, Default::default());
   |                ++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
```

I have tried a number of things (none of which worked) but I felt I was not trying anything particularly informed (as I am new to embedded programming), so I was hoping someone more knowledgeable than myself could help me solve this problem. I feel like I am not doing anything differently than as is instructed in the documentation, so I am wondering if there's something I am missing.

Thanks in advance!

Rahix, 2022-08-03:
> ```rust
> #[no_mangle]
> pub extern "C" fn main() {
> ```

First of all, this doesn't look right.  You should be using the `attiny_hal::entry` macro like this:

```rust
#[attiny_hal::entry]
fn main() -> ! {
    ...
}
```

That out of the way, the problem is that the ADC needs to know about your clock-rate.  As the rate is not fixed, you have to make it explicit in your code.  The best way to do this is to define the clock-rate globally and then reference that type everywhere else.  Then you can easily change the clock-rate in the future.  It works like this:

```rust
pub type CoreClock = attiny_hal::clock::MHz1;

#[attiny_hal::entry]
fn main() -> ! {
    let mut adc = attiny_hal::Adc<CoreClock>::new(...);
}
```

Rahix, 2022-08-03:
On that note, we should really add some examples of using the bare HALs like `atmega-hal` and `attiny-hal`...

tbezemer, 2022-08-03:
Thanks a lot for your quick, and helpful response!

I ended up with:

```rust
let mut adc: attiny_hal::Adc<CoreClock> = attiny_hal::Adc::new(dp.ADC, Default::default());
```

That compiles now! 😄

Concerning your `#[attiny_hal::entry]` tip, it does not seem that `attiny_hal` namespace contains this macro. I also searched the entire `attiny_hal` crate for it, to see if it might be nested in another object, but I don't see it. I only see a similar one in the `arduino_hal` and `avr_device_macros` crates.

When I use the standard rust embedded programming approach (which is what is in my example above) it works perfectly fine; it builds, flashes and runs live on my attiny85 chip without issues. Does this macro do anything specific for AVR that makes it necessary?

About the examples: I would be willing to contribute examples for attiny-hal as my project progresses. I will also be doing similar things with an atmega chip once I get my hands on a few. Perhaps we can discuss a set of examples I could work on for the repo?

tbezemer, 2022-08-03:
For reference, I edited my initial code snippet to include `#[no_std]` and `#[no_main]` which are required for standard embedded programming in Rust.

Rahix, 2022-08-03:
Do you have the `rt` feature enabled for `attiny-hal`?  Without it, the macro won't be included.

You are right that you don't technically need it right now but that only works because we rely on the avr-libc runtime for the time being.  As soon as that changes, the macro will be required.  So to make sure your code continues to work, it should use the `attiny_hal::entry` macro even now.

tbezemer, 2022-08-04:
Thanks! I didn't know about the `rt` feature. It works now. Good to know the difference between the two, and that it's important to make my code future proof.

Rahix, 2022-08-04:
About

> About the examples: I would be willing to contribute examples for attiny-hal as my project progresses. I will also be doing similar things with an atmega chip once I get my hands on a few. Perhaps we can discuss a set of examples I could work on for the repo?

That's great to hear!  My suggestion would be to create a `examples/bare-attiny85/` crate with various examples using the ATtiny85 directly, without e.g. `arduino-hal`.  Starting simple with things like a blink example, maybe UART, an ADC example, etc.  I think the most important part is documenting how one would get those examples running, so the commands to flash it, maybe how to wire it up, etc.