Rahix/avr-hal #299: atmega328p new bootloader

Maik93, 2022-08-17:
Hi, I'd like to work with some Arduino Nano that seems to have a "new" bootloader: on previous projects, working in C++ with PlatformIO, the actual board that I have to select is [nanoatmega328new](https://docs.platformio.org/en/stable/boards/atmelavr/nanoatmega328new.html), since if I try with the usual nanoatmega328 the flashing process gets stuck.

Now here in Rust it seems to be the same problem: if I try to flash the example project (obtained with avr-hal-template by selecting Arduino Nano), `cargo run` gets stuck at `avrdude: stk500_recv(): programmer is not responding`.

I wasn't able to understand the differences between the two PlatformIO configurations, except for the upload speeds: the "new" one is set to 115200.
If they can be useful, here's [nanoatmega328new](https://github.com/platformio/platform-atmelavr/blob/master/boards/nanoatmega328new.json) and [nanoatmega328](https://github.com/platformio/platform-atmelavr/blob/master/boards/nanoatmega328.json) json configuration files used by PlatformIO.

Full terminal output:
```bash
$ cargo run                   
    Finished dev [optimized + debuginfo] target(s) in 0.04s
     Running `ravedude nano -cb 115200 target/avr-atmega328p/debug/nano-base.elf`
       Board Arduino Nano
Error: no matching serial port found, use -P or set RAVEDUDE_PORT in your environment

Caused by: Not able to guess port
$ cargo run -- -P /dev/ttyUSB0
    Finished dev [optimized + debuginfo] target(s) in 0.04s
     Running `ravedude nano -cb 115200 target/avr-atmega328p/debug/nano-base.elf -P /dev/ttyUSB0`
       Board Arduino Nano
 Programming target/avr-atmega328p/debug/nano-base.elf => /dev/ttyUSB0
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x00
```
I'm sure that the target is /dev/ttyUSB0 (it's working in PlatformIO projects) and I've tested both 115200 and 57600 baudrates.

Maik93, 2022-08-18:
Seems fixed in PR #247, just specifying `runner = "ravedude nano-new -cb 115200"` in `.cargo/config.toml`.

Rahix, 2022-08-19:
#247 is merged now so I think we can close this.  If not, let me know, so I will re-open it.

closed 2022-08-19 by Rahix

michaelrommel, 2023-10-18:
Hello,

I think that still there is something missing: the baudrate for ravedude on the commandline is only honoured in the code path dealing with the console opened if the -c flag is given. It is NOT passed on to the avrdude programm command line. Hence there is NO baudrate specified to avrdude and that is why it is failing. It happened to me today as well with a Arduino Pro mini clone that wants to have a -b57600 on the avrdude command line.

I think it would make sense to pass on the Args.baudrate parameter on to the constructor of the avrdude command line and honour it there, if provided.

Thanks for your consideration! 

  Michael.

michaelrommel, 2023-10-18:
ravedude:

```
> /home/rommel/software/rust/keyboard/avr-hal/ravedude/target/release/ravedude -b57600 -P/dev/ttyUSB0 uno /home/rommel/software/rust/keyboard/avr-hal/target/avr-atmega328p/debug/uno-blink.elf
       Board Arduino Uno
 Programming /home/rommel/software/rust/keyboard/avr-hal/target/avr-atmega328p/debug/uno-blink.elf => /dev/ttyUSB0
"avrdude" "-C" "/tmp/.avrdude-8DxVCV.conf" "-c" "arduino" "-p" "atmega328p" "-P" "/dev/ttyUSB0" "-e" "-D" "-U" "flash:w:/home/rommel/software/rust/keyboard/avr-hal/target/avr-atmega328p/debug/uno-blink.elf:e"
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x00
```

avrdude directly with baudrate:

```
avrdude -v -C/tmp/avrdude.conf -patmega328p -carduino -b57600 -P/dev/ttyUSB0 -e -D -Uflash:w:/home/rommel/software/rust/keyboard/avr-hal/target/avr-atmega328p/debug/uno-blink.elf:e

avrdude: Version 6.3-20171130
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/tmp/avrdude.conf"
         User configuration file is "/home/rommel/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : /dev/ttyUSB0
         Using Programmer              : arduino
         Overriding Baud Rate          : 57600
         AVR Part                      : ATmega328P
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
           flash         65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
           lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : Arduino
         Description     : Arduino
         Hardware Version: 2
         Firmware Version: 1.16
         Vtarget         : 0.0 V
         Varef           : 0.0 V
         Oscillator      : Off
         SCK period      : 0.1 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude: Device signature = 0x1e950f (probably m328p)
avrdude: safemode: lfuse reads as 0
avrdude: safemode: hfuse reads as 0
avrdude: safemode: efuse reads as 0
avrdude: erasing chip
avrdude: reading input file "/home/rommel/software/rust/keyboard/avr-hal/target/avr-atmega328p/debug/uno-blink.elf"
avrdude: writing flash (436 bytes):

Writing | ################################################## | 100% 0.26s

avrdude: 436 bytes of flash written
avrdude: verifying flash memory against /home/rommel/software/rust/keyboard/avr-hal/target/avr-atmega328p/debug/uno-blink.elf:
avrdude: load data flash data from input file /home/rommel/software/rust/keyboard/avr-hal/target/avr-atmega328p/debug/uno-blink.elf:
avrdude: input file /home/rommel/software/rust/keyboard/avr-hal/target/avr-atmega328p/debug/uno-blink.elf contains 436 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 0.22s

avrdude: verifying ...
avrdude: 436 bytes of flash verified

avrdude: safemode: lfuse reads as 0
avrdude: safemode: hfuse reads as 0
avrdude: safemode: efuse reads as 0
avrdude: safemode: Fuses OK (E:00, H:00, L:00)

avrdude done.  Thank you.
```

I copied the ravedude-generated .avrdude.conf file to /tmp/avrdude,conf, so there should be no difference, only the baudrate parameter is missing.

Rahix, 2023-10-19:
Hi,

the baudrate from the commandline is only meant for the serial console. it is not passed to avrdude for flashing because your console baudrate may be completely different from the one used for loading the program. the "avrdude baudrate" is instead defined board-specific.  So for example the `uno` board uses the [ default baudrate](https://github.com/johnnynotsolucky/avr-hal/blob/7a82ba21aafbf2b6b001be618f54739a36f6b41b/ravedude/src/board.rs#L63) while the `nano` board uses [57600](https://github.com/johnnynotsolucky/avr-hal/blob/7a82ba21aafbf2b6b001be618f54739a36f6b41b/ravedude/src/board.rs#L125) and the `nano-new` board uses [115200](https://github.com/johnnynotsolucky/avr-hal/blob/7a82ba21aafbf2b6b001be618f54739a36f6b41b/ravedude/src/board.rs#L150).

So I would argue that the solution to your problem is either to add your board to ravedude with correct settings or maybe you can just use the `nano` board config which I think should be completely identical to what you need.