Rahix/avr-hal #301: Attiny85 I2C support

rhblind, 2022-08-24:
Hello,
Please forgive my ignorance, I'm pretty new to Rust and embedded programming as a whole, so this might (or might not) be a silly question.

I'm trying to program an Attiny85 using the [Trinket examples](https://github.com/Rahix/avr-hal/tree/main/examples/trinket).
However, I'd like to add I2C Slave support for communication. Looking at the source code for the Attiny chips in [attiny-hal](https://github.com/Rahix/avr-hal/tree/main/mcu/attiny-hal) directory, I see that the Atmega has a source file for I2C, while the Attiny doesn't. 

I have tried comparing with the examples in [uno-i2cdetect.rs](https://github.com/Rahix/avr-hal/blob/main/examples/arduino-uno/src/bin/uno-i2cdetect.rs), to see if I could make it work, but it won't compile.

Is it something fundamentally I don't get, or is I2C simply not supported for the Attiny family?

Best regards :)

Rahix, 2022-08-29:
Hi,

unfortunately, I2C is not _yet_ supported in `attiny-hal`.  The ATtiny chips use a different peripheral for I2C communication.  It is called `USI` (Universal Serial Interface) and works differently than the `TWI` (Two Wire Interace = I2C) peripheral on ATmega chips.  So what is needed is someone implementing a driver for it...

rhblind, 2022-09-06:
Hello, 
Right, thank you for the explanation. That makes sense :)

Rahix, 2022-09-06:
Let's keep this issue open to track progress on the driver implementation.

rkudryashov, 2022-09-22:
I'm in a pretty similar situation as the author of the issue and not sure if my issue requires a separate ticket, so I'm posting it here.

I am trying to use a bme280 environmental sensor connected to an Arduino Uno using I2C. I do it like this:

```rust
fn measure_bme() {
    let dp = Peripherals::take().unwrap();
    let pins = arduino_hal::pins!(dp);

    let mut i2c = arduino_hal::i2c::I2c::new(
        dp.TWI,
        pins.a4.into_pull_up_input(),
        pins.a5.into_pull_up_input(),
        50000,
    );

    let mut bme280= BME280::new_primary(i2c);
}
```

The compilation error looks like this:

```
error[E0277]: the trait bound `avr_hal_generic::i2c::I2c<Atmega, arduino_hal::pac::TWI, avr_hal_generic::port::Pin<Input, PC4>, avr_hal_generic::port::Pin<Input, PC5>, MHz16>: embedded_hal::i2c::blocking::I2c` is not satisfied
  --> src\main.rs:30:41
   |
30 |     let mut bme280= BME280::new_primary(i2c);
   |                     ------------------- ^^^ the trait `embedded_hal::i2c::blocking::I2c` is not implemented for `avr_hal_generic::i2c::I2c<Atmega, arduino_hal::pac::TWI, avr_hal_generic::port::Pin<Input, PC4>, avr_hal_generic::port::Pin<Input, PC5>, MHz16>`
   |                     |
   |                     required by a bound introduced by this call
   |
   = help: the trait `embedded_hal::i2c::blocking::I2c<A>` is implemented for `&mut T`
note: required by a bound in `bme280::i2c::BME280::<I2C>::new_primary`
  --> C:\Users\User\.cargo\registry\src\github.com-1ecc6299db9ec823\bme280-0.4.4\src\i2c.rs:22:10
   |
22 |     I2C: I2c + ErrorType,
   |          ^^^ required by this bound in `bme280::i2c::BME280::<I2C>::new_primary`

For more information about this error, try `rustc --explain E0277`.
warning: `embedded` (bin "embedded") generated 2 warnings
error: could not compile `embedded` due to previous error; 2 warnings emitted
```

Dependencies are:

```toml
arduino-hal = { git = "https://github.com/rahix/avr-hal", rev = "81edfcd908be8d3b2cd763cf2dc76aedbb22286b", features = ["arduino-uno"] }
embedded-hal = "0.2.7"
panic-halt = "0.2.0"
bme280 = "0.4.4"
```

I tried to implement it as it was in https://github.com/maspetsberger/rust-esp32-bme280/blob/master/src/main.rs#L30.

The library (https://github.com/VersBinarii/bme280-rs) also contains an example of using I2C.

Could you help me fix the compilation error? Is it worth to try using SPI?

Palladinium, 2023-04-30:
I've been working on implementing an I2C driver for the Attiny family, and I think that unfortunately the generic traits as defined in `avr-hal-generic` are not generic enough to work with the USI.

The problem I'm finding is that the `I2cOps` trait only gives access to implementors to the periperal through the `self` parameter, but not the pins.

For the Atmega family, this isn't an issue as the SDA pin stays in input mode and is driven separately by the TWI, but for the USI output driver to control the SDA pin the corresponding data direction register bit must be set. In other words, the pin needs to be switched between input and output mode during different steps of I2C communication. In this library, it looks like I should be using the `into_output`/`into_input` function on the pin, which aren't accessible from the `I2cOps` members such as `raw_start`.

Furthermore, this makes typing and borrowing complicated as the type of the pin structs changes throughout the i2c flow, and while there is a structure for a type-erased runtime pin identifier (`Dynamic`), there doesn't seem to be a similar structure for the pin mode. `into_output` and similar functions move the pin, which can't be done through the `I2c` methods since they take a mutable reference to `I2c`, and therefore to the pins within.

For more context, see the chapter on USI on any of the Attiny datasheets (I've been using an Attiny84), and also compare the application notes AVR310 (Using the USI module as a TWI Master) and AVR315 (Using the TWI Module as I2C Master), as well as their example C source code. Note how AVR315 basically never changes the DDRx register after initialization, while AVR310 sets the SDA bit in DDR_USI (which is just its matching DDRx register) in a few places during communication.

I believe that in order to support I2C on the Attiny family, quite likely breaking changes need to be made to the generic I2C structs in order to better support USI, at least to be able to modify SDA's mode at runtime within `I2cOps`. Alternatively, a separate type for I2C through USI could be added, though the loss of generality is undesirable. Either way, having a pin mode that type-erases the information to a runtime check rather than compile-time would be very useful too.

Palladinium, 2023-05-03:
I've also found an issue where the bit patterns for the `USIWM` bits in `USICR` are incorrectly labelled.
According to the `USIWM_A` enum, a value of 2 (`USIWM1` set, `USIWM0` unset) puts the USI in two-wire slave mode, while a value of 3 (both `USIWM1` and `USIWM0` set) puts it in two-wire master mode.

From my understanding of the datasheet, as well as application notes AVR310 and AVR312 (reference implementations of master and slave i2c respectively), the two patterns are not actually referred to as master or slave modes, however their reference master implementation uses only bit pattern 2 (the one the rust enum calls slave mode), meanwhile their slave implementation uses both bit patterns 2 and 3 during different parts of the communication.

The only difference in behavior between the two modes is that with bit pattern 3 SCL is held low after an overflow of the counter register - something normally done by the slave (the rust code calls this "master" mode).

I think calling these patterns master and slave is a bit misleading in the first place, so perhaps wholly renaming the two variants to more accurate names would be good. Perhaps `two_wire` and `two_wire_with_counter_overflow_hold`.

I'm not actually sure where the name itself comes from - the enum is defined in `avr-device`, but it seems to come from generated code, so I'm not sure where such a change would need to be done.

lopsided98, 2023-07-18:
I just wrote a quick and dirty I2C master driver using USI: https://github.com/lopsided98/acurite-thermometer/blob/main/acurite-thermometer/src/i2c.rs. It is pretty closely based on AVR310, but is missing any and all delays because I only run my ATtiny85 at 1 MHz, making it slow enough already. I also don't wait for SCL to go high, since there is no way to read the `PINx` value of an output pin with avr-hal. I solved the dynamic pin mode problem using an enum that can store either an input or an output pin, allowing me to move the pin between the two types. This probably has some overhead, but I haven't checked the assembly.

A better solution would be to implement the `embedded_hal::digital::v2::IoPin` trait, which is currently guarded behind the `unproven` feature.

I agree that the `USIWM` bits are incorrectly named. The names come from an SVD patch in `avr-device`, see: https://github.com/Rahix/avr-device/blob/main/patch/common/tiny/usi.yaml. The original descriptions from the ATDF are correct, if a little terse; I'm not sure why they needed to be changed: https://github.com/Rahix/avr-device/blob/7631be88f686fab89e008dfa39bd13559909aefb/vendor/attiny85.atdf

Palladinium, 2023-07-24:
I gave this my best shot a few weeks ago, be warned that this compiles but doesn't work: https://gist.github.com/Palladinium/77e8101efa73777fb8290198c24ebe38
It relies on a small patch to `avr-hal` to expose the raw pin, which is included in the gist. My implementation is for an Attiny84, but as far as I know it should all be the same for an Attiny85, just with the pin assignments moved.

While from my testing it works fine on its own, as soon as I start doing anything else alongside it (as little as doing IO on other pins or using PWM) the MCR starts behaving very erratically and unpredictably, including restarting itself at arbitrary times, hanging, or not running at all.

The behavior is consistent until a code change and recompile, and changes in completely unrelated ways to what was changed. For instance, one version where I pull some pins up and down a few times before and after I2C transmissions to light some LEDs for debugging will hang during I2C transmission, and after just changing the order the LEDs are lit or the number of times an LED is blinked will turn it into a restart loop at a line before the I2C is even initialized. Changing something else similarly minor will make the whole MCU do nothing at all.

I double-checked and excluded likely sources of inconsistency, like weak power rails, nearby electrical noise, faulty parts, etc. to the best of my (limited) ability. I eventually rewrote my firmware in C using an existing I2C implementation, and it all behaves correctly in the same electrical environment, with the same parts.

This all makes me think that my code is doing something deeply unsound, and unfortunately I can't spot it and I'm not fluent enough in avr assembly to figure out what's happening from looking at the disassembly. Best of luck to anyone attempting this, I might revisit it some time but I'm a bit too burned out on it from staring at my code too long at the moment.