Rahix/avr-hal #303: Example: uFmt_f32/uFmt_f64

chevdor, 2022-08-31:
I was surprised hitting bigger problems than expected when trying to format a string and writting to a serial port.
This issue is partly related to #302.

In a nutshell, it would be great to see a working example such as:
```
let value = 3.1415926_f32;
let value_write = uFmt_f32::Two(value); // 3.14
ufmt::uwriteln!(&mut serial, "value: {}", value_write);
```

The issue I hit was `the trait `uDisplay` is not implemented for `uFmt_f32` despite the fact that `uFmt_f32` does [implement it](https://docs.rs/ufmt_float/0.1.0/src/ufmt_float/lib.rs.html#47-84).

Rahix, 2022-08-31:
Can you check your dependency tree on whether `avr-hal` and `ufmt_float` both depend on the same version of `ufmt`?  Easiest way to do this is using `cargo tree`.

chevdor, 2022-08-31:
Ah indeed:
```
c tree -i ufmt
error: There are multiple `ufmt` packages in your project, and the specification `ufmt` is ambiguous.
Please re-run this command with `-p <spec>` where `<spec>` is one of the following:
  ufmt@0.1.2
  ufmt@0.2.0
```

chevdor, 2022-08-31:
NOTE: `heapless::string` (and `alloc::...` even less) is, unlike what I initiall thought, **not** required for such an example to work.

chevdor, 2022-09-07:
FYI: `ufmt_loat` was published on crates.io but there was no public repo. I contacted the author who nicely published the repo. I also lined up [a PR](https://github.com/tl8roy/ufmt_float/pull/1) so `ufmt_float` no longer depends on an ancient `ufmt`.