Rahix/avr-hal #341: Example: breadboard-hal, deadbug-hal

steelman, 2022-09-25:
Add an example showing how to use avr-hal without a board like Arduino Uno but with an MCU in a circuit on a custom pcb, a breadboard or a dead bug.

stappersg, 2022-09-25:
_Patches welcome_

stappersg, 2022-09-25:
> _Patches welcome_

and karma bonus points for explain what a _"dead bug"_ is and why it needs a Hardware Abstraction Layer.

steelman, 2022-09-25:
The term deadbug refers to the look of a circuit assembled by soldering wires and components directly to pins of an integrated circuit laying upside down. See https://hackaday.com/tag/deadbug

I suppose these don't require a hal. However,   a project targeted at deadbug circuit still needs to have some defined, that without a board hal remains undefined. I am asking for a minimal list of such stuff. 

I might push a patch as I managed to build a project of mine, but I haven't uploaded it to an MCU so my findings may be wrong.
-- 
Łukasz Stelmach z podróży

stappersg, 2022-09-26:
If I understand correctly: _At software level are breadboard and deadbug the same._

> ... might push a patch ...

Yes, take it for further.  Especially for yourself.

steelman, 2022-09-26:
> At software level are breadboard and deadbug the same.

Yes.

stappersg, 2022-09-26:
This issue has now a link to https://github.com/Rahix/avr-hal/discussions/337
_"How do I write a custom avr-hal for atmega32u4 devices?"_

steelman, 2022-09-28:
Thank you. This is somewhat helpful.