Rahix/avr-hal #342: avr-specs should probably use ISA rather than MCU model

rmsc, 2022-09-29:
I am trying to add support for the attiny202 and similar devices. It turns out avr-gcc doesn't recognize `--mmcu=attiny202`, and instead required me to use `--mmcu=avrxmega3`, which is the MCU's ISA.

Looking at the relevant avr-gcc manual, it seems to me that specifying the **ISA** is all that's needed. Allowing the user to specify the MCU model seems to be just a convenience, and was probably just forgotten in the case of the latest attiny devices.

Looking at what is actually changed between avr-specs, I believe it would be a lot easier to maintain at most 17 json spec files (the current number of ISAs), rather than one file per MCU.

The specs are only needed for the rustc target, and are currently decoupled from the device selection using `features`.

I can submit a PR with these changes, just please let me know if it makes sense.

Rahix, 2022-09-30:
Hi,

IIRC, this setting also influences which linker-script is used.  As we don't currently specify one ourselves but instead rely on `avr-gcc`/`avr-ld`'s linker scripts, this is important.  The linker script will e.g. influence how much SRAM/FLASH space the program may fill.

This is an artifact of how the `avr-device` crate currently works.  When our dependency on `avr-libc` is dropped (ref <https://github.com/Rahix/avr-device/issues/76>), I think this will also not be relevant anymore.  At that point, I think we'd also be able to do the change you propose.

rmsc, 2022-09-30:
I see what you mean. I didn't realize that there were differences between the core avrxmega3 gcc spec and the device-specific ones.

The linking problem I'm experiencing with `--mmcu=attiny202` may actually be a bug in avr-libc:
https://github.com/avrdudes/avr-libc/issues/890

closed 2022-09-30 by rmsc

rmsc, 2022-09-30:
Actually it's a different bug in avr-libc that was fixed after the current stable version was released:
https://github.com/avrdudes/avr-libc/commit/f445c5c96a11096f9aa36b475be51e20d7892492