Rahix/avr-hal #343: attiny2313 targe uses the wrong ISA

rmsc, 2022-09-30:
Not sure if by design or mistake, but the attiny2313 spec sets `--mmcu=at90s2313`:
```
$ grep tiny2313 -A 2 avr-specs/sync-from-upstream.py
    "attiny2313": {
        "cpu": "at90s2313",
    },
```

According to the avr-gcc manual, while code for the at90s2313 can probably run in the attiny2313, they actually have a slightly different ISA:
```
GCC supports the following AVR devices and ISAs:

avr2

    “Classic” devices with up to 8 KiB of program memory.
    mcu = (...) at90s2313, (...) .

avr25

    “Classic” devices with up to 8 KiB of program memory and with the MOVW instruction.
    mcu = (...) attiny2313, (...).
```

Rahix, 2022-09-30:
@urlordjames, as your PR #294 added this, maybe you have any more info on this?

urlordjames, 2022-09-30:
> @urlordjames, as your PR #294 added this, maybe you have any more info on this?

I must have skimmed the manual and missed it.  Good catch!  Thus far I haven't run into any issues in my use case, apologies if this caused a headache on your end.

stappersg, 2022-09-30:
On Fri, Sep 30, 2022 at 05:55:18AM -0700, urlordjames wrote:
> > @urlordjames, as your PR #294 added this,
> > maybe you have any more info on this?
> 
> I must have skimmed the manual and missed it.  Good catch!  Thus far
> I haven't run into any issues in my use case, apologies if this caused
> a headache on your end.

Ideas for getting beyond the headache are welcome

urlordjames, 2022-09-30:
Of course I will submit a PR that resolves this when I find the opportunity.  I don't entirely recall why I used "at90s2313" instead of "attiny2313", maybe it didn't initially work on my system or something.  I will test this on one of the ATtiny2313's I have to verify the suggested change actually works.

Rahix, 2022-09-30:
@urlordjames, perfect, this is very nice of you.  I don't have any of these MCUs on hand so if you can test on real hardware, this will be best.  That said, as `avr2` (what is currently used) is a subset of `avr25` (what the proposed change would use), we're on the safe side right now.  Thus I don't think this issue has high priority at the moment.  So no need to stress over it!

urlordjames, 2022-09-30:
Upon trying to change `at90s2313` to `attiny2313` I stumbled upon what must have made me decide to use it in the first place.  It appears that the Nixpkgs version of avr-gcc fails to find `crtattiny2313.o` and crashes when `-mmcu=attiny2313` is supplied as an argument.  The first time around I must have assumed that `attiny2313` was incorrect somehow, so I used `at90s2313` instead.  I don't think this is intended behavior and I suspect it may be a bug in Nixpkgs, so I will be opening an issue there.  Regardless, after forcibly coercing avr-gcc to find the correct files (by adding `-B/nix/store/fd99zs5zrfbk9pyyj3mqgmwq01ibfgcc-avr-libc-avr-2.0.0/avr/lib/avr25/tiny-stack` to pre-link args) I was able to confirm this works on a real ATtiny2313.  I apologize if this ended up causing incorrect behavior for anyone.

PR #344 (2022-10-01): avr-specs: ATtiny2313 -mmcu "at90s2313" -> "attiny2313"
resolves #343
commit 9eb5ceae (2022-10-01): avr-specs: ATtiny2313 -mmcu "at90s2313" -> "attiny2313"
resolves #343
closed 2022-10-01 by Rahix