Rahix/avr-hal #352: Can't compile not changed cargo generated projec6

leonidv, 2022-10-15:
Hi!

I can't compile simple example, which is auto-generated by cargo-generate. 
What I do:
```
$ cargo generate --git https://github.com/Rahix/avr-hal-template.git
....
🔧   Moving generated files into: `/home/leonidv/temp/blink`...
💡   Initializing a fresh Git repository
✨   Done! New project created /home/leonidv/temp/blink
$ cd blink
$ cargo run
[10/19]   Done: avr-specs/avr-atmega2560.json
[11/19]   Done: avr-specs/avr-atmega328p.json
[12/19]   Done: avr-specs/avr-atmega32u4.json
[13/19]   Done: avr-specs/avr-atmega48p.json
[14/19]   Done: avr-specs/avr-attiny85.json
[15/19]   Done: avr-specs/avr-attiny88.json
[16/19]   Done: avr-specs
[17/19]   Done: rust-toolchain.toml
[18/19]   Done: src/main.rs
[19/19]   Done: src
🔧   Moving generated files into: `/home/leonidv/temp/blink`...
💡   Initializing a fresh Git repository
✨   Done! New project created /home/leonidv/temp/blink
leonidv@lvlinux ~/temp> cd blink/
leonidv@lvlinux ~/t/blink (main)> cargo run
    Updating git repository `https://github.com/rahix/avr-hal`
    Updating crates.io index
   Compiling compiler_builtins v0.1.73
   Compiling core v0.0.0 (/home/leonidv/.rustup/toolchains/nightly-2022-07-10-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/cor
   Compiling proc-macro2 v1.0.46
   Compiling rustversion v1.0.9
   Compiling quote v1.0.21
   Compiling unicode-ident v1.0.5
   Compiling semver-parser v0.7.0
   Compiling syn v1.0.102
   Compiling proc-macro-hack v0.5.19
   Compiling paste v1.0.9
   Compiling semver v0.9.0
   Compiling rustc_version v0.2.3
   Compiling bare-metal v0.2.5
   Compiling avr-device v0.3.4
   Compiling avr-hal-generic v0.1.0 (https://github.com/rahix/avr-hal?rev=1aacefb335517f85d0de858231e11055d9768cdf#1aacefb3)
   Compiling ufmt-macros v0.2.0
   Compiling avr-device-macros v0.3.4
   Compiling rustc-std-workspace-core v1.99.0 (/home/leonidv/.rustup/toolchains/nightly-2022-07-10-x86_64-unknown-linux-gnu/lib/rustlib
/home/leonidv/.rustup/toolchains/nightly-2022-07-10-x86_64-unknown-linux-gnu/bin/../lib/librustc_driver-8fd17062f1c0f9c7.so(+0x544c33)[
/usr/lib/libc.so.6(+0x38a00)[0x7f12c8451a00]
/home/leonidv/.rustup/toolchains/nightly-2022-07-10-x86_64-unknown-linux-gnu/bin/../lib/../lib/libLLVM-14-rust-1.64.0-nightly.so(+0x26a
/home/leonidv/.rustup/toolchains/nightly-2022-07-10-x86_64-unknown-linux-gnu/bin/../lib/../lib/libLLVM-14-rust-1.64.0-nightly.so(+0x26a
/home/leonidv/.rustup/toolchains/nightly-2022-07-10-x86_64-unknown-linux-gnu/bin/../lib/../lib/libLLVM-14-rust-1.64.0-nightly.so(+0x26a
/home/leonidv/.rustup/toolchains/nightly-2022-07-10-x86_64-unknown-linux-gnu/bin/../lib/../lib/libLLVM-14-rust-1.64.0-nightly.so(+0x2e7
/home/leonidv/.rustup/toolchains/nightly-2022-07-10-x86_64-unknown-linux-gnu/bin/../lib/../lib/libLLVM-14-rust-1.64.0-nightly.so(+0x253
/home/leonidv/.rustup/toolchains/nightly-2022-07-10-x86_64-unknown-linux-gnu/bin/../lib/../lib/libLLVM-14-rust-1.64.0-nightly.so(+0x2f0
/home/leonidv/.rustup/toolchains/nightly-2022-07-10-x86_64-unknown-linux-gnu/bin/../lib/../lib/libLLVM-14-rust-1.64.0-nightly.so(+0x372
/home/leonidv/.rustup/toolchains/nightly-2022-07-10-x86_64-unknown-linux-gnu/bin/../lib/../lib/libLLVM-14-rust-1.64.0-nightly.so(_ZN4ll
/home/leonidv/.rustup/toolchains/nightly-2022-07-10-x86_64-unknown-linux-gnu/bin/../lib/../lib/libLLVM-14-rust-1.64.0-nightly.so(_ZN4ll
/home/leonidv/.rustup/toolchains/nightly-2022-07-10-x86_64-unknown-linux-gnu/bin/../lib/../lib/libLLVM-14-rust-1.64.0-nightly.so(+0x313
/home/leonidv/.rustup/toolchains/nightly-2022-07-10-x86_64-unknown-linux-gnu/bin/../lib/../lib/libLLVM-14-rust-1.64.0-nightly.so(_ZN4llvm11PassManagerINS_13LazyCallGraph3SCCENS_15AnalysisManagerIS2_JRS1_EEEJS4_RNS_17CGSCCUpdateResultEEE3runERS2_RS5_S4_S7_+0x6c5)[0x7f12c5e51685]
/home/leonidv/.rustup/toolchains/nightly-2022-07-10-x86_64-unknown-linux-gnu/bin/../lib/../lib/libLLVM-14-rust-1.64.0-nightly.so(_ZN4llvm21DevirtSCCRepeatedPass3runERNS_13LazyCallGraph3SCCERNS_15AnalysisManagerIS2_JRS1_EEES5_RNS_17CGSCCUpdateResultE+0x3ec)[0x7f12c5f6e1cc]
/home/leonidv/.rustup/toolchains/nightly-2022-07-10-x86_64-unknown-linux-gnu/bin/../lib/../lib/libLLVM-14-rust-1.64.0-nightly.so(_ZN4llvm33ModuleToPostOrderCGSCCPassAdaptor3runERNS_6ModuleERNS_15AnalysisManagerIS1_JEEE+0xee0)[0x7f12c57b0cd0]
/home/leonidv/.rustup/toolchains/nightly-2022-07-10-x86_64-unknown-linux-gnu/bin/../lib/../lib/libLLVM-14-rust-1.64.0-nightly.so(+0x27afddd)[0x7f12c57afddd]
/home/leonidv/.rustup/toolchains/nightly-2022-07-10-x86_64-unknown-linux-gnu/bin/../lib/../lib/libLLVM-14-rust-1.64.0-nightly.so(_ZN4llvm11PassManagerINS_6ModuleENS_15AnalysisManagerIS1_JEEEJEE3runERS1_RS3_+0x144)[0x7f12c626e9d4]
/home/leonidv/.rustup/toolchains/nightly-2022-07-10-x86_64-unknown-linux-gnu/bin/../lib/../lib/libLLVM-14-rust-1.64.0-nightly.so(_ZN4llvm24ModuleInlinerWrapperPass3runERNS_6ModuleERNS_15AnalysisManagerIS1_JEEE+0x185)[0x7f12c626e2a5]
/home/leonidv/.rustup/toolchains/nightly-2022-07-10-x86_64-unknown-linux-gnu/bin/../lib/../lib/libLLVM-14-rust-1.64.0-nightly.so(+0x326e10d)[0x7f12c626e10d]
/home/leonidv/.rustup/toolchains/nightly-2022-07-10-x86_64-unknown-linux-gnu/bin/../lib/../lib/libLLVM-14-rust-1.64.0-nightly.so(_ZN4llvm11PassManagerINS_6ModuleENS_15AnalysisManagerIS1_JEEEJEE3runERS1_RS3_+0x144)[0x7f12c626e9d4]
/home/leonidv/.rustup/toolchains/nightly-2022-07-10-x86_64-unknown-linux-gnu/bin/../lib/librustc_driver-8fd17062f1c0f9c7.so(+0x27a06e1)[0x7f12cb1a06e1]
/home/leonidv/.rustup/toolchains/nightly-2022-07-10-x86_64-unknown-linux-gnu/bin/../lib/librustc_driver-8fd17062f1c0f9c7.so(+0x27709c9)[0x7f12cb1709c9]
/home/leonidv/.rustup/toolchains/nightly-2022-07-10-x86_64-unknown-linux-gnu/bin/../lib/librustc_driver-8fd17062f1c0f9c7.so(+0x2770ee4)[0x7f12cb170ee4]
/home/leonidv/.rustup/toolchains/nightly-2022-07-10-x86_64-unknown-linux-gnu/bin/../lib/librustc_driver-8fd17062f1c0f9c7.so(+0x279174a)[0x7f12cb19174a]
/home/leonidv/.rustup/toolchains/nightly-2022-07-10-x86_64-unknown-linux-gnu/bin/../lib/librustc_driver-8fd17062f1c0f9c7.so(+0x278ac2f)[0x7f12cb18ac2f]
/home/leonidv/.rustup/toolchains/nightly-2022-07-10-x86_64-unknown-linux-gnu/bin/../lib/librustc_driver-8fd17062f1c0f9c7.so(+0x2775fe0)[0x7f12cb175fe0]
/home/leonidv/.rustup/toolchains/nightly-2022-07-10-x86_64-unknown-linux-gnu/bin/../lib/libstd-1aab30bac2090a15.so(rust_metadata_std_26f91124698ed936+0xaea33)[0x7f12c86aea33]
/usr/lib/libc.so.6(+0x868fd)[0x7f12c849f8fd]
/usr/lib/libc.so.6(+0x108a60)[0x7f12c8521a60]
   Compiling nb v1.0.0
   Compiling vcell v0.1.3
   Compiling void v1.0.2
   Compiling ufmt-write v0.1.0
   Compiling cfg-if v0.1.10
   Compiling cfg-if v1.0.0
   Compiling panic-halt v0.2.0
   Compiling nb v0.1.3
   Compiling ufmt v0.1.2
   Compiling embedded-hal v0.2.7
   Compiling atmega-hal v0.1.0 (https://github.com/rahix/avr-hal?rev=1aacefb335517f85d0de858231e11055d9768cdf#1aacefb3)
   Compiling arduino-hal v0.1.0 (https://github.com/rahix/avr-hal?rev=1aacefb335517f85d0de858231e11055d9768cdf#1aacefb3)
error: could not compile `core`

Caused by:
  process didn't exit successfully: `rustc --crate-name core --edition=2021 /home/leonidv/.rustup/toolchains/nightly-2022-07-10-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/core/src/lib.rs --error-format=json --json=diagnostic-rendered-ansi,artifacts,future-incompat --crate-type lib --emit=dep-info,metadata,link -C opt-level=s -C panic=abort -C linker-plugin-lto -C debuginfo=2 -C debug-assertions=on -C metadata=2393d4325c150410 -C extra-filename=-2393d4325c150410 --out-dir /home/leonidv/temp/blink/target/avr-atmega328p/debug/deps --target /home/leonidv/temp/blink/avr-specs/avr-atmega328p.json -Z force-unstable-if-unmarked -L dependency=/home/leonidv/temp/blink/target/avr-atmega328p/debug/deps -L dependency=/home/leonidv/temp/blink/target/debug/deps --cap-lints allow` (signal: 11, SIGSEGV: invalid memory reference)
```
cargo and rustc version in blink directory:
```
leonidv@lvlinux ~/t/blink (main) [101]> cargo -V
cargo 1.64.0-nightly (c0bbd42ce 2022-07-03)
leonidv@lvlinux ~/t/blink (main)> rustc -V
rustc 1.64.0-nightly (6dba4ed21 2022-07-09)
```
cargo and rustc version in home directory:
```
leonidv@lvlinux ~> cargo -V
cargo 1.63.0 (fd9c4297c 2022-07-01)
leonidv@lvlinux ~> rustc -V
rustc 1.63.0 (4b91a6ea7 2022-08-08)
```

I attach blink folder after `cargo clean`.
[blink.zip](https://github.com/Rahix/avr-hal/files/9791357/blink.zip)

leonidv, 2022-10-15:
Sorry, It was bug in OS or something else. After reboot all compiled...

stappersg, 2022-10-15:
On Sat, Oct 15, 2022 at 12:03:53AM -0700, Leonid Vygovskiy wrote:
> Sorry, It was bug in OS or something else. After reboot all compiled...

OK

Please close the issue.

Groeten
Geert Stappers
-- 
Silence is hard to parse

closed 2022-10-15 by leonidv