Rahix/avr-hal #355: Dynamic PwmOutput

EZ64cool, 2022-10-22:
I'm pretty new to Arduino so this my be something that shouldn't be allowed, but I was trying to downgrade PwmOutput pins to a Dynamic pin type and couldn't seem to find a way to do this.

I think I understand the reasoning of timers are attached to pins, but was curious anyway.

Thanks.

Rahix, 2022-10-23:
Hi,

there is nothing against supporting that from the technical side but as you noticed, it isn't implemented at this point.  Let's actually use this issue to keep track of this feature.

A few notes:

- I think the easiest option here would be to add a new `DynamicPwm` "timer" type which selects the timer at runtime.  This is similar to what we're already doing with digital I/O and should thus probably also work well for PWM.
- I'm not sure if we can make the `.downgrade()` method work for both digital I/O and PWM.  We might need to introduce some more trait-trickery to get this working.  But I think from an API standpoint, it would be the best option.

Rahix, 2025-01-06:
@suprohub, this feature is open for anyone to implement.