Rahix/avr-hal #363: Support of Atmega8

xfmoulet, 2022-11-09:
Hi, I would like to use the Atmega8 chip, which do not seem to be supported by now.
I don't know enough of the avh-hal system to implement it myself (but I could try with some help)

stappersg, 2022-11-09:
On Wed, Nov 09, 2022 at 03:14:53AM -0800, Xavier-Frédéric Moulet wrote:
> ... but I could try with some help

#362

Rahix, 2022-11-17:
@stappersg, #362 adds a board to `arduino-hal`, that's not what the issue poster needs to do.  They need to add a new MCU to `atmega-hal` which works differently.

@xfmoulet, as a reference for adding a new MCU, maybe PR #306 is a better candidate.  You need to mostly follow what happens there.  You can open a draft PR once you got the bare basics working and need help with some more specific things (e.g. ADC, PWM, ...).  Just let me know when you get stuck anywhere.

xfmoulet, 2022-11-22:
I have an issue (so closing this PR for the moment) : adps() method prescaler_2 does not exist, only bit definitions ...
I try to find a way to temporarily disable the ADC device for the moment (I guess the other possibility would be to enable is automated extraction in avr-device ?)

Rutherther, 2022-12-21:
I think I have done the needed patches in avr-device for this. See https://github.com/Rahix/avr-device/pull/112
I've also tried to implement ATmega8 to avr-hal, see the relevant commit here: https://github.com/Rutherther/avr-hal/commit/67fb8b1ca6d992a486253153a8c191a664ac7c23

There were some problems with that as the rest of the ATmega's are different to ATmega8 (EEPROM does not use erase mode, ADC does not have DIRD). I have added macros to avr-generic-hal specific to ATmega8 (in eeprom.rs and adc.rs), does that make sense? I suppose there are other microcontrollers that will work the same way so it could be useful in the future. What do you think, @Rahix? Do you have any better idea? 

Also name of the register for wdt is different, as a hotfix I added cfg feature to avr-generic-hal wdt.rs. What is the correct way to handle this? Should I add parameters to the macro that will access the correct register? (wdtcsr for other megas, wdtcsr for ATmega8)

I have not yet tested anything except EEPROM, so I won't make a PR yet, but you can expect it in a week.

Rutherther, 2022-12-21:
Another thing is that ATMega8 does not support watchtog at 4000 ms and 8000 ms timeouts, what would the correct way to handle that be? https://github.com/Rahix/avr-hal/compare/main...Rutherther:avr-hal:main#diff-6697f39027bdefb403a8c88673c635515ca9737eb92bf78393b97feffc9b6725R39

Also the rest of the timeouts is not exaclty right either for ATmega8:
[image]

Rahix, 2023-01-05:
As we now have two PRs for ATmega8 open here, #368 by @xfmoulet and #384 by @Rutherther, how shall we proceed?  Should we just drop one or is there work to be merged from both?

xfmoulet, 2023-01-05:
I did not have enough time to pursue patching the generator, and I think @Rutherther went further than I did with its implementation, I can retire mine if you agree with it.

Rahix, 2023-01-05:
Alright, let's go that route.  Thanks anyway for initially starting the work on this topic!

closed 2023-02-23 by Rahix