Rahix/avr-hal #365: the trait bound `(avr_hal_generic::spi::Spi<Atmega, SPI, PB5, PB3, PB4, PB2>, ChipSelectPin<PB2>): embedded_hal::blocking::spi::write::Default<u8>` is not satisfied

sudokit, 2022-11-15:
```the trait bound `(avr_hal_generic::spi::Spi<Atmega, SPI, PB5, PB3, PB4, PB2>, ChipSelectPin<PB2>): embedded_hal::blocking::spi::write::Default<u8>` is not satisfied
the trait `embedded_hal::blocking::spi::write::Default<u8>` is implemented for `avr_hal_generic::spi::Spi<H, SPI, SCLKPIN, MOSIPIN, MISOPIN, CSPIN>`
required because of the requirements on the impl of `_embedded_hal_blocking_spi_Write<u8>` for `(avr_hal_generic::spi::Spi<Atmega, SPI, PB5, PB3, PB4, PB2>, ChipSelectPin<PB2>)```

Its kind of similar to #63, but it doesn't say that it is not implemented.

Rahix, 2022-11-15:
Hello,

please provide your code to reproduce the error you got.  It is impossible to provide proper help from this little information.  You'll find that people are much more willing to help when you spend a bit more effort on your bug/issue reports :)

That said, the error complains that `spi::write::Default` is not implemented for a tuple type `(Spi, ChipSelectPin)`.  I assume you didn't destructure the return value from `Spi::new()`.  It should look like this:

https://github.com/Rahix/avr-hal/blob/533159f6c6a508abe4ecec34bf5013d7a1eb0cf5/examples/arduino-uno/src/bin/uno-spi-feedback.rs#L31-L38

(The `_` could also be a name like `cs`, so `let (mut spi, mut cs) = ...;`.  It's the chip select pin, but not used in the example)

sudokit, 2022-11-15:
Hey,
Yeah sorry i got a little lazy writingt this issue. Destructuring it fixed the error. I was trying to get it to work with the ssd1306 driver and just thought that maybe it didn't work because of that destructuring. 
Thanks!

closed 2022-11-15 by sudokit