Rahix/avr-hal #373: Support Arduino Uno Wifi (rev 2)

bverhagen, 2022-11-30:
Hello,

I followed your guide on generating a new project with avr-hal-template and chose 'Arduino Uno' in the template menu.

When I execute `cargo run`, I get:

```
Finished dev [optimized + debuginfo] target(s) in 7.65s
     Running `ravedude uno -cb 57600 -P /dev/ttyACM0 target/avr-atmega328p/debug/sensor.elf`
       Board Arduino Uno
 Programming target/avr-atmega328p/debug/environment-sensor.elf => /dev/ttyACM0
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x00
```

The Arduino IDE gives exactly the same error.

However, after some investigation with arduino-cli, I found out I need the megaAVR core rather than the (I think) non-mega avr core that is used by ravedude:

```
$ ./arduino-cli board list
Port         Protocol Type              Board Name            FQBN                    Core
/dev/ttyACM0 serial   Serial Port (USB) Arduino Uno WiFi Rev2 arduino:megaavr:uno2018 arduino:megaavr
```

Using the listed FQBN on arduino-cli and the 'blink' example sketch manages to successfully upload, program and run it:

```
$ ./arduino-cli compile --fqbn arduino:megaavr:uno2018 blink
$ ./arduino-cli upload -p /dev/ttyACM0 --fqbn arduino:megaavr:uno2018 blink
avrdude: usbdev_open(): WARNING: failed to set configuration 1: Device or resource busy
```

It seems the warning can be ignored in this case, although I can't say what other process is actually using the device while doing this.

My questions for you:

- Is this a known issue and do you have pointers to solutions?
- If not, can I configure ravedude to use the above FQBN?
- If not, can you point me where I need to extend the ravedude code to support this? I don't seem to be able to locate the core setting in the generated config for avrdude.

bverhagen, 2024-03-25:
No, I switched to the standard tools using C++, which so far have worked flawlessly.

Rahix, 2024-03-31:
The [ARDUINO UNO WiFi REV2](https://store.arduino.cc/products/arduino-uno-wifi-rev2) uses the `ATmega4809` microcontroller which is not yet supported by `avr-hal` unfortunately.  See #97.

bverhagen, 2024-04-02:
In that case I will close this issue in favour of #97 as a duplicate.

Rahix, 2024-04-02:
Actually, let's keep this issue open to track addition of support for this particular board once #97 is resolved.