Rahix/avr-hal #381: Can't Compile error E0635

StormTrooper720, 2022-12-14:
When running `cargo build` I get the following error:
```
   Compiling proc-macro2 v1.0.47
error[E0635]: unknown feature `proc_macro_span_shrink`
  --> .cargo/registry/src/github.com-1ecc6299db9ec823/proc-macro2-1.0.47/src/lib.rs:92:30
   |
92 |     feature(proc_macro_span, proc_macro_span_shrink)
   |                              ^^^^^^^^^^^^^^^^^^^^^^

error: aborting due to previous error

For more information about this error, try `rustc --explain E0635`.
error: could not compile `proc-macro2`

To learn more, run the command again with --verbose.
```

Rahix, 2022-12-14:
Hi,

can you please provide steps for reproducing this error and rustc + cargo versions on your system?

fvilante, 2023-06-03:
What do you think to close this issue: reason lack of information.

?

Rahix, 2023-06-05:
Yeah, let's close this for the time being.  Please speak up if this still occurs, then we can re-open the issue.

DeepHorizons, 2024-03-26:
I appear to be having the same issue. It should be using `nightly-2024-03-22`, and I'm using `main` (`2eb28fa3024ecbf9898c8e900e744d1201ceffca`). When I try to compile the `arduino-uno` blink I get the following error:

```
cargo build --bin uno-blink
   Compiling compiler_builtins v0.1.108
   Compiling core v0.0.0 (/home/user/.rustup/toolchains/nightly-2024-03-22-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/core)
   Compiling proc-macro2 v1.0.59
   Compiling unicode-ident v1.0.9
   Compiling quote v1.0.28
   Compiling syn v1.0.109
   Compiling rustversion v1.0.12
   Compiling embedded-hal-bus v0.1.0
   Compiling paste v1.0.12
error[E0635]: unknown feature `proc_macro_span_shrink`
  --> /home/user/.cargo/registry/src/index.crates.io-6f17d22bba15001f/proc-macro2-1.0.59/src/lib.rs:92:30
   |
92 |     feature(proc_macro_span, proc_macro_span_shrink)
   |                              ^^^^^^^^^^^^^^^^^^^^^^

For more information about this error, try `rustc --explain E0635`.
error: could not compile `proc-macro2` (lib) due to 1 previous error
warning: build failed, waiting for other jobs to finish...
```

I am however able to compile it using the `nightly-2022-07-10` toolchain in the following command: `cargo +nightly-2022-07-10 build --bin uno-blink`

Rahix, 2024-03-26:
Ref <https://github.com/dtolnay/proc-macro2/issues/356>.

A few observations:

- In my own local builds, it seems that `proc-macro2 v1.0.38` is used, potentially due to a cached `Cargo.lock`.
- In CI builds, I don't see any version of `proc-macro2` appearing... ([Example](https://github.com/Rahix/avr-hal/actions/runs/8392815748/job/22986328136#step:5:1)) Not sure what's going on here.

Rahix, 2024-03-26:
> In my own local builds, it seems that proc-macro2 v1.0.38 is used, potentially due to a cached Cargo.lock.

Alright, done a `cargo update` and this indeed pushed `proc-macro2` to version `1.0.79`.  Maybe you simply need to do the same?

DeepHorizons, 2024-03-26:
Yep, that seemed to fix the example. I expected the `Cargo.lock` file to exist along side the `Cargo.toml` in the example folder but it seems the `Cargo.lock` is at the top level of the crate. Thanks for the help!