Rahix/avr-hal #388: Unable to compile example for atmega2560

arnemileswinter, 2023-01-03:
Hello!

I'm pretty new to cross-compiling.

I created a nix environment in which i can compile the .ELF file for some targets.

For example, the target avr-atmega328p.json works:
```
bash-5.1$ cargo build --target avr-specs/avr-atmega328p.json 
   Compiling compiler_builtins v0.1.85
   Compiling core v0.0.0 (/nix/store/r9ns07w4z808rpysvg4z84bx3q525yv7-rust-mixed/lib/rustlib/src/rust/library/core)
   Compiling avr-hal-generic v0.1.0 (https://github.com/rahix/avr-hal?rev=4170a773d4d76cc93433d2455ed8b14e573ebe70#4170a7
   [..... i am skipping some lines ....]
   Compiling helloworld v0.1.0 (/home/arne/Projects/hobby/rust-arduino-led-blink)
    Finished dev [optimized + debuginfo] target(s) in 20.01s
```
And i get `target/avr-atmega328p/debug/helloworld.elf`.

But the arduino that i have at my disposal is the atmega2560, i get the following output:
```
bash-5.1$ cargo build --target avr-specs/avr-atmega2560.json --verbose
       Fresh core v0.0.0 (/nix/store/r9ns07w4z808rpysvg4z84bx3q525yv7-rust-mixed/lib/rustlib/src/rust/library/core)
       Fresh rustc-std-workspace-core v1.99.0 (/nix/store/r9ns07w4z808rpysvg4z84bx3q525yv7-rust-mixed/lib/rustlib/src/rust/library/rustc-std-workspace-core)
       Fresh unicode-ident v1.0.6
       Fresh compiler_builtins v0.1.85
       Fresh proc-macro2 v1.0.49
       Fresh quote v1.0.23
       Fresh nb v1.0.0
       Fresh syn v1.0.107
       Fresh proc-macro-hack v0.5.20+deprecated
       Fresh rustversion v1.0.11
       Fresh ufmt-macros v0.2.0
       Fresh avr-device-macros v0.4.0
       Fresh nb v0.1.3
       Fresh ufmt-write v0.1.0
       Fresh bare-metal v1.0.0
       Fresh cfg-if v0.1.10
       Fresh void v1.0.2
       Fresh vcell v0.1.3
       Fresh ufmt v0.1.2
       Fresh paste v1.0.11
       Fresh embedded-hal v0.2.7
       Fresh avr-device v0.4.0
       Fresh cfg-if v1.0.0
       Fresh avr-hal-generic v0.1.0 (https://github.com/rahix/avr-hal?rev=4170a773d4d76cc93433d2455ed8b14e573ebe70#4170a773)
       Fresh panic-halt v0.2.0
       Fresh atmega-hal v0.1.0 (https://github.com/rahix/avr-hal?rev=4170a773d4d76cc93433d2455ed8b14e573ebe70#4170a773)
       Fresh arduino-hal v0.1.0 (https://github.com/rahix/avr-hal?rev=4170a773d4d76cc93433d2455ed8b14e573ebe70#4170a773)
   Compiling helloworld v0.1.0 (/home/arne/Projects/hobby/rust-arduino-led-blink)
     Running `rustc --crate-name helloworld --edition=2021 src/main.rs --error-format=json --json=diagnostic-rendered-ansi,artifacts,future-incompat --diagnostic-width=149 --crate-type bin --emit=dep-info,link -C opt-level=s -C panic=abort -C lto -C debuginfo=2 -C debug-assertions=on -C metadata=fbe7e5f7c81a8527 -C extra-filename=-fbe7e5f7c81a8527 --out-dir /home/arne/Projects/hobby/rust-arduino-led-blink/target/avr-atmega2560/debug/deps --target /home/arne/Projects/hobby/rust-arduino-led-blink/avr-specs/avr-atmega2560.json -C incremental=/home/arne/Projects/hobby/rust-arduino-led-blink/target/avr-atmega2560/debug/incremental -L dependency=/home/arne/Projects/hobby/rust-arduino-led-blink/target/avr-atmega2560/debug/deps -L dependency=/home/arne/Projects/hobby/rust-arduino-led-blink/target/debug/deps --extern arduino_hal=/home/arne/Projects/hobby/rust-arduino-led-blink/target/avr-atmega2560/debug/deps/libarduino_hal-b20db3fb6d901e6a.rlib --extern 'noprelude:compiler_builtins=/home/arne/Projects/hobby/rust-arduino-led-blink/target/avr-atmega2560/debug/deps/libcompiler_builtins-03a22ee3553a16ba.rlib' --extern 'noprelude:core=/home/arne/Projects/hobby/rust-arduino-led-blink/target/avr-atmega2560/debug/deps/libcore-8e43a845006832d0.rlib' --extern embedded_hal=/home/arne/Projects/hobby/rust-arduino-led-blink/target/avr-atmega2560/debug/deps/libembedded_hal-2eda8d6ea831d148.rlib --extern nb=/home/arne/Projects/hobby/rust-arduino-led-blink/target/avr-atmega2560/debug/deps/libnb-1af58d0fb3475ee9.rlib --extern panic_halt=/home/arne/Projects/hobby/rust-arduino-led-blink/target/avr-atmega2560/debug/deps/libpanic_halt-d2460f22359f919f.rlib --extern ufmt=/home/arne/Projects/hobby/rust-arduino-led-blink/target/avr-atmega2560/debug/deps/libufmt-a9534496297e0ed1.rlib -Z unstable-options`
error: linking with `avr-gcc` failed: exit status: 1
  |
  = note: LC_ALL="C" PATH="/nix/store/r9ns07w4z808rpysvg4z84bx3q525yv7-rust-mixed/lib/rustlib/x86_64-unknown-linux-gnu/bin:/nix/store/g2nfbpind37b3v3y6zbszmcgfmrj5v7x-patchelf-0.15.0/bin:/nix/store/wn31i3dzwahz6ccws8bs1nwyqrpgsvj7-gcc-wrapper-11.3.0/bin:/nix/store/sxdx80lmk4zkhb51f4x5dgqvxgmx55wl-gcc-11.3.0/bin:/nix/store/s7ip867mrpnnjlppbnxlcsq10gv13x2x-glibc-2.35-224-bin/bin:/nix/store/h8gvq6r4hgpa71h44dmg9qfx03mj81sv-coreutils-9.1/bin:/nix/store/62fb427ncxaaksa2k59rhbilfg68v20x-binutils-wrapper-2.39/bin:/nix/store/frh9l9nrdysasdi2gs7i241s241ngjw2-binutils-2.39/bin:/nix/store/bzc9c4x9fqwjaxs0g45qkahabc0gv75f-avrdude-7.0/bin:/nix/store/fznym4mxp2q5r0bg6zdpska27xcq1iv3-ravedude-0.1.5/bin:/nix/store/r9ns07w4z808rpysvg4z84bx3q525yv7-rust-mixed/bin:/nix/store/88awvrn16car4z5xx516drhpbl2sp5h8-avr-stage-final-gcc-wrapper-11.3.0/bin:/nix/store/7zlcl5m0vq5cl5kcl1swb67wczms78al-avr-stage-final-gcc-11.3.0/bin:/nix/store/5kh0ddszrd1d04z61kw1b2302z7mbnkp-avr-libc-avr-2.1.0/bin:/nix/store/qyrv5scy0pda6cm10mdn66nd6shh907a-avr-binutils-wrapper-2.39/bin:/nix/store/kzjyiw72pi5d7hgyrs0l1fgqq31zj6cs-avr-binutils-2.39/bin:/nix/store/h8gvq6r4hgpa71h44dmg9qfx03mj81sv-coreutils-9.1/bin:/nix/store/zml88vnkpm8if114qkbbqd1q7n3ypqqy-findutils-4.9.0/bin:/nix/store/49y3r0gr9m6k20d91kl6dgp4b9a6m72v-diffutils-3.8/bin:/nix/store/5dv5cq1lwvsijr9p16p2kp79g1dbajk3-gnused-4.8/bin:/nix/store/bcvccw6y9bfil6xrl5j7psza7hnd16ry-gnugrep-3.7/bin:/nix/store/l1fp0hyca54xbb85vfhppd16bskzx8dg-gawk-5.1.1/bin:/nix/store/89zbjdkb48ma61k76l2mzn3s0ra0wn2c-gnutar-1.34/bin:/nix/store/qs8qb1swpivkfq7i9yd52n0mw6z4ij81-gzip-1.12/bin:/nix/store/wwkyfg8b34xy16zzc9p6rkh59p4q37qx-bzip2-1.0.8-bin/bin:/nix/store/i0x4pzj96qwvkrm94317l6jbi53a2rdj-gnumake-4.4/bin:/nix/store/4xw8n979xpivdc46a9ndcvyhwgif00hz-bash-5.1-p16/bin:/nix/store/793iwfbjvg7wgpqq7r83a1qjl1yg02sf-patch-2.7.6/bin:/nix/store/a3mwv26f99ycsv9w6hrx0jjjjywvcb1n-xz-5.2.9-bin/bin:/nix/store/camlh5laf1wsklghk0vcaw7gvx4rpzd1-file-5.43/bin:/home/arne/.nix-profile/bin:/nix/var/nix/profiles/default/bin:/home/arne/.cargo/bin:/home/arne/.nix-profile/bin:/nix/var/nix/profiles/default/bin:/usr/local/bin:/usr/bin:/usr/local/sbin:/usr/lib/jvm/default/bin:/usr/bin/site_perl:/usr/bin/vendor_perl:/usr/bin/core_perl:/home/arne/.cabal/bin:/home/arne/.ghcup/bin:/home/arne/.yarn/bin:/home/arne/.emacs.d/bin:/home/arne/.yarn/bin:/home/arne/.emacs.d/bin" VSLANG="1033" "avr-gcc" "-mmcu=atmega2560" "-Wl,--as-needed" "/tmp/nix-shell.pCNOuF/rustcaIPZ4U/symbols.o" "/home/arne/Projects/hobby/rust-arduino-led-blink/target/avr-atmega2560/debug/deps/helloworld-fbe7e5f7c81a8527.arduino_hal-b20db3fb6d901e6a.arduino_hal.4ff7873c-cgu.0.rcgu.o.rcgu.o" "-Wl,--as-needed" "-L" "/home/arne/Projects/hobby/rust-arduino-led-blink/target/avr-atmega2560/debug/deps" "-L" "/home/arne/Projects/hobby/rust-arduino-led-blink/target/debug/deps" "-L" "/nix/store/r9ns07w4z808rpysvg4z84bx3q525yv7-rust-mixed/lib/rustlib/avr-atmega2560/lib" "-Wl,-Bstatic" "/home/arne/Projects/hobby/rust-arduino-led-blink/target/avr-atmega2560/debug/deps/libcompiler_builtins-03a22ee3553a16ba.rlib" "-Wl,-Bdynamic" "-lgcc" "-Wl,-znoexecstack" "-L" "/nix/store/r9ns07w4z808rpysvg4z84bx3q525yv7-rust-mixed/lib/rustlib/avr-atmega2560/lib" "-o" "/home/arne/Projects/hobby/rust-arduino-led-blink/target/avr-atmega2560/debug/deps/helloworld-fbe7e5f7c81a8527.elf" "-Wl,--gc-sections" "-no-pie"
  = note: /nix/store/kzjyiw72pi5d7hgyrs0l1fgqq31zj6cs-avr-binutils-2.39/bin/avr-ld: skipping incompatible /nix/store/5kh0ddszrd1d04z61kw1b2302z7mbnkp-avr-libc-avr-2.1.0/avr/lib/avr5/libm.a when searching for -lm
          /nix/store/kzjyiw72pi5d7hgyrs0l1fgqq31zj6cs-avr-binutils-2.39/bin/avr-ld: skipping incompatible /nix/store/5kh0ddszrd1d04z61kw1b2302z7mbnkp-avr-libc-avr-2.1.0/avr/lib/avr3/libm.a when searching for -lm
          /nix/store/kzjyiw72pi5d7hgyrs0l1fgqq31zj6cs-avr-binutils-2.39/bin/avr-ld: skipping incompatible /nix/store/5kh0ddszrd1d04z61kw1b2302z7mbnkp-avr-libc-avr-2.1.0/avr/lib/avr4/libm.a when searching for -lm
          /nix/store/kzjyiw72pi5d7hgyrs0l1fgqq31zj6cs-avr-binutils-2.39/bin/avr-ld: skipping incompatible /nix/store/5kh0ddszrd1d04z61kw1b2302z7mbnkp-avr-libc-avr-2.1.0/avr/lib/avr5/libc.a when searching for -lc
[... cut ...] e/Projects/hobby/rust-arduino-led-blink/target/avr-atmega2560/debug/deps/libpanic_halt-d2460f22359f919f.rlib --extern ufmt=/home/arne/Projects/hobby/rust-arduino-led-blink/target/avr-atmega2560/debug/deps/libufmt-a9534496297e0ed1.rlib -Z unstable-options` (exit status: 1)
```

I don't understand why it  doesn't work for the latter target.

The repository is here: https://github.com/arnemileswinter/rust-arduino-cross-compile-nix-blinking-led
and is based on the cargo template.

Thanks in advance!

Patryk27, 2023-01-04:
It looks like something on the way (rustc, llvm or avr-gcc?) doesn't like AVR6 as an architecture - which is weird, since in principle everything should be compatible.

I've checked avr-binutils (as a fellow NixOS user) and it seems that AVR6's standard library is present there, and the `symbols.o` file mentioned in the error message seems to be the same for AVR5 (e.g. atmega328p) and AVR6 (e.g. atmega2560) - maybe when rustc generates this file it doesn't set some "compatible with avr6" flag?

(I haven't checked other systems, though - maybe it's a NixOS specific thing related to how it packages avr-gcc, for instance.)

arnemileswinter, 2023-01-04:
@PaulRaUnite fixed it! 

My repository now has a working nix-powered development environment.

https://github.com/arnemileswinter/rust-atmega2560-cross-compile-nix-blinking-led

I am unsure whether we should close this issue now until the upstream error in `nightly` is fixed, or if this can be closed.

Rahix, 2023-01-04:
Let's keep it open for visibility.  Is there an upstream issue we can link to?

agausmann, 2023-01-07:
I am having the same issue with attiny404 when attempting to compile #357 on latest nightly. Can also confirm that something happened between 2022-10-22 and 2022-10-23 nightlies

agausmann, 2023-01-07:
Bisecting leads to rust-lang/rust@5642a751ecc76fff232351f5686d49475fb06a09 as the first commit with this issue.

agausmann, 2023-01-07:
Prior to this commit, `symbols.o` would not be created at all.

The function [`add_linked_symbol_object`](https://github.com/BelovDV/rustc/blob/5642a751ecc76fff232351f5686d49475fb06a09/compiler/rustc_codegen_ssa/src/back/link.rs#L1729-L1741) is responsible for creating `symbols.o` file, but there is an early return if `create_object_file` returns `None` (which it did before the mentioned commit, which adds support for AVR in `create_object_file`)

We still need to figure out why the generated object file's architecture is incompatible and how to correct it, but the `create_object_file` function is a good place to start looking.

agausmann, 2023-01-07:
I was close - the low 7 bits of `e_flags` in the ELF header are the "ISA revision", which isn't exactly the same as the MCU:

```c
#define EF_AVR_MACH 0x7F
#define E_AVR_MACH_AVR1     1
#define E_AVR_MACH_AVR2     2
#define E_AVR_MACH_AVR25   25
#define E_AVR_MACH_AVR3     3
#define E_AVR_MACH_AVR31   31
#define E_AVR_MACH_AVR35   35
#define E_AVR_MACH_AVR4     4
#define E_AVR_MACH_AVR5     5
#define E_AVR_MACH_AVR51   51
#define E_AVR_MACH_AVR6     6
#define E_AVR_MACH_AVRTINY 100
#define E_AVR_MACH_XMEGA1  101
#define E_AVR_MACH_XMEGA2  102
#define E_AVR_MACH_XMEGA3  103
#define E_AVR_MACH_XMEGA4  104
#define E_AVR_MACH_XMEGA5  105
#define E_AVR_MACH_XMEGA6  106
#define E_AVR_MACH_XMEGA7  107
```
(Copied from `binutils-2.39/include/elf/avr.h`)

The `e_flags` field should probably be set to one of these values in `create_object_file`, since that is already done for some other target architectures.

LoganDark, 2023-07-21:
this still seems to be an issue more than 6 months later, my uno is totally bricked and my mega is apparently a glitch in the matrix so i guess i will not be playing with avr today

Patryk27, 2023-07-21:
Fortunately, https://reviews.llvm.org/D154785 was merged just two days ago - I'll try to bump the rustc's llvm over the weekend :-)

LoganDark, 2023-07-21:
> Fortunately, https://reviews.llvm.org/D154785 (that solves this issue) was merged just two days ago - I'll try to bump the rustc's llvm over the weekend :-)

epic :3

<details><summary>i did end up using the super outdated rustc and succeeded in making blue</summary>

[image]

after spending way too long trying to pin down exactly which ic this completely generic and unbranded display is using

</details>

Patryk27, 2023-07-25:
Status: over https://github.com/rust-lang/llvm-project/pull/149 I've got an information that the team will be bumping-up LLVM soon :tm: -- so after they do it, all the recent fixes to the AVR backend will be brought as well 🙂

closed 2023-08-10 by Rahix