Rahix/avr-hal #390: Can't run on Gentoo

w1png, 2023-01-09:
### Problem
Can't run examples/avr-hal-template under gentoo. Rustup was setup using `rustup-init-gentoo -s`. Everything was working fine under other Distros. 

### Some info:
Kernel: `6.1.3-gentoo-dist `

❯ cargo run --bin uno-blink
```
error: failed to run `rustc` to learn about target-specific information

Caused by:
  process didn't exit successfully: `rustc - --crate-name ___ --print=file-names --target /home/w1png/dev/rust/arduino/avr-hal/avr-specs/avr-atmega328p.json --crate-type bin --crate-type rlib --crate-type dylib --crate-type cdylib --crate-type staticlib --crate-type proc-macro --print=sysroot --print=cfg` (exit status: 1)
  --- stderr
  error: Could not create LLVM TargetMachine for triple: avr-unknown-unknown: No available targets are compatible with triple "avr-unknown-unknown"
```

❯ rustup show
```
Default host: x86_64-unknown-linux-gnu
rustup home:  /home/w1png/.rustup

installed toolchains
--------------------

nightly-2022-07-10-x86_64-unknown-linux-gnu
nightly-x86_64-unknown-linux-gnu
gentoo (default)

active toolchain
----------------

nightly-2022-07-10-x86_64-unknown-linux-gnu (overridden by '/home/w1png/dev/rust/arduino/avr-hal/rust-toolchain.toml')
rustc 1.64.0-nightly (6dba4ed21 2022-07-09)
```

Would appreciate the help :)

Rahix, 2023-01-09:
Hm, I am a bit lost here...  The error message makes me believe your LLVM was not compiled with the AVR target enabled.  But rustup is supposed to ship vendored LLVM instead of relying on something from the distro?  Does this work differently on gentoo maybe?

w1png, 2023-01-10:
The AVR target is enabled:
```
[ebuild   R    ] sys-devel/llvm-15.0.6-r1:15::gentoo  USE="binutils-plugin libffi ncurses xml -debug -doc -exegesis -libedit -test -verify-sig -xar -z3 -zstd" ABI_X86="32 (64) (-x32)" LLVM_TARGETS="(AArch64) (AMDGPU) (ARM) (AVR) (BPF) (Hexagon) (Lanai) (MSP430) (Mips) (NVPTX) (PowerPC) (RISCV) (Sparc) (SystemZ) (VE) (WebAssembly) (X86) (XCore) (-ARC) (-CSKY) (-DirectX) (-LoongArch) (-M68k) (-SPIRV)" 0 KiB
```

w1png, 2023-01-10:
Ok I recompiled rust with llvm target use flags and this error dissapeared.

> /etc/portage/package.use
`>=dev-lang/rust-1.66.0:stable/1.66::gentoo llvm_targets_AVR llvm_targets_ARM`

w1png, 2023-01-10:
But now there is a different problem:
```
error[E0463]: can't find crate for `core`
  |
  = note: the `avr-atmega328p` target may not be installed
  = help: consider downloading the target with `rustup target add avr-atmega328p`

error[E0463]: can't find crate for `compiler_builtins`

error: cannot find attribute `derive` in this scope
  --> /home/w1png/.cargo/registry/src/github.com-1ecc6299db9ec823/bare-metal-1.0.0/src/lib.rs:50:3
   |
50 | #[derive(Debug)]
   |   ^^^^^^

error[E0463]: can't find crate for `core`
   --> /home/w1png/.cargo/registry/src/github.com-1ecc6299db9ec823/nb-1.0.0/src/lib.rs:189:5
    |
189 | use core::fmt;
    |     ^^^^ can't find crate
    |
    = note: the `avr-atmega328p` target may not be installed
    = help: consider downloading the target with `rustup target add avr-atmega328p`

error: cannot find attribute `derive` in this scope
  --> /home/w1png/.cargo/registry/src/github.com-1ecc6299db9ec823/bare-metal-1.0.0/src/lib.rs:15:3
   |
15 | #[derive(Clone, Copy, Debug)]
   |   ^^^^^^

error: cannot find attribute `derive` in this scope
   --> /home/w1png/.cargo/registry/src/github.com-1ecc6299db9ec823/nb-1.0.0/src/lib.rs:198:3
    |
198 | #[derive(Clone, Copy, PartialEq, Eq, PartialOrd, Ord, Hash)]
    |   ^^^^^^

error[E0463]: can't find crate for `core`
   --> /home/w1png/.cargo/registry/src/github.com-1ecc6299db9ec823/nb-1.0.0/src/lib.rs:192:27
    |
192 | pub type Result<T, E> = ::core::result::Result<T, Error<E>>;
    |                           ^^^^ can't find crate
    |
    = note: the `avr-atmega328p` target may not be installed
    = help: consider downloading the target with `rustup target add avr-atmega328p`

error[E0405]: cannot find trait `Copy` in this scope
  --> /home/w1png/.cargo/registry/src/github.com-1ecc6299db9ec823/vcell-0.1.3/src/lib.rs:31:18
   |
31 |         where T: Copy
   |                  ^^^^ not found in this scope

error[E0405]: cannot find trait `Copy` in this scope
  --> /home/w1png/.cargo/registry/src/github.com-1ecc6299db9ec823/vcell-0.1.3/src/lib.rs:39:18
   |
39 |         where T: Copy
   |                  ^^^^ not found in this scope

error[E0405]: cannot find trait `Sync` in this scope
  --> /home/w1png/.cargo/registry/src/github.com-1ecc6299db9ec823/bare-metal-1.0.0/src/lib.rs:87:16
   |
87 | unsafe impl<T> Sync for Mutex<T> where T: Send {}
   |                ^^^^ not found in this scope

error[E0405]: cannot find trait `Send` in this scope
  --> /home/w1png/.cargo/registry/src/github.com-1ecc6299db9ec823/bare-metal-1.0.0/src/lib.rs:87:43
   |
87 | unsafe impl<T> Sync for Mutex<T> where T: Send {}
   |                                           ^^^^ not found in this scope

error[E0463]: can't find crate for `core`
[... 104 lines cut ...]
117 |             Err(e) => e
    |             ^^^ not found in this scope

For more information about this error, try `rustc --explain E0463`.
Some errors have detailed explanations: E0405, E0463.
For more information about an error, try `rustc --explain E0405`.
error: could not compile `bare-metal` due to 6 previous errors
warning: build failed, waiting for other jobs to finish...
error: could not compile `cfg-if` due to 2 previous errors
Some errors have detailed explanations: E0405, E0412, E0463, E0531.
error: requires `receiver` lang_item

Some errors have detailed explanations: E0412, E0463.
For more information about an error, try `rustc --explain E0412`.
error: could not compile `void` due to 14 previous errors
error: could not compile `vcell` due to 4 previous errors
error: could not compile `ufmt-write` due to 6 previous errors
error: could not compile `nb` due to 7 previous errors
error: requires `sized` lang_item

error: could not compile `cfg-if` due to 2 previous errors
error: could not compile `panic-halt` due to 3 previous errors
```

I tried running `rustup target add avr-atmega328p` but it "does not contain component"

Rahix, 2023-01-10:
What project are you building and where from?  You need to use `build-std` to build libcore and friends alongside your code.  This is usually achieved by some settings in [`.cargo/config.toml`](https://github.com/Rahix/avr-hal/blob/main/examples/arduino-uno/.cargo/config.toml):

https://github.com/Rahix/avr-hal/blob/4c9c44c314eb061ee20556ef10d45dea36e75ee4/examples/arduino-uno/.cargo/config.toml#L1-L8

w1png, 2023-01-10:
I am trying to run `cargo run --bin uno-blink -- -P /dev/ttyUSB0` in arduino uno examples. 
I also tried running the template repo.

Rahix, 2023-01-10:
Hm, and it is not picking up the cargo config?  Can you try running this please:

```bash
cargo build -Zbuild-std=core --bin uno-blink
```

w1png, 2023-01-13:
`cargo build -Zbuild-std=core --bin uno-blink` output that I need to be running nightly version

<hr>

After I tried every method to use nightly build using rustup nothing worked.
Then I checked the use flags again and it turned out that I needed to enable:

```
>=dev-lang/rust-1.66.0:stable/1.66::gentoo llvm_targets_AVR rust-src nightly
```

Everything works now. Thanks for your help :)

<hr>

Are you interested in me writing a `gentoo_installation.md` file?

closed 2023-01-13 by w1png

w1png, 2023-01-13:
Also a quick recap for everyone who found this issue:

- Emerge rust with the following use flags: 
```
llvm_targets_AVR llvm_targets_ARM rust-src nightly
```
- Init rustup: 
```bash
rustup-init-gentoo -s
```
- Install the avr/arduino toolchain ([wiki link](https://wiki.gentoo.org/wiki/Arduino#Recommended:_Install_the_toolchain_using_crossdev))