Rahix/avr-hal #391: Arduino Uno Example Millis

Mikmist, 2023-01-09:
The arduino uno example for millis does not work with the upgraded version of avr-device. The interrupt seems to have been removed from the root avr-device crate.

Rahix, 2023-01-09:
Hm, can you be a bit more specific about the problem you're encountering?  The `TIMER0_COMPA` interrupt we're using in the millis example is available when I look at the new avr-device sources: https://docs.rs/crate/avr-device-macros/0.5.0/source/src/vector.rs

Do you get a compiler error?  Or does the interrupt just not fire when it should?

Mikmist, 2023-01-10:
When following along with the version upgrade that you did a few days ago of the `avr-device` and when I did I got a compiler error which I will include. It might be just me being a noob in rust still but I figured it was because it might have been removed since the version upgrade broke the build.

```
error[E0433]: failed to resolve: could not find `interrupt` in `avr_device`
  --> src/millis.rs:43:15
   |
43 | #[avr_device::interrupt(atmega328p)]
   |               ^^^^^^^^^ could not find `interrupt` in `avr_device`
``` 

This error dissapears when using `0.4.0`.

Rahix, 2023-01-10:
Can you prepare a reproducer project somewhere?  It seems to be working in the examples here...

Mikmist, 2023-01-11:
I might be just overlooking something, but in my attempt at reacreating the example it seems to be failing. In this [repo](https://github.com/Mikmist/temperature-sensor). If you change the version to 0.4.0 it works.

Rahix, 2023-01-12:
Ah, that's a nasty one.  You were referencing an old version of `arduino-hal` which only enabled the interrupt-runtime-support for it's 0.4 version of `avr-device` while you were separately depending on `avr-device` 0.5.  So there were two versions of `avr-device` in your dependency graph, actually.  This is a quirk of cargo's dependency resolution algorithm, unfortunately.

You can fix this by depending on the latest version of `arduino-hal` instead:

```diff
diff --git a/Cargo.toml b/Cargo.toml
index 73c0bfc..10136fa 100644
--- a/Cargo.toml
+++ b/Cargo.toml
@@ -19,12 +19,11 @@ dht11 = "0.3.1"
 
 [dependencies.arduino-hal]
 git = "https://github.com/rahix/avr-hal"
-rev = "4170a773d4d76cc93433d2455ed8b14e573ebe70"
+rev = "4c9c44c314eb061ee20556ef10d45dea36e75ee4"
 features = ["arduino-uno"]
 
 [dependencies.avr-device]
 version = "0.5.0"
-features = ["atmega328p"]
 
 [dependencies.either]
 version = "1.6.1"
```

Now the version of `arduino-hal` will also depend on `avr-device` 0.5 and enable interrupt support for it.

I'll go and upgrade the version referenced in the `avr-hal-template` ASAP....

closed 2023-01-12 by Mikmist