Rahix/avr-hal #393: External interrupts for mega2560

ZettaScript, 2023-01-17:
The following code works for Atmega328:

```rust
let dp = arduino_hal::Peripherals::take().unwrap();
let pins = arduino_hal::pins!(dp);

dp.EXINT.eicra.modify(|_, w| w.isc0().bits(INT_CHANGE));
dp.EXINT.eimsk.modify(|_, w| w.int0().set_bit());
dp.EXINT.eicra.modify(|_, w| w.isc1().bits(INT_CHANGE));
dp.EXINT.eimsk.modify(|_, w| w.int1().set_bit());
```

However it does not compile for Atmega2560:

```
error[E0599]: no method named `int0` found for mutable reference `&mut arduino_hal::pac::exint::eimsk::W` in the current scope
   --> src/main.rs:149:34
    |
149 |         dp.EXINT.eimsk.modify(|_, w| w.int0().set_bit());
    |                                        ^^^^ help: there is a method with a similar name: `int`

error[E0599]: no method named `int1` found for mutable reference `&mut arduino_hal::pac::exint::eimsk::W` in the current scope
   --> src/main.rs:151:34
    |
151 |         dp.EXINT.eimsk.modify(|_, w| w.int1().set_bit());
    |                                        ^^^^ help: there is a method with a similar name: `int`
```

I don't see any difference regarding EIMSK or INT0 between the 328's and 2560's atdf files so I don't understand.
It compiles when the eimsk lines are commented.

### System info
ArchLinux. `uname -a` -> `Linux 5.15.87-1-lts #1 SMP Thu, 12 Jan 2023 15:51:39 +0000 x86_64 GNU/Linux`
`channel = "nightly-2022-10-22"`
```
arduino-hal v0.1.0 (https://github.com/Rahix/avr-hal#4c9c44c3)
│   ├── atmega-hal v0.1.0 (https://github.com/Rahix/avr-hal#4c9c44c3)
│   │   ├── avr-device v0.5.0
```
Using the provided JSON AVR specs.

ZettaScript, 2023-01-18:
Looking at the code, datasheet and [attachInterrupt](https://github.com/arduino/ArduinoCore-avr/blob/9f8d27f09f3bbd1da1374b5549a82bda55d45d44/cores/arduino/WInterrupts.c), this looks fine but I can't make it work yet:

```rust
dp.EXINT.eicra.modify(|_, w| w.isc0().bits(INT_CHANGE));
dp.EXINT.eicra.modify(|_, w| w.isc1().bits(INT_CHANGE));
dp.EXINT.eimsk.modify(|_, w| w.int().bits(1<<4 | 1<<5));
```

closed 2023-01-18 by ZettaScript

Tellurian-Ul, 2024-03-02:
Just want to post that I'm having the same problem. I'm surprised you closed this issue.

Rahix, 2024-03-05:
@Tellurian-Ul can you elaborate on the issue a bit?  Is the problem that the generated field accessors in `avr-device` don't match register descriptions from the datasheet?

Rahix, 2024-03-05:
And for reference, it seems we are patching EIMSK for ATmega328P:

https://github.com/Rahix/avr-device/blob/1b0b406b376d0d12eb4ac189b011458a7c906ed7/patch/atmega328p.yaml#L13C1-L18C18