Rahix/avr-hal #396: Any idea why linking with `avr-gcc` fails?

xphoniex, 2023-02-09:
I'm trying to compile with `libsecp256k1` crate which requires `alloc`, so I added a dummy one just to see if it'd compile and:

```rust
error: linking with `avr-gcc` failed: exit status: 1
  |
  = note: "avr-gcc" "-mmcu=atmega328p" "-Wl,--as-needed" "/arduino-lib-rs/target/avr-atmega328p/debug/deps/arduino_lib-eba03eb7ff231966.secp256k1-39ff10e3da6aca26.secp256k1.20a321c8-cgu.0.rcgu.o.rcgu.o" "-Wl,--as-needed" "-L" "/arduino-lib-rs/target/avr-atmega328p/debug/deps" "-L" "/arduino-lib-rs/target/debug/deps" "-L" "/arduino-lib-rs/target/avr-atmega328p/debug/build/secp256k1-sys-3814b449d734bc91/out" "-L" "/.rustup/toolchains/nightly-2022-07-10-x86_64-unknown-linux-gnu/lib/rustlib/avr-atmega328p/lib" "-Wl,-Bstatic" "/tmp/rustcbzOoYn/libsecp256k1_sys-3a3f6b652e34c422.rlib" "/arduino-lib-rs/target/avr-atmega328p/debug/deps/libcompiler_builtins-7aa2be695e3a21ba.rlib" "-Wl,-Bdynamic" "-lgcc" "-Wl,-znoexecstack" "-L" "/.rustup/toolchains/nightly-2022-07-10-x86_64-unknown-linux-gnu/lib/rustlib/avr-atmega328p/lib" "-o" "/arduino-lib-rs/target/avr-atmega328p/debug/deps/arduino_lib-eba03eb7ff231966.elf" "-Wl,--gc-sections"
  = note: /tmp/rustcbzOoYn/libsecp256k1_sys-3a3f6b652e34c422.rlib: error adding symbols: File format not recognized
          collect2: error: ld returned 1 exit status
```

Any idea where this link error comes from?

Patryk27, 2023-02-18:
While I'm not sure where the error comes from, I'm pretty certain that this crate is much too heavy to run on an AVR anyway - for instance, it generates [this huge array](https://github.com/paritytech/libsecp256k1/blob/de9f27521461ad6497c4149541933a94f1a56d1b/gen/ecmult/src/lib.rs), with `pre_g` consisting of 16k elements - that's not going to fit in RAM 👀

xphoniex, 2023-02-18:
Yes. I think the error comes from the fact that `libsecp256k1` is compiled using `cc` which is not something `avr-gcc` recognizes, hence the `File format not recognized` error.

Rahix, 2023-02-20:
I guess we can close this issue for the reason @Patryk27 outlined.  If the same error is observed with other crates, we can reopen it.

closed 2023-02-20 by Rahix