Rahix/avr-hal #398: address 0x801468 of `.elf` section `.data' is not within region `data'

xphoniex, 2023-03-02:
I have a main that creates a struct using a 3rd crate, e.g. `other_crate::new()` then immediately calls `panic!("")` with nothing else on stack.

I have copied the `panic_handler` from [this](https://github.com/Rahix/avr-hal/blob/main/examples/arduino-uno/src/bin/uno-panic.rs) example. It compiles but then I get:
```cc
  = note: /usr/lib/gcc/avr/5.4.0/../../../avr/bin/ld: address 0x801468 of /project/target/avr-atmega328p/release/deps/lib-59d76aad33ce9344.elf section `.data' is not within region `data'
          /usr/lib/gcc/avr/5.4.0/../../../avr/bin/ld: address 0x801469 of /project/target/avr-atmega328p/release/deps/lib-59d76aad33ce9344.elf section `.bss' is not within region `data'
          /usr/lib/gcc/avr/5.4.0/../../../avr/bin/ld: address 0x801468 of /project/target/avr-atmega328p/release/deps/lib-59d76aad33ce9344.elf section `.data' is not within region `data'
          /usr/lib/gcc/avr/5.4.0/../../../avr/bin/ld: address 0x801469 of /project/target/avr-atmega328p/release/deps/lib-59d76aad33ce9344.elf section `.bss' is not within region `data'
          collect2: error: ld returned 1 exit status
```

something doesn't fit in the ram? or in the static area? or do I need to supply a linker parameter?

Rahix, 2023-03-03:
Hi, what chip are you compiling for?

xphoniex, 2023-03-03:
arduino uno, atmega328p

xphoniex, 2023-03-03:
btw, I'm also getting unstable behavior, e.g.:

* compiling with opt-level = "s" shows serial output as
```
5.1
��00 = 79be667ef9dcbbac55a
```
and compiling with `opt-level = "z"` returns the correct output:
```
5.1
public-key = 79be667ef9dcbbac55
```

also, declaring extra stuff on stack causes weird/unexpected behavior. in my code, if I add 
```rust
let var_i_wont_use = num::new();
```
my program won't return the same output, depending on the variation, sometimes it's garbage output and sometimes it's program re-starting over and over. (I'm testing in `qemu`)

Rahix, 2023-03-05:
Without knowing more, this looks a lot like you are using way too much memory and thus corrupting your stack and data in cases where it does compile and overflowing available program flash when it doesn't compile.

Maybe you are using a crate in your project which needs way too much space?

xphoniex, 2023-03-05:
Yeah, it looks like it. I think I'm overflowing stack, however original issue still remains, I can't add panic when creating a struct on stack which only takes ~200 bytes. If I check `avr-size`, flash size is ~10K and data size (which is in ram?) is 500 bytes. 

Unless adding a panic_handler takes 1000+ bytes, I don't see how that should not work.

Rahix, 2023-03-05:
Can you maybe prepare a minimal reproducer project?  It's easier to troubleshoot this when I can also reproduce the exact error.

> Unless adding a panic_handler takes 1000+ bytes

This may happen when your panicking with a message.  In such situations, a lot of the rust standard library formatting code is pulled in which is huge.  But right now it is hard for me to tell whether this is what is happening in your case...

xphoniex, 2023-03-06:
I moved a lot of stuff around to optimize and now I can't re-produce it. I guess struct having less impl made it possible to compile.

Still, I'm very close to the limit, already using 2040 bytes of ram!

I've tried other targets just to have more ram, such as:
* avr-atmega1280.json
* avr-atmega1284.json
* avr-atmega2560.json

But I can't run any of them in `qemu-system-avr`, usually something like `CPU 5.1` is not supported but `CPU 5` is. Any idea how to fix it?

How do your normally debug an avr program?

xphoniex, 2023-03-06:
> I prefer going with primitive "printf debugging" over the serial line.

I don't know how to do that. I couldn't extract the type into an static value. Can you add an example?

Rahix, 2023-04-08:
Sorry, this issue got lost in my inbox... I assume you mean how to put the UART object into a static so it can be accessed everywhere?  If yes, I pushed an example here: https://github.com/Rahix/avr-hal/blob/main/examples/arduino-uno/src/bin/uno-println.rs

closed 2023-04-14 by xphoniex

Rahix, 2023-04-16:
> You can get a more detailed message if you add --print-memory-usage in your avr-atmega328p.json file:

That's a nice trick, maybe we should just include that by default?

kallemooo, 2023-04-16:
> > You can get a more detailed message if you add --print-memory-usage in your avr-atmega328p.json file:
> 
> That's a nice trick, maybe we should just include that by default?

Yes, as this is a common problem in embedded projects. Specially with small devices such as the AVR.
And as memory usage printout works even on linker memory overflow error it will help the developer a lot.

Rahix, 2023-04-16:
@kallemooo, can you maybe send a PR to integrate it into our https://github.com/Rahix/avr-hal/blob/main/avr-specs/sync-from-upstream.py script (and then regenerate the specs files)?  I'm super short on time right now so it'd take a while until I can sit down to take care of it.