Rahix/avr-hal #405: Usage of `void` and replacement with `core::convert::Infallible`

aleokdev, 2023-03-31:
Hello, I've noticed a crate named `void` is used for infallible situations. However, since Rust 1.34.0, we have [`Infallible`](https://doc.rust-lang.org/std/convert/enum.Infallible.html), which performs this task. Am I missing something, or could the void type the crate offers be replaced by this type?

Rahix, 2023-03-31:
The missing pieces is the [`ResultVoidExt`](https://docs.rs/void/latest/void/trait.ResultVoidExt.html) trait which gives us the handy `void_unwrap()` method.

`core` already has an equivalent in the making, called [`into_ok()`](https://doc.rust-lang.org/core/result/enum.Result.html#method.into_ok) but unfortunately this is still unstable.

So far, my take on the situation was to wait on stabilization of `into_ok()` and switch then.  That is, unless the use of `void` proves to be problematic for downstream users in some place.  If that is the case, please speak up :)

Beanow, 2023-10-19:
See also https://github.com/rust-embedded/embedded-hal/issues/329
At least in `embedded-hal 0.2.x` there were some instances of void too. https://github.com/rust-embedded/embedded-hal/pull/150/files so regardless of what `void_unwrap` alternative to use, I think a migration to `embedded-hal 1.0` should drop the void crate too.

Rahix, 2023-10-19:
Nice, thanks for digging into this!  I wasn't aware of the `unwrap-infallible` crate until now - I'd be totally on board with replacing `void` with `unwrap-infallible` here in `avr-hal`.

You are right, the best time is probably when adopting `e-h 1.0` - until then I think the disruption is not worth it...

closed 2024-01-13 by Rahix