Rahix/avr-hal #410: running eeprom example fails on atmega328p

pppout, 2023-04-22:
I tried running the eeprom example uno-eeprom.rs on my Arduino Uno, but I couldn't get anything printed on my console.
But chainging `for i in data {` to `for  i in &data {` got it working (got 1024 `255` printed)
I ran it several times on the same device but got the same result.
Could this be a bug?

- rustc 1.64.0-nightly (6dba4ed21 2022-07-09)
- arch linux
- avr-gcc (GCC) 12.2.0

I'm a complete newbie, I have no idea how things are working...
Thanks in advance.

fvilante, 2023-05-31:
Thanks for the report!

This is probably due current example's code [is reading the entire EEPROM data capacity into SRAM at once](https://github.com/Rahix/avr-hal/blob/7b3212216899d51a9ab3b63852264147ae642e84/examples/arduino-uno/src/bin/uno-eeprom.rs#L20). This seems to cause some kind of memory overlap at run-time. The eeprom driver by the way seems to being working accordingly, just example's code appears to be not working.

As a fast fix proposal we may just reduce the chunk of bytes being read at once for something more safe for the example's context, like 10 bytes, and add some additional documentation for end user.

**Old code:**
```rust
let mut data = [0_u8; arduino_hal::Eeprom::CAPACITY as usize];
```

**Proposal:**
```rust
// KNOWN ISSUE: Avoid to read entire eeprom capacity at once 
// (see: https://github.com/rust-lang/rust/issues/63466)
// let buf_size = arduino_hal::Eeprom::CAPACITY as usize; 
let buf_size = 10; 
let mut data = [0_u8; buf_size];
```

cc: @flyingyizi

Rahix, 2023-06-01:
@fvilante, this sounds like a great idea!  Can you send a PR for it?

fvilante, 2023-06-01:
@Rahix sure!

fvilante, 2023-06-03:
> The eeprom driver by the way seems to being working accordingly, just example's code appears to be not working.

According to this #406 issue the eeprom driver must not be ok as I previously said.

Buy the way the example bug will be fixed by PR #421 

And we still have a bug in the driver.