Rahix/avr-hal #429: Uart driver for peripherals

Firstyear, 2023-07-11:
Hi there,

I have recently been using an atmega328p with avr-rs. I have a seeduino MR24HPC1 human presence sensor, which I would like to use with this board. However, the communication of the MR24HPC1 is over uart, and the atmega328p appears to only support uart on pins 0 and 1 - which are not exposed on the board, and are already used for usb-uart communication.

So provided I haven't missed something hugely wrong, from what I can see currently there is no way to use uart peripherals via avr-rs with boards like this. Curiously it works in arduino, likely via a software uart driver rather than a hardware one. 

Is there a way that I can solve the problem? Does avr-rs support software uart? Or is there a way I have missed to assign extra pins to be supported for uart peripherals? 

Thanks!

Wolfus20, 2023-10-24:
I have the exact same issue. I'm writing a driver for a sensor that uses UART as its only communication method but it's blocked on boards with only one hardware serial pin used for USB communication. The [USART struct](https://rahix.github.io/avr-hal/atmega_hal/usart/struct.Usart.html#method.new) seems to be locked into using the default hardware serial pins with no way to use other pins.

From what I can tell, a library probably has to be written to support a software serial feature, similar to the [official Arduino library](https://docs.arduino.cc/learn/built-in-libraries/software-serial).

However, I'm not sure if that's a feasible thing to achieve. Perhaps someone more experienced and familiar with the matter could chime in?

Thank you.

stappersg, 2023-10-29:
> I'll check it out and report back.

_"Stop procrastinating"_

And for those that did not get the joke:

[image]

Wolfus20, 2023-10-30:
LOL! So sorry for taking so long. I've been trying bitbang_hal but I got stuck trying to create a serial object from the crate and I was too embarrassed to admit it.

Apparently, it requires _something_ that impls CountDown and Periodic defined in embedded_hal, but I couldn't find anything like that in arduino_hal (I'm using the Arduino Uno board to test).

I figured it wasn't wise to spend so much time on a mere school project since I'm on a time limit. However, if anyone can help me get past this issue. I'd be happy to give it another shot.

Cheers.