Rahix/avr-hal #440: ravedude: Apply baudrate for console also for uploading via avrdude

michaelrommel, 2023-10-19:
Hello,

the baudrate for ravedude on the commandline is only honoured in the code path dealing with the console opened if the -c flag is given. It is NOT passed on to the avrdude programm command line. Hence there is NO baudrate specified to avrdude and that is why it is failing. It happened to me today as well with a Arduino Pro mini clone that wants to have a -b57600 on the avrdude command line.

I think it would make sense to pass on the Args.baudrate parameter on to the constructor of the avrdude command line and honour it there, if provided.

ravedude:

```
> /home/rommel/software/rust/keyboard/avr-hal/ravedude/target/release/ravedude -b57600 -P/dev/ttyUSB0 uno /home/rommel/software/rust/keyboard/avr-hal/target/avr-atmega328p/debug/uno-blink.elf
       Board Arduino Uno
 Programming /home/rommel/software/rust/keyboard/avr-hal/target/avr-atmega328p/debug/uno-blink.elf => /dev/ttyUSB0
"avrdude" "-C" "/tmp/.avrdude-8DxVCV.conf" "-c" "arduino" "-p" "atmega328p" "-P" "/dev/ttyUSB0" "-e" "-D" "-U" "flash:w:/home/rommel/software/rust/keyboard/avr-hal/target/avr-atmega328p/debug/uno-blink.elf:e"
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x00
```

avrdude directly with baudrate:

```
avrdude -v -C/tmp/avrdude.conf -patmega328p -carduino -b57600 -P/dev/ttyUSB0 -e -D -Uflash:w:/home/rommel/software/rust/keyboard/avr-hal/target/avr-atmega328p/debug/uno-blink.elf:e

avrdude: Version 6.3-20171130
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/tmp/avrdude.conf"
         User configuration file is "/home/rommel/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : /dev/ttyUSB0
         Using Programmer              : arduino
         Overriding Baud Rate          : 57600
         AVR Part                      : ATmega328P
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
           flash         65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
           lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : Arduino
         Description     : Arduino
         Hardware Version: 2
         Firmware Version: 1.16
         Vtarget         : 0.0 V
         Varef           : 0.0 V
         Oscillator      : Off
         SCK period      : 0.1 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude: Device signature = 0x1e950f (probably m328p)
avrdude: safemode: lfuse reads as 0
avrdude: safemode: hfuse reads as 0
avrdude: safemode: efuse reads as 0
avrdude: erasing chip
avrdude: reading input file "/home/rommel/software/rust/keyboard/avr-hal/target/avr-atmega328p/debug/uno-blink.elf"
avrdude: writing flash (436 bytes):

Writing | ################################################## | 100% 0.26s

avrdude: 436 bytes of flash written
avrdude: verifying flash memory against /home/rommel/software/rust/keyboard/avr-hal/target/avr-atmega328p/debug/uno-blink.elf:
avrdude: load data flash data from input file /home/rommel/software/rust/keyboard/avr-hal/target/avr-atmega328p/debug/uno-blink.elf:
avrdude: input file /home/rommel/software/rust/keyboard/avr-hal/target/avr-atmega328p/debug/uno-blink.elf contains 436 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 0.22s

avrdude: verifying ...
avrdude: 436 bytes of flash verified

avrdude: safemode: lfuse reads as 0
avrdude: safemode: hfuse reads as 0
avrdude: safemode: efuse reads as 0
avrdude: safemode: Fuses OK (E:00, H:00, L:00)

avrdude done.  Thank you.
```

I copied the ravedude-generated .avrdude.conf file to /tmp/avrdude,conf, so there should be no difference, only the baudrate parameter is missing.

Thanks for considering it. In the meantime I just solved it by using a shell script calling avrdude as runner instead of ravedude.

Rahix, 2023-10-19:
For reference, see https://github.com/Rahix/avr-hal/issues/299#issuecomment-1771719832 why the `-b` isn't passed on.

I agree that #435 is the solution for your board specifically.

About adding a "flash baudrate" parameter anyway: I'm not so sure about this because it is rather arbitrary to allow changing the baudrate but none of the other avrdude parameters... I think _if_ we want an escape hatch, it should allow overriding the entire avrdude commandline instead of just some single setting.

michaelrommel, 2023-10-19:
Yes, that is okay. I only discovered the setting in the boards file after I opened the issue. I will close it now. Thanks for the responses!

closed 2023-10-19 by michaelrommel