Rahix/avr-hal #444: Missing a release of Ravedude

emilbonnek, 2023-11-05:
There is some new stuff in ravedude which isn't installable via `cargo +stable install ravedude`. Namely the `promini-5v` code added [here](https://github.com/Rahix/avr-hal/pull/435).

It seems there is a release missing and the semver hasn't been updated.

To replicate:
```sh
> cargo +stable install ravedude
> ravedude --version
ravedude v0.1.5 (no git)
> ravedude promini-5v
thread 'main' panicked at .cargo\registry\src\index.crates.io-6f17d22bba15001f\ravedude-0.1.5\src\main.rs:90:47:
board not found
```

```sh
> cargo install --git https://github.com/Rahix/avr-hal ravedude --branch main
> ravedude --version
ravedude 190f2c3cb8d2
> ravedude promini-5v
Board SparkFun Pro Mini 5V (16MHz)
```

stappersg, 2023-11-05:
For what it is worth:
* ravedude version 0.1.5 is from 2022-09-18
* merge request #435 is from 2023-09-04

> It seems there is a release missing and the semver hasn't been updated.

As I see it: There was no ravedude release since september 2022.

Rahix, 2023-11-06:
Fair enough, published `ravedude` version 0.1.6 with all the changes since the last release (see [CHANGELOG](https://github.com/Rahix/avr-hal/blob/main/ravedude/CHANGELOG.md)).

closed 2023-11-06 by Rahix