Rahix/avr-hal #446: Support pins A6 and A7 of Arduino Nano

Fundevoge, 2023-11-07:
As can be seen in the image, the Arduino Nano gives access to `ADC_MUX` `111` and `ADC_MUX` `110` via the pins A6 and A7.
It would be nice if these pins could be used somehow but it probably would require some refactoring as the availability of these pins depends on the **package/board** variant (_TQFP and VQFN Package Only_; _Arduino Uno_ has the wrong package, _Arduino Pro Mini_ does not forward it) and not on the **processor** variant.

[image]

Rahix, 2023-11-07:
They are supported, access just works a bit different than for all the other pins.  Check the example code here:

https://github.com/Rahix/avr-hal/blob/06ad11ffdbc64b1e0e43c4910be4110d488662e1/examples/arduino-nano/src/bin/nano-adc.rs#L47-L53

Fundevoge, 2023-11-07:
I see, thanks for pointing it out. 
It would be convenient to be able to use `pins.a6` and then get a `Pin<Analog, ADC6>` or similar that does not implement the `PinOps` so it cannot be converted, but implements `AdcChannel` so `a6.analog_read()` is possible.
But I also understand that this is not the most important feature...

Rahix, 2023-11-19:
If you manage to make this work nicely, I'd be happy to take a PR implementing this :)

So far my reasoning was that writing code generic over ADC channels is already possible using `AdcChannel` so I didn't bother going any further here.