Rahix/avr-hal #450: generic spi?

marrsus, 2023-11-12:
i am working with an pololu ZUMO 32U4 oled. it uses an atmega32u4. The problem is i can't figure out how to make a spi connection with the oled. the problem being that the makers of this board uses other pins and i can't seam to find a spi implementation that allow for me to make the spi interface

Rahix, 2023-11-16:
Hi, we only implement a driver for the hardware SPI peripheral here in `avr-hal`.  If you need to talk SPI with different pins you have to use a software implementation of the protocol instead.  Maybe the one from [`bitbang-hal`](https://docs.rs/bitbang-hal/latest/bitbang_hal/) does the trick?

marrsus, 2023-11-17:
looks promising, do you have a embedded-hal::timer implementation for avr-hal or do i have to try my luck at making my own

Rahix, 2023-11-19:
Hm, I'm afraid we don't have an `e-h::timer` implementation right now :/  Your best bet right now is building your own using a hardware timer and its registers directly...