Rahix/avr-hal #455: Arduino UNO R4 WiFi Support? Compilation Error `Error: no matching serial port found`

MarcusWentz, 2023-11-24:
This script is working on my Arduino UNO R3:

```rust
// Tutorial from: https://blog.logrocket.com/complete-guide-running-rust-arduino/
// Template generated with: 
// cargo generate --git https://github.com/Rahix/avr-hal-template.git
// Run and automatically flash Arduino running:
// cargo run

#![no_std]
#![no_main]

use panic_halt as _;

#[arduino_hal::entry]
fn main() -> ! {
    let dp = arduino_hal::Peripherals::take().unwrap();
    let pins = arduino_hal::pins!(dp);

    let mut led = pins.d13.into_output();

    loop {
        led.toggle();
        arduino_hal::delay_ms(1000);
    }
}
```

However on the Arduino UNO R4 WiFi, I get:

```rust
Finished dev [optimized + debuginfo] target(s) in 11.80s
     Running `ravedude uno -cb 57600 target/avr-atmega328p/debug/test-rust-ard.elf`
       Board Arduino Uno
Error: no matching serial port found, use -P or set RAVEDUDE_PORT in your environment

Caused by: Serial port not found.
```

minirop, 2023-11-25:
The R4 isn't AVR but ARM. So you need another HAL. You might be able to look at the atsamd-rs project which is for ARM boards by Microchip.

MarcusWentz, 2023-11-26:
I will look into atsamd-rs.

Thank you @minirop .

closed 2023-11-26 by MarcusWentz