Rahix/avr-hal #460: core::fmt::write! causes unexpected behavior

djdisodo, 2023-12-07:
```rust
#![no_std]
#![no_main]

use core::fmt::Write;
use core::panic::PanicInfo;
use arduino_hal::delay_ms;
use ufmt::uWrite;

#[arduino_hal::entry]
fn main() -> ! {
    let dp = arduino_hal::Peripherals::take().unwrap();
    let pins = arduino_hal::pins!(dp);

    /*
     * For examples (and inspiration), head to
     *
     *     https://github.com/Rahix/avr-hal/tree/main/examples
     *
     * NOTE: Not all examples were ported to all boards!  There is a good chance though, that code
     * for a different board can be adapted for yours.  The Arduino Uno currently has the most
     * examples available.
     */

    delay_ms(1000);
    panic!("UwU");

    loop {
    }
}

//modified from uno-panic.rs
#[panic_handler]
fn panic(info: &PanicInfo) -> ! {
    avr_hal_generic::avr_device::interrupt::disable();
    let dp = unsafe { arduino_hal::Peripherals::steal() };
    let pins = arduino_hal::pins!(dp);
    let mut serial = arduino_hal::default_serial!(dp, pins, 57600);
    serial.write_str("this should print properly at least but doesn't");
    delay_ms(1000);
    let mut writer = Writer {
        u_write: serial
    };
    writeln!(&mut writer, "{}", info).ok();
    loop {

    }
}

struct Writer<T> {
    u_write: T
}

impl<T: uWrite> Write for Writer<T> {
    fn write_str(&mut self, s: &str) -> core::fmt::Result {
        self.u_write.write_str(s).map_err(|x| match x {
            _ => core::fmt::Error
        })
    }

    fn write_char(&mut self, c: char) -> core::fmt::Result {
        self.u_write.write_char(c).map_err(|x| match x {
            _ => core::fmt::Error
        })
    }
}
```
this is my code
what i expected:
[image]
this screenshot is actually from qemu
(the bug is not reproducible on qemu)

what i got:
[image]
it somehow repeats

i tried to modify the code so that it uses heapless
```rust
//modified from uno-panic.rs
#[panic_handler]
fn panic(info: &PanicInfo) -> ! {
    avr_hal_generic::avr_device::interrupt::disable();
    let dp = unsafe { arduino_hal::Peripherals::steal() };
    let pins = arduino_hal::pins!(dp);
    let mut serial = arduino_hal::default_serial!(dp, pins, 57600);
    serial.write_str("this should print properly at least but doesn't");
    delay_ms(1000);
    let mut buffer = heapless::String::<256>::new();
    writeln!(&mut buffer, "{}", info).ok();
    serial.write_str(buffer.as_str()).ok();
    loop {

    }
}
```
different output but still
[image]

i later tried to make buffer static but didn't worked
i won't post the result tho
```rust
static mut BUFFER: heapless::String<256> = String::new();

//modified from uno-panic.rs
#[panic_handler]
fn panic(info: &PanicInfo) -> ! {
    avr_hal_generic::avr_device::interrupt::disable();
    let dp = unsafe { arduino_hal::Peripherals::steal() };
    let pins = arduino_hal::pins!(dp);
    let mut serial = arduino_hal::default_serial!(dp, pins, 57600);
    serial.write_str("this should print properly at least but doesn't");
    delay_ms(1000);
    unsafe {
        writeln!(&mut BUFFER, "{}", info).ok();
        serial.write_str(BUFFER.as_str()).ok();
    }
    loop {

    }
}
```

closed 2023-12-09 by djdisodo

Rahix, 2023-12-25:
The problem is that `core::fmt` uses a lot of memory for its formatting.  For any non-trivial cases, it usually surpasses the capabilities of an AVR chip so essentially you just cannot use it at all.  That's why all the examples make use of the `ufmt` crate instead.

However, there is one place where `core::fmt` sneaks in and cannot be replaced so easily: panic message formatting.  The only real solution at the moment is to make sure that all panics do not provide a formatted message and instead simply panic with a static message / no message.  For example, you need to replace all `Result::unwrap()` calls with something like

```rust
trait ResultExt {
    type T;
    fn unwrap_avr(self) -> Self::T;
}

impl<T, E> UnwrapExt for Result<T, E> {
    type T = T;

    fn unwrap_avr(self) -> Self::T {
        match self {
            Ok(t) => t,
            Err(_) => panic!(),
        }
    }
}
```

(Or alternatively follow your suggestion from https://github.com/Rahix/avr-hal/discussions/463 but that means you don't get any custom panic behavior at all)

djdisodo, 2023-12-25:
@Rahix i thought it might be caused by insufficient sram but i didn't thought further because compiling with release profile fixed it
(maybe core::fmt heavily rely on compiler optimizations to reduce ram usage)

about immediate abort, iirc just avoiding print of error message didn't strip out all of the error messages

Rahix, 2023-12-25:
Relying on optimizations with `core::fmt` is very hit or miss.  Compiling with LTO manages to eliminate a lot of overhead a surprising number of times, but it does not always work.  And when it doesn't, there is little you can do, unfortunately.

> iirc just avoiding print of error message didn't strip out all of the error messages

The things is that you need to ensure all panicking code paths do not lead to a panic with a message.  This is not an easy feat of course.  The `ResultExt` shown above just takes care of the worst offender - `Result::unwrap()` which usually pulls in the full `core::fmt::Debug` implementations of the error types.  But to get rid of all error message strings, you'll of course need to do a lot more.  A list that certainly isn't exhaustive:

- `Option::unwrap()`
- Any uses of `ops::Index`/`ops::IndexMut` where the bounds check isn't optimized away - replace with fallible `.get()`/`.get_mut` calls and then work around the following `Option::unwrap()` as before.
- If overflow checks are enabled, you get panicking code paths from a lot of maths operations.  Either disable overflow checks or use the fallible methods on integer types instead (e.g. `u32::checked_add()`)