Rahix/avr-hal #471: roadmap to stable rust

rursprung, 2023-12-23:
it'd be great if you could add a roadmap which identifies what still needs to be done in order to build & use this with a stable rust toolchain (instead of nightly). thanks!

Rahix, 2023-12-24:
I don't think we will see AVR on stable rust anytime soon, there is so much that still needs to be worked on... In any case, here are the big topics that come to my mind:

- [ ] The LLVM backend for AVR is still not entirely stable.  I don't have good overview at this time where the limitations are - maybe @Patryk27 can comment on that.  He's responsible for a large number of codegen fixes without which AVR would still be mostly unusable.  Ideally, we also want a stable lld for AVR so we can drop all dependencies on avr-gcc and avr-binutils.
- [ ] We depend on the use of json-target specs which are not really well supported throughout the rust toolchain and surrounding ecosystem - many things still break when using them.
- [ ] We need the beast that is https://github.com/rust-lang/wg-cargo-std-aware without which we cannot build libcore for the different AVR architecture flavors.

Patryk27, 2023-12-24:
> The LLVM backend for AVR is still not entirely stable. 

I think nowadays it's pretty good, there's not [that many](https://github.com/llvm/llvm-project/issues?q=is%3Aissue+is%3Aopen+label%3Abackend%3AAVR+) opened bugs - @benshi001 is doing great job maintaining the backend. Most of the issues in the recent months were caused by linking wrong intrinsics, not by codegen itself being wrong. 

> Ideally, we also want a stable lld for AVR so we can drop all dependencies on avr-gcc and avr-binutils.

This shouldn't be extremely difficult - IIRC we use avr-gcc mostly because it brings [certain hand-written intrinsics](https://github.com/gcc-mirror/gcc/blob/3eb9cae6d375d222787498b15ac87f383b3834fe/libgcc/config/avr/lib1funcs.S) (e.g. for i32 multiplication) - in principle those could be sourced from [compiler-builtins](https://github.com/rust-lang/compiler-builtins) instead, but the issue is that AVR requires [a custom calling convention](https://gcc.gnu.org/wiki/avr-gcc#Exceptions_to_the_Calling_Convention) that hasn't been yet implemented in LLVM (and thus exposed in rustc).

Moving from hand-written intrinsics to auto-generated ones would also warrant a benchmark, since e.g. making i32 multiplication ten times slower wouldn't be that great 👀 

(or maybe rustc/lld could just automatically link those hand-written intrinsics, to reuse the work already done? I'm not sure on potential license issues here, though.)

benshi001, 2023-12-25:
I am not familiar with rust, and I am now taking the responsibility of maintaining the llvm-avr backend (though it is my spare time work). Some known issues:

1. The lld does not support some odd linker script syntax of avr-ld, such as `KEEP(SORT(*)(.dtors))` as shown in [binutils](https://sourceware.org/git/?p=binutils-gdb.git;a=blob;f=ld/scripttempl/avr.sc;h=00ff49aacf46b46f150e0a980585ff20f4d3938d;hb=HEAD#l223)

benshi001, 2023-12-25:
The [custom calling convention](https://gcc.gnu.org/wiki/avr-gcc#Exceptions_to_the_Calling_Convention) does not break the standard version, which is llvm's current implementation. So current AVR assembly generated by llvm should work, though a little less efficient.

And of course, I am happy if you can offer rust's implementation of int32 mul/div to compiler-rt, as shown at [here](https://github.com/llvm/llvm-project/tree/main/compiler-rt/lib/builtins/avr).

benshi001, 2023-12-25:
I think the `llvm-avr` backend should be more stable than `lld`, and I really appreciate if you can try `lld` and feedback me issues !

tones111, 2025-04-20:
@Patryk27 @benshi001 

Is there much left to do before we can start experimenting with using `lld` (or `rust-lld`?)? Maybe we're close to being able to start working on the runtime initialization code and linker scripts?

From rust-lang/compiler-builtins#711 it sounds like the calling convention shouldn't be problematic any more (as long as the app doesn't attempt signed division).

It looks like we should be able to tell `rustc` to [use `lld` as the linker](https://github.com/rust-lang/rust/commit/78ddabf31d3147a19be77c3b38b49fe60915d75a#diff-3f87e69f5e21712681d6fca5b6b70542c10eaf49bf48ff16445ddd17f1ae6dadR156), but then it's hard-coded to `avr-gcc` in [the avr spec](https://github.com/rust-lang/rust/commit/78ddabf31d3147a19be77c3b38b49fe60915d75a#diff-e0ff5e2d420b8a97a85b2df82e0bfc8e963d2a8b3e732bc61a236f68b55871b9R18).

I tried to modify the spec file to have it call `rust-lld`, but my `rustc` build errors trying to invoke `rust-lld`.  I imagine `lld` likely needs different arguments.  Thanks in advance for any pointers on modifying `rustc` to drop the avr-gcc dependencies so we can find out what's still missing.

Patryk27, 2025-04-20:
I haven't played around lld much, but from what I know, it should just-work, so it's mostly the matter of figuring out command line arguments; I'll take a quick look tomorrow.

benshi001, 2025-04-21:
I am not familiar with rustc and rust-lld, the original lld of the llvm-project's main branch should work, except a known issue:

https://github.com/llvm/llvm-project/issues/115266

There may be other unknown issues, please submit them to https://github.com/llvm/llvm-project/issues/ and assign to me, when you encounter them.

benshi001, 2025-04-21:
You can check the linker arguments via compiling a c++ helloworld program with `-###`, and check the arguments passed to avr-ld. such as `clang helloworld.c -O3 -Wall -### --target=avr -mmcu=atmega328`.

All the arguments in passed to avr-ld in the above command should work for lld, except a tiny modification to the linker script to avoid https://github.com/llvm/llvm-project/issues/115266

LuigiPiucco, 2025-04-22:
> but then it's hard-coded to `avr-gcc` in [the avr spec](https://github.com/rust-lang/rust/commit/78ddabf31d3147a19be77c3b38b49fe60915d75a#diff-e0ff5e2d420b8a97a85b2df82e0bfc8e963d2a8b3e732bc61a236f68b55871b9R18).

@tones111 To circumvent this temporarily, you can use a .json spec like we did before `avr-none`. Later a proper target can be upstreamed when we have at least a working proposal. Just beware that `avr-ld` (and I assume `lld` as well) doesn't understand the `-mmcu=*` machine specification like `avr-gcc` does, it has to be `-m*` where `*` is the architecture revision, not the MCU name. You'll have to adjust the flags accordingly. And the linker flavor should probably be changed as well.