Rahix/avr-hal #477: Linker errors when compiling for atmega328p with more than one instance of libm::atan referenced in a given function

pmnlla, 2023-12-27:
I have a piece of code which compiles the base joint angle for a 2 joint mechanical arm. It's a piece for internal use and context isn't a necessity for this issue. However, I've narrowed it down to having the exact line that produces the following linker error:
```rust
	pub fn produce_base_joint_angle(self) -> f64 {
        if self.mid_joint_angle == 0.0 {
            false; // todo: replace this with actual error handling.
        }

        let a:f64 = &self.mid_pole_length * (libm::sin(self.mid_joint_angle).to_degrees());
        let b:f64 = &self.base_pole_length + (&self.mid_pole_length * libm::cos(self.mid_joint_angle).to_degrees());

        let c:f64 = libm::atan(a/b).to_degrees();

        let y_x_tan:f64 = &self.end_point.y / &self.end_point.x;

        let q1:f64 = libm::atan(y_x_tan).to_degrees() + c; // when this specific line is commented out, the code compiles successfully into an ELF binary.
        // println!("{}", &q1); // commented out when porting to AVR - not applicable
        q1
    }
```
The linker error, as taken from my CI workflow, is as follows:
```
= note: LC_ALL="C" PATH="/home/runner/.rustup/toolchains/nightly-2023-08-08-x86_64-unknown-linux-gnu/lib/rustlib/x86_64-unknown-linux-gnu/bin:/snap/bin:/home/runner/.local/bin:/opt/pipx_bin:/home/runner/.cargo/bin:/home/runner/.config/composer/vendor/bin:/usr/local/.ghcup/bin:/home/runner/.dotnet/tools:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin" VSLANG="1033" "avr-gcc" "-mmcu=atmega328p" "/tmp/rustcHzhFbH/symbols.o" "/home/runner/work/nari/nari/nari-avr-target/target/avr-atmega328p/debug/deps/nari_avr_target-1e0badf000416126.arduino_hal-47c3b92a15d537f7.arduino_hal.b706e65ca0090060-cgu.0.rcgu.o.rcgu.o" "-Wl,--as-needed" "-L" "/home/runner/work/nari/nari/nari-avr-target/target/avr-atmega328p/debug/deps" "-L" "/home/runner/work/nari/nari/nari-avr-target/target/debug/deps" "-L" "/home/runner/.rustup/toolchains/nightly-2023-08-08-x86_64-unknown-linux-gnu/lib/rustlib/avr-atmega328p/lib" "-Wl,-Bstatic" "/home/runner/work/nari/nari/nari-avr-target/target/avr-atmega328p/debug/deps/libcompiler_builtins-e009e55f786d72c4.rlib" "-Wl,-Bdynamic" "-lgcc" "-Wl,-z,noexecstack" "-L" "/home/runner/.rustup/toolchains/nightly-2023-08-08-x86_64-unknown-linux-gnu/lib/rustlib/avr-atmega328p/lib" "-o" "/home/runner/work/nari/nari/nari-avr-target/target/avr-atmega328p/debug/deps/nari_avr_target-1e0badf000416126.elf" "-Wl,--gc-sections"
  = note: /usr/lib/gcc/avr/5.4.0/../../../avr/bin/ld: address 0x800908 of /home/runner/work/nari/nari/nari-avr-target/target/avr-atmega328p/debug/deps/nari_avr_target-1e0badf000416126.elf section `.data' is not within region `data'
          /usr/lib/gcc/avr/5.4.0/../../../avr/bin/ld: address 0x800909 of /home/runner/work/nari/nari/nari-avr-target/target/avr-atmega328p/debug/deps/nari_avr_target-1e0badf0004[161](https://github.com/pmnlla/nari/actions/runs/7333887546/job/19969915408#step:4:162)26.elf section `.bss' is not within region `data'
          /usr/lib/gcc/avr/5.4.0/../../../avr/bin/ld: address 0x800908 of /home/runner/work/nari/nari/nari-avr-target/target/avr-atmega328p/debug/deps/nari_avr_target-1e0badf000416126.elf section `.data' is not within region `data'
          /usr/lib/gcc/avr/5.4.0/../../../avr/bin/ld: address 0x800909 of /home/runner/work/nari/nari/nari-avr-target/target/avr-atmega328p/debug/deps/nari_avr_target-1e0badf000416126.elf section `.bss' is not within region `data'
          /home/runner/work/nari/nari/nari-avr-target/target/avr-atmega328p/debug/deps/libcompiler_builtins-e009e55f786d72c4.rlib(compiler_builtins-e009e55f786d72c4.compiler_builtins.8061183a12bc77bf-cgu.1.rcgu.o): In function `compiler_builtins::float::mul::mul::hc52f2fcbbb6f2e34':
          /home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/compiler_builtins-0.1.98/src/float/mul.rs:(.text._ZN17compiler_builtins5float3mul8__muldf317h90a980e6381d6d16E+0x952): undefined reference to `core::panicking::panic::hecd845025dfa9f2e'
          /home/runner/work/nari/nari/nari-avr-target/target/avr-atmega328p/debug/deps/libcompiler_builtins-e009e55f786d72c4.rlib(compiler_builtins-e009e55f786d72c4.compiler_builtins.8061183a12bc77bf-cgu.1.rcgu.o): In function `_$LT$f64$u20$as$u20$compiler_builtins..float..Float$GT$::normalize::h71176ea743fba888':
          /home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/compiler_builtins-0.1.98/src/float/mod.rs:[164](https://github.com/pmnlla/nari/actions/runs/7333887546/job/19969915408#step:4:165): undefined reference to `core::panicking::panic::hecd845025dfa9f2e'
          /home/runner/work/nari/nari/nari-avr-target/target/avr-atmega328p/debug/deps/libcompiler_builtins-e009e55f786d72c4.rlib(compiler_builtins-e009e55f786d72c4.compiler_builtins.8061183a12bc77bf-cgu.2.rcgu.o): In function `compiler_builtins::float::trunc::trunc::hbc7046c939c84170':
          /home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/compiler_builtins-0.1.98/src/float/trunc.rs:(.text._ZN17compiler_builtins5float5trunc12__truncdfsf217hf6ffce40c3a8ccfaE+0x50a): undefined reference to `core::panicking::panic::hecd845025dfa9f2e'
          /home/runner/work/nari/nari/nari-avr-target/target/avr-atmega328p/debug/deps/libcompiler_builtins-e009e55f786d72c4.rlib(compiler_builtins-e009e55f786d72c4.compiler_builtins.8061183a12bc77bf-cgu.3.rcgu.o): In function `compiler_builtins::float::div::div64::h651fa15c21321d46':
          /home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/compiler_builtins-0.1.98/src/float/div.rs:(.text._ZN17compiler_builtins5float3div8__divdf317h2c2f49c433083acaE+0xdc6): undefined reference to `core::panicking::panic::hecd845025dfa9f2e'
          /home/runner/work/nari/nari/nari-avr-target/target/avr-atmega328p/debug/deps/libcompiler_builtins-e009e55f786d72c4.rlib(compiler_builtins-e009e55f786d72c4.compiler_builtins.8061183a12bc77bf-cgu.0.rcgu.o): In function `compiler_builtins::float::add::add::h0562b3e6e1b1dcc4':
          /home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/compiler_builtins-0.1.98/src/float/add.rs:(.text._ZN17compiler_builtins5float3add8__adddf317h966457fb95ac8e72E+0xcea): undefined reference to `core::panicking::panic::hecd845025dfa9f2e'
          /home/runner/work/nari/nari/nari-avr-target/target/avr-atmega328p/debug/deps/nari_avr_target-1e0badf000416126.arduino_hal-47c3b92a15d537f7.arduino_hal.b706e65ca0090060-cgu.0.rcgu.o.rcgu.o: In function `core::f64::_$LT$impl$u20$f64$GT$::is_nan::h018e74ede0409540':
          /home/runner/.rustup/toolchains/nightly-2023-08-08-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/core/src/num/f64.rs:428: undefined reference to `__unorddf2'
          collect2: error: ld returned 1 exit status
```

I've ran out of sane ways to attempt to fix the issue. I know that certain libraries need to be compiled with all optimizations at all times, however adding (basically) an -O3 flag to my cargo config didn't help. Outsourcing the arctan to another function and calling that instead did not work either.  I don't know where to go as this error refers to a panic instruction, which I have never implemented. My next step is likely to rewrite arctan tbh, the more dependencies I can get out of the way the better imo

This is an operation that requires inverse tangent, thus libm is required as rust's stdlib cannot be provided.

pmnlla, 2023-12-27:
The issue with that is that, to my understanding, most float operations in libm require f64. I can try storing my values as f32 and converting them to f64 when I provide them, but I don't know how much help that will be. Will report back when Ive tried it.

Rahix, 2023-12-27:
No, converting them won't help I think.  But you should be able to use [`libm::atanf`](https://docs.rs/libm/latest/libm/fn.atanf.html) instead of [`libm::atan`](https://docs.rs/libm/latest/libm/fn.atan.html), no?

pmnlla, 2023-12-28:
Also figured this error might be related. it doesn't stop the code from compiling, just linking with avr-gcc I assume. rustc doesn't even warn me about this, just the faulty rust-analyzer instance in vscode.
[image]

I swapped to f32 for all of my values. <func>f instead of <func>. Still getting the linking error.  I don't see a difference, but I'll send the updated version in case you do:
```
error: linking with `avr-gcc` failed: exit status: 1
  |
  = note: LC_ALL="C" PATH="/home/runner/.rustup/toolchains/nightly-2023-08-08-x86_64-unknown-linux-gnu/lib/rustlib/x86_64-unknown-linux-gnu/bin:/snap/bin:/home/runner/.local/bin:/opt/pipx_bin:/home/runner/.cargo/bin:/home/runner/.config/composer/vendor/bin:/usr/local/.ghcup/bin:/home/runner/.dotnet/tools:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin" VSLANG="1033" "avr-gcc" "-mmcu=atmega328p" "/tmp/rustcbhRBap/symbols.o" "/home/runner/work/nari/nari/nari-avr-target/target/avr-atmega328p/debug/deps/nari_avr_target-1e0badf000416126.arduino_hal-47c3b92a15d537f7.arduino_hal.b706e65ca0090060-cgu.0.rcgu.o.rcgu.o" "-Wl,--as-needed" "-L" "/home/runner/work/nari/nari/nari-avr-target/target/avr-atmega328p/debug/deps" "-L" "/home/runner/work/nari/nari/nari-avr-target/target/debug/deps" "-L" "/home/runner/.rustup/toolchains/nightly-2023-08-08-x86_64-unknown-linux-gnu/lib/rustlib/avr-atmega328p/lib" "-Wl,-Bstatic" "/home/runner/work/nari/nari/nari-avr-target/target/avr-atmega328p/debug/deps/libcompiler_builtins-e009e55f786d72c4.rlib" "-Wl,-Bdynamic" "-lgcc" "-Wl,-z,noexecstack" "-L" "/home/runner/.rustup/toolchains/nightly-2023-08-08-x86_64-unknown-linux-gnu/lib/rustlib/avr-atmega328p/lib" "-o" "/home/runner/work/nari/nari/nari-avr-target/target/avr-atmega328p/debug/deps/nari_avr_target-1e0badf000416126.elf" "-Wl,--gc-sections"
  = note: /home/runner/work/nari/nari/nari-avr-target/target/avr-atmega328p/debug/deps/libcompiler_builtins-e009e55f786d72c4.rlib(compiler_builtins-e009e55f786d72c4.compiler_builtins.8061183a12bc77bf-cgu.1.rcgu.o): In function `compiler_builtins::float::mul::mul::h24cd00a6a6a419fb':
          /home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/compiler_builtins-0.1.98/src/float/mul.rs:(.text._ZN17compiler_builtins5float3mul8__mulsf317h3d0aff828d32e933E+0x5e8): undefined reference to `core::panicking::panic::hecd845025dfa9f2e'
          /home/runner/work/nari/nari/nari-avr-target/target/avr-atmega328p/debug/deps/libcompiler_builtins-e009e55f786d72c4.rlib(compiler_builtins-e009e55f786d72c4.compiler_builtins.8061183a12bc77bf-cgu.1.rcgu.o): In function `_$LT$f32$u20$as$u20$compiler_builtins..float..Float$GT$::normalize::ha5ce90c65520e46b':
          /home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/compiler_builtins-0.1.98/src/float/mod.rs:164: undefined reference to `core::panicking::panic::hecd845025dfa9f2e'
          /home/runner/work/nari/nari/nari-avr-target/target/avr-atmega328p/debug/deps/libcompiler_builtins-e009e55f786d72c4.rlib(compiler_builtins-e009e55f786d72c4.compiler_builtins.8061183a12bc77bf-cgu.3.rcgu.o): In function `compiler_builtins::float::div::div32::ha54c26540da8b084':
          /home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/compiler_builtins-0.1.98/src/float/div.rs:(.text._ZN17compiler_builtins5float3div8__divsf317h74a7c6c4ec02fd98E+0x764): undefined reference to `core::panicking::panic::hecd845025dfa9f2e'
          /home/runner/work/nari/nari/nari-avr-target/target/avr-atmega328p/debug/deps/libcompiler_builtins-e009e55f786d72c4.rlib(compiler_builtins-e009e55f786d72c4.compiler_builtins.8061183a12bc77bf-cgu.0.rcgu.o): In function `compiler_builtins::float::add::add::h6d8e598238daa834':
          /home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/compiler_builtins-0.1.98/src/float/add.rs:(.text._ZN17compiler_builtins5float3add8__addsf317h05779f8ebaf03a11E+0x7d0): undefined reference to `core::panicking::panic::hecd845025dfa9f2e'
          collect2: error: ld returned 1 exit status
          ```

Rahix, 2023-12-28:
Ah, this is now another known kind of error: `compiler-builtins` must never panic but when built in some configurations, it does include panicking codepaths.  Are you doing a release build or a debug build right now?

pmnlla, 2023-12-28:
I assume debug.

Rahix, 2023-12-28:
Does it also happen in release builds?

Please also try whether this fix helps:

https://github.com/rust-lang/wg-cargo-std-aware/issues/62#issuecomment-732245421

pmnlla, 2023-12-28:
Doesn't happen in release builds. Got a successful CI run when compiling for release.

pmnlla, 2023-12-29:
Adding `overflow-checks = false` to my cargo config fixed the issue for debug builds as well. I dislike nulling over error reporting in debug builds but in this case this kind of reporting is not needed, so I'll mark this as closed.

Thank you so much! 🎉

closed 2023-12-29 by pmnlla

Rahix, 2023-12-29:
If you limit the disabled overflow checks to `compiler-builtins`, it is fine to do because `compiler-builtins` is proven to never run into overflow conditions.  You can do this using this configuration:

```toml
[profile.dev.package.compiler_builtins]
overflow-checks = false
```

However, IIRC this leads to a spurious warning from cargo (that can be safely ignored).