Rahix/avr-hal #483: add easy way to build the whole workspace

rursprung, 2023-12-28:
running `cargo build` on the workspace level fails because no features have been selected.

it'd be great if there'd be a way to run the build for the whole workspace, as it'd make development a lot easier. this is also visible in the CI script which needs to list everything explicitly (running the risk to miss adding new crates there in which case it won't build/test them).

i'm not sure what the best option would be. maybe a [`justfile`](https://github.com/casey/just)? it seems that this is being used regularly in the (embedded) rust world (though i've no personal experience with it).

Rahix, 2024-01-11:
My personal suggestion would be to just (ab)use the CI for this ;)

I do like how the CI currently builds each target individually as this parallelizes builds and also nicely reports the status of each target separately.

I don't think a Justfile or other kind of script would really improve the build coverage situation.  You still have to explictly add every target in there which people won't do unless you explicitly ask them to... And that's what I am currently doing with the CI matrix anyway.

Another reason why I'd like to leave this up to the CI is that for such a world build it makes sense to have the environment more controlled than when doing "simple" single target builds.  If the intention is to check for "everything still works as intended", then the reference environment for "everything works" should be used - and that environment is the CI build at the moment...

Workflow wise, feel free to send as many draft PRs as you want, to check for build issues.

Would this be a "solution" that's acceptable for you?

Rahix, 2024-01-11:
(and yeah, it sucks that cargo workspaces interact so badly with projects where different crates need to be built for different targets)