Rahix/avr-hal #507: avrdude failing, possible due to ravedude

Johnwf01john, 2024-02-16:
Her is output when cargo run is entered
PS C:\Users\john\blink> cargo run
warning: blink v0.1.0 (C:\Users\john\blink) ignoring invalid dependency `ravedude` which is missing a lib target
    Finished dev [optimized + debuginfo] target(s) in 0.22s
     Running `ravedude micro -P COM4 target\avr-atmega32u4\debug\blink.elf`
       Board Arduino Micro
Warning: this board cannot reset itself.

    Reset the board by pressing the reset button once.

Once reset, press ENTER here:
 Programming target\avr-atmega32u4\debug\blink.elf => COM4
avrdude warning: mosi is deprecated, will be removed in v8.0, use sdo [C:\Users\john\AppData\Local\Temp\.avrdude-YjVonk.conf:399]
avrdude warning: miso is deprecated, will be removed in v8.0, use sdi [C:\Users\john\AppData\Local\Temp\.avrdude-YjVonk.conf:400]
avrdude warning: mosi is deprecated, will be removed in v8.0, use sdo [C:\Users\john\AppData\Local\Temp\.avrdude-YjVonk.conf:428]
avrdude warning: miso is deprecated, will be removed in v8.0, use sdi [C:\Users\john\AppData\Local\Temp\.avrdude-YjVonk.conf:429]
avrdude warning: mosi is deprecated, will be removed in v8.0, use sdo [C:\Users\john\AppData\Local\Temp\.avrdude-YjVonk.conf:461]
avrdude warning: miso is deprecated, will be removed in v8.0, use sdi [C:\Users\john\AppData\Local\Temp\.avrdude-YjVonk.conf:462]
avrdude warning: mosi is deprecated, will be removed in v8.0, use sdo [C:\Users\john\AppData\Local\Temp\.avrdude-YjVonk.conf:495]
avrdude warning: miso is deprecated, will be removed in v8.0, use sdi [C:\Users\john\AppData\Local\Temp\.avrdude-YjVonk.conf:496]
avrdude warning: mosi is deprecated, will be removed in v8.0, use sdo [C:\Users\john\AppData\Local\Temp\.avrdude-YjVonk.conf:522]
avrdude warning: miso is deprecated, will be removed in v8.0, use sdi [C:\Users\john\AppData\Local\Temp\.avrdude-YjVonk.conf:523]
avrdude warning: mosi is deprecated, will be removed in v8.0, use sdo [C:\Users\john\AppData\Local\Temp\.avrdude-YjVonk.conf:563]
avrdude warning: miso is deprecated, will be removed in v8.0, use sdi [C:\Users\john\AppData\Local\Temp\.avrdude-YjVonk.conf:564]
avrdude warning: mosi is deprecated, will be removed in v8.0, use sdo [C:\Users\john\AppData\Local\Temp\.avrdude-YjVonk.conf:583]
avrdude warning: miso is deprecated, will be removed in v8.0, use sdi [C:\Users\john\AppData\Local\Temp\.avrdude-YjVonk.conf:584]
avrdude warning: mosi is deprecated, will be removed in v8.0, use sdo [C:\Users\john\AppData\Local\Temp\.avrdude-YjVonk.conf:602]
avrdude warning: miso is deprecated, will be removed in v8.0, use sdi [C:\Users\john\AppData\Local\Temp\.avrdude-YjVonk.conf:603]
avrdude warning: mosi is deprecated, will be removed in v8.0, use sdo [C:\Users\john\AppData\Local\Temp\.avrdude-YjVonk.conf:648]
avrdude warning: miso is deprecated, will be removed in v8.0, use sdi [C:\Users\john\AppData\Local\Temp\.avrdude-YjVonk.conf:649]
avrdude warning: miso is deprecated, will be removed in v8.0, use sdi [C:\Users\john\AppData\Local\Temp\.avrdude-YjVonk.conf:732]
avrdude warning: mosi is deprecated, will be removed in v8.0, use sdo [C:\Users\john\AppData\Local\Temp\.avrdude-YjVonk.conf:734]
avrdude warning: miso is deprecated, will be removed in v8.0, use sdi [C:\Users\john\AppData\Local\Temp\.avrdude-YjVonk.conf:743]
avrdude warning: mosi is deprecated, will be removed in v8.0, use sdo [C:\Users\john\AppData\Local\Temp\.avrdude-YjVonk.conf:745]
avrdude warning: miso is deprecated, will be removed in v8.0, use sdi [C:\Users\john\AppData\Local\Temp\.avrdude-YjVonk.conf:755]
avrdude warning: mosi is deprecated, will be removed in v8.0, use sdo [C:\Users\john\AppData\Local\Temp\.avrdude-YjVonk.conf:757]
avrdude warning: miso is deprecated, will be removed in v8.0, use sdi [C:\Users\john\AppData\Local\Temp\.avrdude-YjVonk.conf:768]
avrdude warning: mosi is deprecated, will be removed in v8.0, use sdo [C:\Users\john\AppData\Local\Temp\.avrdude-YjVonk.conf:770]
avrdude warning: miso is deprecated, will be removed in v8.0, use sdi [C:\Users\john\AppData\Local\Temp\.avrdude-YjVonk.conf:791]
avrdude warning: mosi is deprecated, will be removed in v8.0, use sdo [C:\Users\john\AppData\Local\Temp\.avrdude-YjVonk.conf:793]
avrdude warning: miso is deprecated, will be removed in v8.0, use sdi [C:\Users\john\AppData\Local\Temp\.avrdude-YjVonk.conf:815]
avrdude warning: mosi is deprecated, will be removed in v8.0, use sdo [C:\Users\john\AppData\Local\Temp\.avrdude-YjVonk.conf:817]
avrdude warning: mosi is deprecated, will be removed in v8.0, use sdo [C:\Users\john\AppData\Local\Temp\.avrdude-YjVonk.conf:1184]
avrdude warning: miso is deprecated, will be removed in v8.0, use sdi [C:\Users\john\AppData\Local\Temp\.avrdude-YjVonk.conf:1185]
avrdude warning: mosi is deprecated, will be removed in v8.0, use sdo [C:\Users\john\AppData\Local\Temp\.avrdude-YjVonk.conf:1195]
avrdude warning: miso is deprecated, will be removed in v8.0, use sdi [C:\Users\john\AppData\Local\Temp\.avrdude-YjVonk.conf:1197]
avrdude warning: mosi is deprecated, will be removed in v8.0, use sdo [C:\Users\john\AppData\Local\Temp\.avrdude-YjVonk.conf:1218]
avrdude warning: miso is deprecated, will be removed in v8.0, use sdi [C:\Users\john\AppData\Local\Temp\.avrdude-YjVonk.conf:1219]
avrdude warning: mosi is deprecated, will be removed in v8.0, use sdo [C:\Users\john\AppData\Local\Temp\.avrdude-YjVonk.conf:1236]
avrdude warning: miso is deprecated, will be removed in v8.0, use sdi [C:\Users\john\AppData\Local\Temp\.avrdude-YjVonk.conf:1237]
avrdude warning: mosi is deprecated, will be removed in v8.0, use sdo [C:\Users\john\AppData\Local\Temp\.avrdude-YjVonk.conf:1252]
avrdude warning: miso is deprecated, will be removed in v8.0, use sdi [C:\Users\john\AppData\Local\Temp\.avrdude-YjVonk.conf:1253]
avrdude warning: mosi is deprecated, will be removed in v8.0, use sdo [C:\Users\john\AppData\Local\Temp\.avrdude-YjVonk.conf:1263]
avrdude warning: miso is deprecated, will be removed in v8.0, use sdi [C:\Users\john\AppData\Local\Temp\.avrdude-YjVonk.conf:1264]
avrdude warning: mosi is deprecated, will be removed in v8.0, use sdo [C:\Users\john\AppData\Local\Temp\.avrdude-YjVonk.conf:1274]
avrdude warning: miso is deprecated, will be removed in v8.0, use sdi [C:\Users\john\AppData\Local\Temp\.avrdude-YjVonk.conf:1275]
avrdude warning: mosi is deprecated, will be removed in v8.0, use sdo [C:\Users\john\AppData\Local\Temp\.avrdude-YjVonk.conf:1285]
avrdude warning: miso is deprecated, will be removed in v8.0, use sdi [C:\Users\john\AppData\Local\Temp\.avrdude-YjVonk.conf:1286]
avrdude warning: mosi is deprecated, will be removed in v8.0, use sdo [C:\Users\john\AppData\Local\Temp\.avrdude-YjVonk.conf:1302]
avrdude warning: miso is deprecated, will be removed in v8.0, use sdi [C:\Users\john\AppData\Local\Temp\.avrdude-YjVonk.conf:1306]
avrdude warning: mosi is deprecated, will be removed in v8.0, use sdo [C:\Users\john\AppData\Local\Temp\.avrdude-YjVonk.conf:1319]
avrdude warning: miso is deprecated, will be removed in v8.0, use sdi [C:\Users\john\AppData\Local\Temp\.avrdude-YjVonk.conf:1320]
avrdude warning: mosi is deprecated, will be removed in v8.0, use sdo [C:\Users\john\AppData\Local\Temp\.avrdude-YjVonk.conf:1330]
avrdude warning: miso is deprecated, will be removed in v8.0, use sdi [C:\Users\john\AppData\Local\Temp\.avrdude-YjVonk.conf:1331]
avrdude warning: mosi is deprecated, will be removed in v8.0, use sdo [C:\Users\john\AppData\Local\Temp\.avrdude-YjVonk.conf:1341]
avrdude warning: miso is deprecated, will be removed in v8.0, use sdi [C:\Users\john\AppData\Local\Temp\.avrdude-YjVonk.conf:1342]
avrdude warning: miso is deprecated, will be removed in v8.0, use sdi [C:\Users\john\AppData\Local\Temp\.avrdude-YjVonk.conf:1351]
avrdude warning: mosi is deprecated, will be removed in v8.0, use sdo [C:\Users\john\AppData\Local\Temp\.avrdude-YjVonk.conf:1353]
avrdude warning: mosi is deprecated, will be removed in v8.0, use sdo [C:\Users\john\AppData\Local\Temp\.avrdude-YjVonk.conf:1375]
avrdude warning: miso is deprecated, will be removed in v8.0, use sdi [C:\Users\john\AppData\Local\Temp\.avrdude-YjVonk.conf:1376]
[... 76 lines cut ...]
avrdude warning: has_jtag is deprecated, will be removed in v8.0, use prog_modes [C:\Users\john\AppData\Local\Temp\.avrdude-YjVonk.conf:13554]
avrdude warning: has_jtag is deprecated, will be removed in v8.0, use prog_modes [C:\Users\john\AppData\Local\Temp\.avrdude-YjVonk.conf:13643]
[... cut ...]
avrdude warning: has_jtag is deprecated, will be removed in v8.0, use prog_modes [C:\Users\john\AppData\Local\Temp\.avrdude-YjVonk.conf:14912]
avrdude warning: System wide configuration file version ()
        does not match Avrdude build version (7.3)
avrdude error: cannot find programmer id avr109
use -c? to see all possible programmers
Error: avrdude failed

error: process didn't exit successfully: `ravedude micro -P COM4 target\avr-atmega32u4\debug\blink.elf` (exit code: 1)
PS C:\Users\john\blink>
arduino ver2.3.1 programes micro OK but using rust in windows 11 PowerShell does not. 

any idea's welcolme

vs-mark-graham-smith, 2024-02-16:
I have a similar issue on macOS

The output below:

    Finished dev [optimized + debuginfo] target(s) in 0.12s
     Running `ravedude mega2560 -cb 57600 target/avr-atmega2560/debug/rus-blnk.elf`
       Board Arduino Mega 2560
 Programming target/avr-atmega2560/debug/rus-blnk.elf => /dev/cu.usbmodem11134201
avrdude warning: mosi is deprecated, will be removed in v8.0, use sdo [/private/var/folders/ly/yj8sly2x4fq749c_v5mglcnm0000gn/T/.avrdude-EhlIHL.conf:399]
avrdude warning: miso is deprecated, will be removed in v8.0, use sdi [/private/var/folders/ly/yj8sly2x4fq749c_v5mglcnm0000gn/T/.avrdude-EhlIHL.conf:400]
avrdude warning: mosi is deprecated, will be removed in v8.0, use sdo [/private/var/folders/ly/yj8sly2x4fq749c_v5mglcnm0000gn/T/.avrdude-EhlIHL.conf:428]
avrdude warning: miso is deprecated, will be removed in v8.0, use sdi [/private/var/folders/ly/yj8sly2x4fq749c_v5mglcnm0000gn/T/.avrdude-EhlIHL.conf:429]
avrdude warning: mosi is deprecated, will be removed in v8.0, use sdo [/private/var/folders/ly/yj8sly2x4fq749c_v5mglcnm0000gn/T/.avrdude-EhlIHL.conf:461]
avrdude warning: miso is deprecated, will be removed in v8.0, use sdi [/private/var/folders/ly/yj8sly2x4fq749c_v5mglcnm0000gn/T/.avrdude-EhlIHL.conf:462]
avrdude warning: mosi is deprecated, will be removed in v8.0, use sdo [/private/var/folders/ly/yj8sly2x4fq749c_v5mglcnm0000gn/T/.avrdude-EhlIHL.conf:495]
avrdude warning: miso is deprecated, will be removed in v8.0, use sdi [/private/var/folders/ly/yj8sly2x4fq749c_v5mglcnm0000gn/T/.avrdude-EhlIHL.conf:496]
avrdude warning: mosi is deprecated, will be removed in v8.0, use sdo [/private/var/folders/ly/yj8sly2x4fq749c_v5mglcnm0000gn/T/.avrdude-EhlIHL.conf:522]
avrdude warning: miso is deprecated, will be removed in v8.0, use sdi [/private/var/folders/ly/yj8sly2x4fq749c_v5mglcnm0000gn/T/.avrdude-EhlIHL.conf:523]
avrdude warning: mosi is deprecated, will be removed in v8.0, use sdo [/private/var/folders/ly/yj8sly2x4fq749c_v5mglcnm0000gn/T/.avrdude-EhlIHL.conf:563]
avrdude warning: miso is deprecated, will be removed in v8.0, use sdi [/private/var/folders/ly/yj8sly2x4fq749c_v5mglcnm0000gn/T/.avrdude-EhlIHL.conf:564]
avrdude warning: mosi is deprecated, will be removed in v8.0, use sdo [/private/var/folders/ly/yj8sly2x4fq749c_v5mglcnm0000gn/T/.avrdude-EhlIHL.conf:583]
avrdude warning: miso is deprecated, will be removed in v8.0, use sdi [/private/var/folders/ly/yj8sly2x4fq749c_v5mglcnm0000gn/T/.avrdude-EhlIHL.conf:584]
avrdude warning: mosi is deprecated, will be removed in v8.0, use sdo [/private/var/folders/ly/yj8sly2x4fq749c_v5mglcnm0000gn/T/.avrdude-EhlIHL.conf:602]
avrdude warning: miso is deprecated, will be removed in v8.0, use sdi [/private/var/folders/ly/yj8sly2x4fq749c_v5mglcnm0000gn/T/.avrdude-EhlIHL.conf:603]
avrdude warning: mosi is deprecated, will be removed in v8.0, use sdo [/private/var/folders/ly/yj8sly2x4fq749c_v5mglcnm0000gn/T/.avrdude-EhlIHL.conf:648]
avrdude warning: miso is deprecated, will be removed in v8.0, use sdi [/private/var/folders/ly/yj8sly2x4fq749c_v5mglcnm0000gn/T/.avrdude-EhlIHL.conf:649]
avrdude warning: miso is deprecated, will be removed in v8.0, use sdi [/private/var/folders/ly/yj8sly2x4fq749c_v5mglcnm0000gn/T/.avrdude-EhlIHL.conf:732]
avrdude warning: mosi is deprecated, will be removed in v8.0, use sdo [/private/var/folders/ly/yj8sly2x4fq749c_v5mglcnm0000gn/T/.avrdude-EhlIHL.conf:734]
avrdude warning: miso is deprecated, will be removed in v8.0, use sdi [/private/var/folders/ly/yj8sly2x4fq749c_v5mglcnm0000gn/T/.avrdude-EhlIHL.conf:743]
avrdude warning: mosi is deprecated, will be removed in v8.0, use sdo [/private/var/folders/ly/yj8sly2x4fq749c_v5mglcnm0000gn/T/.avrdude-EhlIHL.conf:745]
avrdude warning: miso is deprecated, will be removed in v8.0, use sdi [/private/var/folders/ly/yj8sly2x4fq749c_v5mglcnm0000gn/T/.avrdude-EhlIHL.conf:755]
avrdude warning: mosi is deprecated, will be removed in v8.0, use sdo [/private/var/folders/ly/yj8sly2x4fq749c_v5mglcnm0000gn/T/.avrdude-EhlIHL.conf:757]
avrdude warning: miso is deprecated, will be removed in v8.0, use sdi [/private/var/folders/ly/yj8sly2x4fq749c_v5mglcnm0000gn/T/.avrdude-EhlIHL.conf:768]
avrdude warning: mosi is deprecated, will be removed in v8.0, use sdo [/private/var/folders/ly/yj8sly2x4fq749c_v5mglcnm0000gn/T/.avrdude-EhlIHL.conf:770]
avrdude warning: miso is deprecated, will be removed in v8.0, use sdi [/private/var/folders/ly/yj8sly2x4fq749c_v5mglcnm0000gn/T/.avrdude-EhlIHL.conf:791]
avrdude warning: mosi is deprecated, will be removed in v8.0, use sdo [/private/var/folders/ly/yj8sly2x4fq749c_v5mglcnm0000gn/T/.avrdude-EhlIHL.conf:793]
avrdude warning: miso is deprecated, will be removed in v8.0, use sdi [/private/var/folders/ly/yj8sly2x4fq749c_v5mglcnm0000gn/T/.avrdude-EhlIHL.conf:815]
avrdude warning: mosi is deprecated, will be removed in v8.0, use sdo [/private/var/folders/ly/yj8sly2x4fq749c_v5mglcnm0000gn/T/.avrdude-EhlIHL.conf:817]
avrdude warning: mosi is deprecated, will be removed in v8.0, use sdo [/private/var/folders/ly/yj8sly2x4fq749c_v5mglcnm0000gn/T/.avrdude-EhlIHL.conf:1184]
avrdude warning: miso is deprecated, will be removed in v8.0, use sdi [/private/var/folders/ly/yj8sly2x4fq749c_v5mglcnm0000gn/T/.avrdude-EhlIHL.conf:1185]
avrdude warning: mosi is deprecated, will be removed in v8.0, use sdo [/private/var/folders/ly/yj8sly2x4fq749c_v5mglcnm0000gn/T/.avrdude-EhlIHL.conf:1195]
[... cut ...]
avrdude warning: ATtiny1634's flash writepage misses a necessary address bit a5 [/private/var/folders/ly/yj8sly2x4fq749c_v5mglcnm0000gn/T/.avrdude-EhlIHL.conf:14723]
avrdude warning: has_tpi is deprecated, will be removed in v8.0, use prog_modes [/private/var/folders/ly/yj8sly2x4fq749c_v5mglcnm0000gn/T/.avrdude-EhlIHL.conf:14789]
avrdude warning: has_jtag is deprecated, will be removed in v8.0, use prog_modes [/private/var/folders/ly/yj8sly2x4fq749c_v5mglcnm0000gn/T/.avrdude-EhlIHL.conf:14912]
avrdude warning: System wide configuration file version ()
        does not match Avrdude build version (7.3)
avrdude error: cannot find programmer id wiring

Rahix, 2024-02-16:
Can reproduce.  It seems our avrdude configuration is finally too old to work with the most recent avrdude versions.  I'm not sure if we can simply update the config as I have no clue how many people are still using older avrdude versions...

There are some long overdue ravedude changes I want to make anyway - one of them is to have ravedude manage its own avrdude version instead of relying on a system-provided one.  That way we have full control over the avrdude version that's used.  I think that's the best solution to the problem at hand for now...

Rahix, 2024-02-16:
Ah, this is a duplicate of #498.  Closing in favor of the other issue.

closed 2024-02-16 by Rahix